MarlinFirmware/Marlin #4416: Delta dual extruders, second extruder slanted yet the first exturder work fine.

prophet-id, 2016-07-26:
1.1.0 RC6 without auto bed leveling.
Slicer: S3D

So if I print with the 1st extruder ONLY, it works well. If I print with 2nd extruder ONLY, works well.
But when I use BOTH extruder (as object and support), the 2nd extruder print on slanted plane, while the 1st extruder is fine, printing on the bed plane normally.

Try to change the EXTRUDER_OFFSET_X and EXTRUDER_OFFSET_Y to (0.0 , 0.0) for both, the slanted issue is gone, although as expected the print from the 1st extruder and the 2nd extruder has offset as their physical nozzle's positions are offset.
My current hypothesis, something wrong related with the EXTRUDER_OFFSET_X and Y.

Is there anyone running the same issue? (sorry if I missed)
Any idea for debug (fixing)?

thinkyhead, 2016-07-27:
@prophet-id What version of Marlin are you using?

prophet-id, 2016-07-27:
it is 1.1.0 RC6 (my first line on the message)