MarlinFirmware/Marlin #4418: G29 on corexy moves bed away from endstop first

MichaelBrock, 2016-07-26:
I have a Corxy printer.  In "rest" position the bed sits at the bottom and it is raised to home.  With the latest RCBugFix, the printer attempts to push the bed downwards before raising it for homing.  Back in the day we would set Z_RAISE_BEFORE_HOMING  to 0 to prevent this.  A while ago I updated to RC 6 and configured MIN_Z_HEIGHT_FOR_HOMING to 0  this seemed to do the trick.  I started using RCBugFix a few weeks ago with no issues.   However, yesterday I I just jumped to the latest RcBugFix and before homing Z the printer is again driving the bed downwards (or attempting to).  To be clear it is homing X and Y (safe homing is set) first with no issue but before doing Z it is moving the bed downwards.  I'm digging my way through the code right now.  The issue isn't in the "Raise Z before homing" code as that happens  before either x or y is homed (z is homed last with same homing obviously).  Any ideas?

Blue-Marlin, 2016-07-26:
The second 'raise' is for deploying the probe. (Z_RAISE_PROBE_DEPLOY_STOW)

Park your bed some cm above.
Lift your bed (decrease the distance between nozzle and bed) before homing. 
Use a z-max endstop for homing.
Use a conventional z-min endstop.

MichaelBrock, 2016-07-26:
I just found that in the code.  Z_RAISE_PROBE_DEPLOY_STOW can't be set to 0 unfortunately. I realize that manually lifting the bed off the bottom to give it room to be lowered will "fix" the problem.   I would prefer to fix this without having to do that or reconfigure the entire printer though.  It seems to me that the code to deploy the probe shouldn't be run at all if FIX_MOUNTED_PROBE is being used.  I guess I'm going to jump into using a customized version of Marlin.

Blue-Marlin, 2016-07-27:
There is __NO__ way to make homing with a probe foolproof. 
Ether you risk to scratch the bed or banging z-max. Usually scratching the bed is the more expensive problem (Nozzle damage, broken nozzle holder, damaged bed (surface), broken probe )

If you want to complain about unexpected 'extra' raises please join #4282.

thinkyhead, 2016-07-27:
If you want to compile with those set to zero, comment out or make the following change to `SanityCheck.cpp`:
```diff
  /**
   * Make sure Z raise values are set
   */
  #if defined(Z_RAISE_BEFORE_PROBING) || defined(Z_RAISE_AFTER_PROBING)
    #error "Z_RAISE_(BEFORE|AFTER)_PROBING are deprecated. Use Z_RAISE_PROBE_DEPLOY_STOW instead."
  #elif !defined(Z_RAISE_PROBE_DEPLOY_STOW)
    #error "You must set Z_RAISE_PROBE_DEPLOY_STOW in your configuration."
  #elif !defined(Z_RAISE_BETWEEN_PROBINGS)
    #error "You must set Z_RAISE_BETWEEN_PROBINGS in your configuration."
- #elif Z_RAISE_PROBE_DEPLOY_STOW < 1
-   #error "Probes need Z_RAISE_PROBE_DEPLOY_STOW >= 1."
- #elif Z_RAISE_BETWEEN_PROBINGS < 1
-   #error "Probes need Z_RAISE_BETWEEN_PROBINGS >= 1."
  #endif
```

thinkyhead, 2016-07-27:
> the code to deploy the probe shouldn't be run at all if `FIX_MOUNTED_PROBE` is being used

We might as well keep the option, for the possible case where one starts `G29` with the nozzle already touching the bed after a `G28`. The extra raise might be needed to disengage the switch in that case.

closed 2016-07-27 by thinkyhead