MarlinFirmware/Marlin #4421: UBL Branch testing on ML2 FreeFab3d

fr33dimension, 2016-07-27:
I am new to mesh abl, but I am using save to Eeprom ABL for years now with a proximity sensor.
What I noticed after successfully flashing and running a G29 P1, that it doesn't appear to save the matrix to eeprom. What am I missing

Roxy-3D, 2016-07-27:
You can do a G29 W to see what it says.

But the way the system is organized, you have a Mesh in memory.  That is where you build and edit it.   But it is in memory until you save it.   Depending on the Mesh size you have declared, and how much EEPROM you have, you get to save more or less different Mesh's.   

If you do a G29 P1 M  and let it sample a few points, then stop it by holding the Encoder Wheel pressed.   You can do a G29 M and see the Mesh at it's current state.   Then do a G29 S 0 and it should save.
Reset your machine, and the last selected (saved) Mesh should be automatically loaded.   Do a G29 M and you should see it in the same state as when you saved it (even though you reset the machine).

fr33dimension, 2016-07-27:
G29 W
21:17:49.138 : Mesh invalidated.  Probing mesh.
21:25:23.482 : echo:Settings Stored (263 bytes)
21:27:37.146 : Unified Bed Leveling System Inactive.
21:27:37.187 : Mesh: FFFF Loaded.
21:27:37.191 : G29_Correction_Fade_Height : 10.00
21:27:37.334 : Free EEPROM space starts at: 0x0120
21:27:37.376 : end of EEPROM              : 0FFF
21:27:37.376 : z_value[][] size: 534  400
21:27:37.416 : EEPROM free for UBL: 0x0EDF
21:27:37.457 : EEPROM can hold 0x0009 meshes.
21:27:37.498 : sizeof(stat)     :0043
21:27:37.498 : MESH_NUM_X_POINTS  10
21:27:37.503 : MESH_NUM_Y_POINTS  10
21:27:37.503 : MESH_MIN_X         0
21:27:37.503 : MESH_MIN_Y         0
21:27:37.503 : MESH_MAX_X         195
21:27:37.503 : MESH_MAX_Y         200
21:27:37.507 : MESH_X_DIST        21.67
21:27:37.507 : MESH_Y_DIST        22.22
21:27:37.507 : z_offset           0.00
21:27:37.510 : Unified Bed Leveling sanity checks passed.
21:28:11.749 : Mesh saved at offset 0x0E2C
21:28:11.749 : Done.

Roxy-3D, 2016-07-27:
If you do a G29 M you can see what the current Mesh looks like in memory.  But the Mesh: FFFF Loaded means you have never saved or loaded a Mesh.   You have room to store 9 Mesh's.  0 through 8.

fr33dimension, 2016-07-27:
Thanks Roxy, I get the info below when resetting and running a G29 W. G29 M has not effect.
07:51:20.616 : Unified Bed Leveling:
07:51:20.616 : echo:System is: Deactive
07:51:20.616 : Active Mesh Slot: 1
07:51:20.616 : EEPROM can hold 9 meshes.
07:51:20.616 : MESH_NUM_X_POINTS  10
07:51:20.620 : MESH_NUM_Y_POINTS  10
07:51:20.620 : MESH_MIN_X         0
07:51:20.620 : MESH_MIN_Y         0
07:51:20.620 : MESH_MAX_X         195
07:51:20.624 : MESH_MAX_Y         200
07:51:20.624 : MESH_X_DIST        21.67
07:51:20.624 : MESH_Y_DIST        22.22
07:51:20.625 : echo:Material heatup parameters:
07:51:20.628 : echo:  M145 S0 H180 B50 F0
07:51:20.628 : echo:  M145 S1 H240 B85 F0
07:51:20.628 : echo:PID settings:
07:51:20.633 : echo:  M301 P20.81 I1.34 D80.55 C100.00 L20
07:51:20.633 : echo:Filament settings: Disabled
07:51:20.633 : echo:  M200 D1.75
07:51:20.633 : echo:  M200 D0
07:51:20.636 : echo:Z-Probe Offset (mm):
07:51:20.636 : echo:  M851 Z-0.90
07:51:24.119 : echo:DEBUG:INFO,ERRORS

fr33dimension, 2016-07-27:
Ran another G29 P1, then saved G29 S 0, reset machine and got this with G29 W (G29 M does nothing)

08:05:25.364 : Unified Bed Leveling System Inactive.
08:05:25.405 : Mesh: 0000 Loaded.
08:05:25.409 : G29_Correction_Fade_Height : 10.00
08:05:25.552 : Free EEPROM space starts at: 0x0120
08:05:25.593 : end of EEPROM              : 0FFF
08:05:25.593 : z_value[][] size: 534  400
08:05:25.634 : EEPROM free for UBL: 0x0EDF
08:05:25.675 : EEPROM can hold 0x0009 meshes.
08:05:25.716 : sizeof(stat)     :0043
08:05:25.716 : MESH_NUM_X_POINTS  10
08:05:25.721 : MESH_NUM_Y_POINTS  10
08:05:25.721 : MESH_MIN_X         0
08:05:25.721 : MESH_MIN_Y         0
08:05:25.721 : MESH_MAX_X         195
08:05:25.721 : MESH_MAX_Y         200
08:05:25.725 : MESH_X_DIST        21.67
08:05:25.725 : MESH_Y_DIST        22.22
08:05:25.725 : z_offset           0.00
08:05:25.728 : Unified Bed Leveling sanity checks passed.

fr33dimension, 2016-07-27:
Actually, G29 M makes the board freeze, I have to reset to get control again.

Roxy-3D, 2016-07-27:
That doesn't make any sense.   (I believe it is happening.   I just have never seen anything like that.)

The 'M' option says to print a Map of the bed's current Mesh Topology.   It can be added to most parameters or G29 commands.   And I just loaded the most recent code on my printer to make sure the 'M' map option still works.  It does.

Let me check the code and see if I can find something...   But try adding an M to some other G29 commands and see what it does.    Does it do the commands and then lock up?  Does it lock up before doing anything?   That would be helpful to know to isolate what the failure is.

fr33dimension, 2016-07-27:
No I cannot see the map printed at each point, I can't see anything of that nature. I tried G29 P1 M, it prints TEST12ok

fr33dimension, 2016-07-27:
G29 P0 M freezes the board.

Roxy-3D, 2016-07-27:
I don't understand what you mean it prints TEST12ok.   Are you talking about the test patterns?   Are you saying you can print a map of the test patterns?   

One thing we can do very quickly is see how far into the G29 code it is getting before it hangs.  We can do a G29 Q 99 and see if it complains.   Then a G29 P 99 and see if it complains.   Then a G29 L 99 and see if it complains.  And then a G29 S 99 to see if it complains.

The next thing in the code after the 'S' (save) parameter is the M parameter.      

We might need to add some debug code to see where it is getting to.

fr33dimension, 2016-07-27:
I will try this next. When I said print I meant that it displayed TEST12ok, did not see anything other than that.
[image]

fr33dimension, 2016-07-27:
Here is what the G29 Q99, -P99, -L99, -S99 did generate

12:11:15.784 : Invalid Test_Pattern value.  (0-4)
12:11:51.093 : Invalid Phase value.  (0-4)
12:12:08.992 : ?EEPROM storage not available for use.
12:12:30.214 : ?EEPROM storage not available for use.

fr33dimension, 2016-07-27:
This is what caught my eye earlier, it says system deactive

10:47:42.179 : Unified Bed Leveling:
10:47:42.179 : echo:System is: Deactive
10:47:42.179 : Active Mesh Slot: 0
10:47:42.179 : EEPROM can hold 9 meshes.

Roxy-3D, 2016-07-27:
Yes, but even with the system deactivated, the Mesh Map should be printable.

And I reviewed the code to display the Mesh Map.  There is nothing fancy in it.    Did you see this question:

> When you do a G29 M does it print out anything? One of the first things the Map code does is print a title: SERIAL_PROTOCOLLNPGM("Bed Topography Report:\n"); Do you see the Bed Topography Report: title show up on your PronterFace screen?

fr33dimension, 2016-07-27:
No, nothing like a topography report
12:25:40.847 : 0B42
12:25:40.847 : Mesh invalidated.  Probing mesh.
12:33:36.731 : Mesh saved at offset 0x0C9C
12:33:36.731 : Done.
This is what I see in the log during and after the G29 P 1 and G29 S 1

Roxy-3D, 2016-07-27:
And G29 P1 M locks up the machine????

fr33dimension, 2016-07-27:
No G29 M does, G29 P1 M displays "TEST12ok"

fr33dimension, 2016-07-27:
And G29 M P1 after reset does Home all then displays "Mesh zeroed"

Roxy-3D, 2016-07-27:
It almost sounds like the firmware did not get flashed correctly.   What kind of board (and processor) do you have?   Can you download the latest code and put your current configuration.h and configuration_adv.h file with it and flash the board again?

Something else is going on here.  I don't know what is causing the problem but the Map code is very straight forward and should not be causing problems.   Can you also paste your Configuration.h file into a post here and I'll take a closer look at it.

fr33dimension, 2016-07-27:
I will try that later today. It's a Getech Chinese rambo clone usually works alright, not sure of the chip.
I have other machines, but I have work on them right now. 

[Configuration H.txt](https://github.com/MarlinFirmware/Marlin/files/386654/Configuration.H.txt)
[Configuration_adv H.txt](https://github.com/MarlinFirmware/Marlin/files/386656/Configuration_adv.H.txt)

Roxy-3D, 2016-07-27:
Did you see that at line 1005 you have the discount RepRap LCD controller turned on?
That doesn't seem compatible with Viki controller.   

#define REPRAP_DISCOUNT_SMART_CONTROLLER

fr33dimension, 2016-07-27:
That must have been selected before, the viki is not working right with RC6 yet. I will change that too . 
Thanks

fr33dimension, 2016-07-27:
Didn't make any difference...FYI.....So I downloaded this file again https://github.com/MarlinFirmware/Marlin/tree/devel-ubl and flashed it, same thing. It's is probing again now in if I have some new developments I will post them.

fr33dimension, 2016-07-27:
Wait a minute....I think I got it. Had acknowledgements turned off in repetier. Now I turned it on and I can see stuff like this
15:51:26.596 : ok T:24.9 /0.0 B:24.2 /0.0 B@:0 @:0
15:51:29.664 : ok T:24.9 /0.0 B:24.2 /0.0 B@:0 @:0
15:51:32.729 : ok T:24.9 /0.0 B:24.3 /0.0 B@:0 @:0
15:51:35.784 : ok T:24.8 /0.0 B:24.2 /0.0 B@:0 @:0
15:51:38.943 : ok T:24.8 /0.0 B:24.3 /0.0 B@:0 @:0
15:51:41.945 : ok T:24.9 /0.0 B:24.3 /0.0 B@:0 @:0
15:51:44.975 : ok T:24.9 /0.0 B:24.3 /0.0 B@:0 @:0
15:51:46.441 : ok
15:51:46.441 : ok
15:51:46.445 : ok
15:51:47.006 : ok
15:51:47.011 : ok
15:51:47.011 : ok
15:51:48.036 : ok T:24.8 /0.0 B:24.2 /0.0 B@:0 @:0
15:51:53.798 : X:0.00 Y:0.00 Z:7.90 E:0.00 Count X: 0 Y:0 Z:25280
15:51:53.798 : ok
15:51:53.800 : ok T:24.9 /0.0 B:24.2 /0.0 B@:0 @:0
15:51:54.163 : ok T:24.8 /0.0 B:24.2 /0.0 B@:0 @:0
15:51:57.224 : ok T:24.7 /0.0 B:24.3 /0.0 B@:0 @:0
15:52:00.284 : ok T:24.8 /0.0 B:24.3 /0.0 B@:0 @:0
15:52:03.351 : ok T:24.7 /0.0 B:24.3 /0.0 B@:0 @:0
15:52:06.416 : ok T:24.8 /0.0 B:24.3 /0.0 B@:0 @:0
15:52:09.475 : ok T:24.9 /0.0 B:24.2 /0.0 B@:0 @:0
15:52:12.539 : ok T:25.0 /0.0 B:24.3 /0.0 B@:0 @:0

Roxy-3D, 2016-07-27:
I loaded your Configuration.h and Configuration_adv.h file into the build directory and I sent it to my RAMPS board.   I could not duplicate the hang.    But I did almost melt my nozzle off of the machine.  Apparently, the Rambo board does something to lock on the pin that RAMPS uses for the first nozzle.

I then changed the Rambo to RAMPS and left everything else the way you had it.   It seems to work fine and the 'M' option does not cause any issues.   I don't know what we are fighting.   Do you have another machine with different boards and such we could try it on?

fr33dimension, 2016-07-28:
I will try it on a ramps board tomorrow.

thinkyhead, 2016-07-28:
> I then changed the Rambo to RAMPS

You flashed a RAMPS board with `MOTHERBOARD` set to `BOARD_RAMBO`? D'oh!

Roxy-3D, 2016-07-28:
> You flashed a RAMPS board with MOTHERBOARD set to BOARD_RAMBO? D'oh!

Well, I knew it wouldn't work.    I didn't expect it to turn on the heaters at full blast and keep them on.   Live and learn.    But my intentions were good.  I wanted the generated firmware to be identical to what he had so I could see if I could reduplicate the problem.

fr33dimension, 2016-07-28:
I just noticed when flashing the ramps board that the UBL requires and LCD. I don't have one on the machine with the ramps board but just turned on the reprap discount lcd and was able to flash. The viki that I have on the ML2, doesn't work right with the UBL firmware and RC6. It displays the info screen, but the menus don't even come on when the encoder wheel is pushed. Could this be part of the issue?

Roxy-3D, 2016-07-28:
In the ReadMe file for UBL, I put this text:
~~~~
The System was developed on a Prusa i3 machine with a 20x4 LCD display. The closer you are to the development system, the less likely you are to have issues.
~~~~
If you have an i3 type machine with a REPRAP_DISCOUNT_SMART_CONTROLLER, I expect the software will come up with no problems.   If you want to get it running on a different type of machine I'll help you as much as I can, but it is very difficult to debug these kind of things without hardware.

Do you have an i3 machine and a Discount 20x4 controller available?  Even if you really want this on a different machine, bringing it up on a system where it is known to work first is helpful because you will see how everything is setup and functions.

fr33dimension, 2016-07-28:
Upload to Ramps 13 EFB is stuck.....never stops uploading. I had the same the first time I flashed the UBL to the Rambo and had to reburn the bootloader. Any suggestions

fr33dimension, 2016-07-28:
I don't have an i3, but a Prusa Mendel. Should be very close of a setup. I got some LCD's around and can try that later.

Roxy-3D, 2016-07-28:
OK..  Please try the Prusa Mendel!    With a 20x4 LCD, I think it will come up.   

But on your flashing problem, that really confuses me.   There is nothing special about how the firmware image is generated.  It is some extra C code so the image ends up being a little bit bigger, but that has nothing to do with the boot loader.    The only thing this code does that hasn't been done in the past is it grabs the tail end (unused area) of the EEPROM and uses that to store Mesh data.    But the system has to be up and running for it to do that.

Are you trying to flash this onto a processor that only has 128Kb of program space?   That might be the explanation for the problems we are seeing.   The firmware image is very close to that size and the boot loader takes some of that space away from the user.    If you tried to put this on a processor without enough program space, we would see the kind of problems you described.   Incidentally,  all of the development was done on an Atmega2560 and I have roughly half of the program memory free.

fr33dimension, 2016-07-28:
On the ramps and rambo, there are Atmega 2560's . I will try the Prusa Mendel, as soon as I can flash stuff to it again....will be a day or so.

fr33dimension, 2016-07-28:
Was able to flash without burning the bootloader. Unplugged USB wire until right before the upload process starts and that did it.

Blue-Marlin, 2016-07-28:
You experienced a watchdog reset. (fast blinking LED)

fr33dimension, 2016-07-28:
Okay, that makes sense. Should I have waited longer?

Roxy-3D, 2016-07-28:
My guess (and Blue-Marlin will probably be able to confirm), by having the wrong display type defined, the firmware got stuck and wasn't able to continue.   It wasn't able to service the heaters and eventually the Watch Dog code said "For safety, we have to shut everything down."     The problem is, if the system can't get all the way back up, the Watch Dog fires again and the system never comes up again.

fr33dimension, 2016-07-28:
Uploaded the UBL to the ramps board, I didn't change all the configs, just wanted to try the G29 M.
And it does the same thing as on the rambo. G29 M stops the board, communication timeout.....but here too, I had no LCD attached. Will try that next.

Roxy-3D, 2016-07-28:
Yes, right now you need to have an LCD Panel.

fr33dimension, 2016-07-28:
Went on to a different machine with a real rambo and reprap disc. LCD, did the mesh and it bedlevels during the print....good step, the leveing is not good, its not even much harder at the front of bed than back and I can still not get anything other then freeze the board with g29 M. The ramps unfortunately doens't have the lcd adapter pins broken out. I got one with that, but need some time to install.

Roxy-3D, 2016-07-28:
It doesn't really matter if the leveling is good.   Once you can get the G26 to print a Mesh Validation Pattern on the bed, it is very quick and easy to dial in any corrections you need.    But it certainly looks like we need to have a RepRap Discount 20x4 LCD Controller right now.    Once you get that on the board, things should go much smoother.

fr33dimension, 2016-07-29:
Perfect. So it needs the full graphic LCD not the reprap discount smart controller. I will wait to hear from you in that case.

Roxy-3D, 2016-07-29:
@fr33dimension Did you see this post and the subsequent posts?

https://github.com/MarlinFirmware/Marlin/issues/4437#issuecomment-236008669

It might be that is all you have to do to get your Full Graphics LCD Controller to work?   He is on a MakerFarm i3V with a Rambo board.

fr33dimension, 2016-07-29:
Thanks .....tried that, but since I only have the regular lcd I guess it doesn't work perfect. Will have one this afternoon. I noticed though that it measures fine, I can even manually probe the out of reach areas, just the G29 M is not liked without the full graphic LCD. 
Liked your comment about reprapers not reading instructions......I am definitely one of them...

Roxy-3D, 2016-07-29:
> Liked your comment about reprapers not reading instructions......I am definitely one of them...

Yeah, that was a joke.  I'm definitely a RepRap'er and I don't read directions very well.   

> Perfect. So it needs the full graphic LCD not the reprap discount smart controller. I will wait to hear from you in that case.

No.  I did not mean to imply that.   I think AdamFilip got his graphics controller going with that one line change.   He posted a picture of it.   I don't know which one that is, but you might.  The picture is here:  https://github.com/MarlinFirmware/Marlin/issues/4437#issuecomment-235993850

I have this one on its way to me from eBay:  http://www.ebay.com/itm/262174846139?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

We will get all of the LCD Panels going as soon as possible!   But right now, for sure, the Discount 20x4 LCD Panel does work with UBL !   That is what it was developed on.

fr33dimension, 2016-07-29:
Thanks Roxy, yes these are the Full Graphic LCD's, while that one that was selected on your config.H is the regular reprap discount lcd this one right here or similar http://www.ebay.com/itm/Geeetech-LCD-2004-20X4-display-with-controller-adaptor-for-RepRap-Sanguinololu-/231191036294?hash=item35d40f4186:g:oAQAAOxyTMdTNPJd  
This is what I have and it doesn't work.

Roxy-3D, 2016-07-29:
This is the one that UBL was developed on:

http://www.ebay.com/itm/251242416836?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

But I think all of the 20x4 LCD Panels are the same electrically.  I don't know that for a fact.   But we will get this LCD problem figured out ASAP.

fr33dimension, 2016-07-29:
Okay, that is the one that was selected and the one I have. The Full Graphic doesn't work with the rep discount selected and visa versa. I will try the Full in the next few days and get back.

Roxy-3D, 2016-07-29:
If you copy the settings in the Roxy folder's Configuration.h file...   That definitely works with the Discount 20x4 display.    I'm certainly not an expert at LCD Displays and all the different versions.   But there are 'Discount' versions of both the Graphics LCD Panel and the 20x4 LCD Panel.

fr33dimension, 2016-07-29:
Okay, I got the a full graphic LCD installed and it's doing the same thing. G29 M or G29 P1 M freezes the board. Evert hing else seams to work.

Roxy-3D, 2016-07-29:
Was that a 'Full Graphics LCD' in the picture that AdamFilip posted?   Or is that a 'Discount Full Graphics LCD' ?    I don't understand why the 20x4 display doesn't work either.   That is very strange.   At worst case, in 5 or 6 days, I'll have a Graphics LCD plugged into my printer and I'll be able to figure out what is causing trouble.  It can't be much to fix, if we just knew what was getting it sick.

fr33dimension, 2016-07-29:
Roxy, as you mentioned earlier they are very similar, the Discount Full Graphic LCD is kind of square and displays graphics while the Discount Smart LCD is rectangular. They have two different selections in config.h, but they did not make a change at all in my case. LCD can be somewhat tricky to install since they need some pins to be adjusted and they can be plugged in wrong. Mine where both working with my normal firmware, but did not with the UBL. I am not sure and willl dig deeper but here is what I got working. G29 P1 probes, G29 P2 lets me adjust. I can print G26.  What I cannot get is the G29 P1 M to work and G29 M freezes the board. I will try some more tomorrow.

Roxy-3D, 2016-07-29:
G29 P2 uses the Encoder Wheel to move the nozzle down to the bed.   That is working?   Because that is a lot of interaction with the LCD Panel to do that.   And what about a G29 P4  ???   Do the display and Encoder Wheel work?    

The reason I'm asking is the 'M' option doesn't even use the LCD Panel.  It sends a lot of text to PronterFace (or the host program) via SERIAL_PROTOCOL() and SERIAL_ECHO() calls.  But no where in the display_map() function does it touch the LCD Panel.   And I've looked at the few places where display_map() is called and the LCD Panel isn't touched as part of the setup to call it.   It just doesn't make sense that calling that function should cause problems.   And it also doesn't make sense that you don't even see the 'Topology Report:' title come out.     When we figure it out, I'm sure it will all make sense.  But right now, this doesn't make any sense.

fr33dimension, 2016-07-30:
Yes, both (reprap discount)LCD's  work fine. The Viki does not. I thought that the M has nothing to do with the LCD, actually this firmware would work without LCD, you just cannot adjust the off bed points.
I didn't try the P4 yet.
I am using repetier, which displays the same things as pronterface, but no, I am not even getting the title.

Roxy-3D, 2016-07-30:
> Yes, both (reprap discount)LCD's work fine. 

Can you tell me this:   Ignoring the Bed Leveling System, can you use the Encoder Wheel to move around on the LCD Menu and Select items?

fr33dimension, 2016-07-30:
Yes, on both reprap discount LCD's.

fr33dimension, 2016-07-30:
With G29 P4 I get the mesh point editor on the LCD so everything works, besides the M command.

Roxy-3D, 2016-07-30:
OK...    Is it accurate to say the problem with the 'M' option on G29 is not tied to a LCD Panel?   I'm trying to connect the bad behavior of the 'M' option with the LCD Controller not working correctly.   But I think I'm hearing you say the LCD Panel problems are gone if you are using the 'right' LCD Panel.  However you still have the problem with the Map option hanging the machine.

fr33dimension, 2016-07-30:
Roxy, 
I switched over to Printrun and the M command works fine, it doesn't work in Repetier. Hopefully this can be fixed. 

Bed Topography Report:
(0,9)                                                                                                                                                             (9,9)
(0,164)                                                                                                                                                         (186,164)
       .                 .                 .                 .                 .                 .                 .                 .                 .                 .

       .                 .                 .                 .                 .                 .                 .                 .                 .                 .

  0.18250     0.08250     0.01000    -0.08000    -0.09750    -0.19250    -0.19000    -0.23000          .                 .

  0.25000     0.14250     0.08250    -0.01250    -0.02000    -0.12000   [-0.13000]  -0.13750          .                 .

  0.29500     0.20000     0.13000     0.05500     0.02500    -0.06250    -0.06500    -0.08750          .                 .

  0.36500     0.27250     0.21250     0.11250     0.09250     0.01250     0.00000    -0.01750          .                 .

  0.42750     0.34250     0.30750     0.19000     0.14500     0.16000     0.06500     0.03500          .                 .

  0.50000     0.43500     0.38750     0.27000     0.23500     0.16750     0.14750     0.12750          .                 .

  0.55500     0.47500     0.44000     0.33250     0.31000     0.24750     0.23000     0.20250          .                 .

  0.64000     0.57000     0.45500     0.43000     0.41750     0.39250     0.33250     0.32250          .                 .
(0,0)                                                                                                                                                             (186,0)
(0,0)                                                                                                                                                                (9,0)

[16 more replies]

closed 2016-07-31 by fr33dimension