MarlinFirmware/Marlin #4423: SD_DETECT_PIN goes LOW but then goes HIGH

thinkyhead, 2016-07-27:
I'm using the `G3D_PANEL` with 1.1.0-RC7.

- Inserting or removing the SD card causes the LCD to blank for longer than usual
- The display always says "Card removed" and "No SD card" even when it can read the card
- `M20` works to display the contents of the SD card
- `M23` and `M24` don't do anything when sent from the console
- When printing from SD via the controller, `M23` and `M24` are enqueued, but then nothing happens.
- Tones played on the speaker sound worse than in earlier Marlin versions

I did some debugging to see what was going on. The `SD_DETECT_PIN` is acting oddly. The bit goes to LOW when the card is inserted or removed but then soon reverts back to HIGH, and wants to stay HIGH.
```
Startup:
SD_DETECT_PIN : 1
SD_DETECT_PIN : 1  IS_SD_INSERTED : 0
New lcd_sd_status=0
SD_DETECT_PIN : 1
SD_DETECT_PIN : 1
. . .
```
Insert Card:
```
SD_DETECT_PIN : 0  IS_SD_INSERTED : 1
New lcd_sd_status=1
echo:SD card ok
SD_DETECT_PIN : 1
SD_DETECT_PIN : 1  IS_SD_INSERTED : 0
New lcd_sd_status=0
SD_DETECT_PIN : 1
SD_DETECT_PIN : 1
. . .
```
```
Remove Card:
SD_DETECT_PIN : 0  IS_SD_INSERTED : 1
New lcd_sd_status=1
echo:SD init fail
SD_DETECT_PIN : 1
SD_DETECT_PIN : 1  IS_SD_INSERTED : 0
New lcd_sd_status=0
SD_DETECT_PIN : 1
SD_DETECT_PIN : 1
. . .
```
Is this a bug where some other code could be stepping on the `SD_DETECT_PIN` — 49 in my case? Have we seen this before?

thinkyhead, 2016-07-27:
**SOLVED.** If the SD Card is inserted with the tab in the LOCK position, this is the resulting behavior. I can't think of any way to work around it so that we can produce better behavior. So we need to include this as a troubleshooting tip.

thinkyhead, 2016-07-27:
After experimenting, I decided to come up with a workaround. First, whenever the pin goes from LOW (or, "inserted") back to HIGH (or, "removed") the change occurs less than one second afterward. So if a small delay is detected, the code will retry card initialization. If it succeeds then the card is considered "detected."

Blue-Marlin, 2016-07-27:
The hardware designer used the 'card write protected' instead of the 'card present' connector of the socket. The LOW time likely depends on the speed you are pushing in the card.

thinkyhead, 2016-07-27:
@Blue-Marlin Is this common?

Blue-Marlin, 2016-07-27:
I don't know.
A short test with my panels does no not show the problem. A bunch of REPRAP_DISCOUNT_SMART_CONTROLLER and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER clones from different producers.

Blue-Marlin, 2016-07-27:
May also be a damaged socket, or some conductive dirt shorting the both pins. Non't know the exact layout of the G3D-Panel.

thinkyhead, 2016-07-27:
It doesn't surprise me if it's using the wrong pin. But I wonder if the other pin might still be connected, so that I can simply change the pin number and get the same behavior. The "schematic" doesn't offer much hope here…
[image]

Grogyan, 2016-07-27:
@thinkyhead if the cause is really down to the latch on thde SD card itself, then this should be respected for its designed purpose to not enable writing to the SD card

jbrazio, 2016-07-28:
@thinkyhead have you low level formatted the SD Card ? We usually see this random SD card errors and so far 99.9% of them are solved by low level formatting the card.

thinkyhead, 2016-07-28:
@jbrazio If it was an issue that could be fixed with low-level formatting, wouldn't I see the same behavior all the time, and not just when the card is write-protected? Also, I don't recall anybody having issues where the `SD_DETECT_PIN` acted bizarrely due to the contents of the card.

Blue-Marlin, 2016-07-29:
I guess this is ether specific to `Panel_ONE` or specific to a special series of `Panel_ONE` or specific to your special display. Depends on, if this is a design error, a charge of different sockets or a soldering error.

thinkyhead, 2016-07-29:
Is the `G3D_PANEL` a variant of the `Panel_ONE`.

Blue-Marlin, 2016-07-29:
Sorry. I mend the display type you are talking about. (Corrected that above.) For some reason my fuzzy memory told me, you used to have a PANEL_ONE

thinkyhead, 2016-07-29:
I now have a RDSC I can test with (thanks @esenapaj!). I will compare to the pinouts on the G3D and see if the flaw is obvious. I'm sure you're right about the wrong pin being connected. I might revive my workaround as an option buried in `Configuration_adv.h`.

boelle, 2016-08-24:
if this is not a bug.. ie bug invalid why not close it?

@thinkyhead @jbrazio

thinkyhead, 2016-08-24:
I have a proposed fix, but it's such a niche case that I think it should be one of those "fixer-up" options "buried" in `Configuration_adv.h`. If I decide to make a PR I'll just link back to this.

closed 2016-08-24 by thinkyhead