MarlinFirmware/Marlin #4425: MBL again broken because of probe

Alex9779, 2016-07-27:
Ok this is odd, I wanted sanity checks for Z_SAFE_HOMING, #4398 did that but also introduced that I need a probe now.
I commented the PR but no answer, I had a problem with MBL with a probe described here #4170 so I had to deactivate.
Now I have to activate which bricks MBL again, but I need Z_SAFE_HOMING...
And this stuff is now in RC7...
Why do I need a probe for Z_SAFE_HOMING?

Alex9779, 2016-07-27:
I disabled the sanity check, that way it works.
I think this is because of the `Z_RAISE_PROBE_DEPLOY_STOW`and `Z_RAISE_BETWEEN_PROBINGS`.
Those interfere when manually leveling the bed....

thinkyhead, 2016-07-27:
I use `MIN_Z_HEIGHT_FOR_HOMING` to improve the behavior of MBL without a probe.

Alex9779, 2016-07-27:
Ok sorry, maybe I have to explain again...
My Z endstop is not over the bed when X and Y endstops are hit. So I have to move the head to get the Z endstop over the bed. So I need `Z_SAFE_HOMING`.
But if I now wanna use it you want me to have a probe defined, for this I activate `FIX_MOUNTED_PROBE`.
But if I do so and then wanna use MBL and try to level the bed then there are more Z moves than without. The thing is that it homes, after honing it wants me to start with the first point, it moves to the first point but Z is too far away because there was an additional Z move which was not reversed before it wants me to set the offset...

brainscan, 2016-07-27:
I struggled to get mbl working again too after it was changed, I don't understand the min z height thing as it just adds extra z moves that aren't needed and it doesn't need z raise between probing either if set up using the original instructions.

thinkyhead, 2016-07-27:
> if set up using the original instructions

I'm curious what MBL does in-between points if you don't specify any kind of raise. Does it just zip over the bed at the last height it was at? Personally, I prefer having it raise, move, then lower again, but only a small raise.

thinkyhead, 2016-07-27:
> but Z is too far away because there was an additional Z move

That sounds like a bug in itself. MBL should always lower down to the bed at every point.

Alex9779, 2016-07-27:
@thinkyhead this PR is ok for me, I never understood why the probe offsets are used even if no probe was active.
But this does not fix the real issue... It fixes it for me, because I need no probe to get my machine working, you just forced me to define one again with the sanity check...

The issue with within MBL and a probe after the changes to the probe with the new settings I mentioned above.
I have defined `MIN_Z_HEIGHT_FOR_HOMING`and but this has no influence on the MBL sequence. When running MBL you level one point, then it moves to the next and raises the bed in between...

thinkyhead, 2016-07-27:
> I never understood why the probe offsets are used even if no probe was active.

Well, we're starting with the z-safe-homing code that someone made a while ago, which was oriented around probes. As time has gone on, we've been working on removing various artificial restrictions. We are (that is to say I am) just getting around to this one.

brainscan, 2016-07-27:
No it never zipped across the bed at the last height it would always return to the height it found the z end stop, move to next point and wait. I would give myself 2mm clearance over the whole bed (by default you have 4mm search distance) and never once had a problem.

thinkyhead, 2016-07-27:
@brainscan Thanks for the explanation.

thinkyhead, 2016-07-27:
> I have defined `MIN_Z_HEIGHT_FOR_HOMING` and but this has no influence on the MBL sequence.

Here is the logic used to decide the raise between points in MBL. (The `MESH_HOME_SEARCH_Z` is actually "zero" in this context – the Z home position).
```cpp
current_position[Z_AXIS] = MESH_HOME_SEARCH_Z
  #if Z_RAISE_BETWEEN_PROBINGS > MIN_Z_HEIGHT_FOR_HOMING
    + Z_RAISE_BETWEEN_PROBINGS
  #elif MIN_Z_HEIGHT_FOR_HOMING > 0
    + MIN_Z_HEIGHT_FOR_HOMING
  #endif
;
```

Alex9779, 2016-07-27:
Ok I made two videos:
MBL without a probe activated: https://youtu.be/zjm2tBCWjIM
MBL with probe activated: https://youtu.be/IoXcMuM791g

```
#define MIN_Z_HEIGHT_FOR_HOMING 1
```

MBL settings:
```#if ENABLED(MESH_BED_LEVELING)
  #define MESH_INSET 5        // Mesh inset margin on print area
  #define MESH_NUM_X_POINTS 3  // Don't use more than 7 points per axis, implementation limited.
  #define MESH_NUM_Y_POINTS 3
  #define MESH_HOME_SEARCH_Z 3  // Z after Home, bed somewhere below but above 0.0.

  //#define MESH_G28_REST_ORIGIN // After homing all axes ('G28' or 'G28 XYZ') rest at origin [0,0,0]

  #define MANUAL_BED_LEVELING  // Add display menu option for bed leveling.

  #if ENABLED(MANUAL_BED_LEVELING)
    #define MBL_Z_STEP 0.025  // Step size while manually probing Z axis.
  #endif  // MANUAL_BED_LEVELING

#endif  // MESH_BED_LEVELING
```

Settings for the probe:
```
#define FIX_MOUNTED_PROBE
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define X_PROBE_OFFSET_FROM_EXTRUDER 52  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -54  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 3   // Z offset: -below +above  [the nozzle]
#define Z_RAISE_PROBE_DEPLOY_STOW 5 // Raise to make room for the probe to deploy / stow
#define Z_RAISE_BETWEEN_PROBINGS 5  // Raise between probing points.
```

You can see that in the second video the Z rain is bigger when starting the sequence... If I lower `Z_RAISE_BETWEEN_PROBINGS`that distance shrink but I can't go lower than 1 because of a sanity check...

thinkyhead, 2016-07-27:
> I can't go lower than 1 because of a sanity check

When there's a real probe, we might want to keep that sanity check, but I'll modify it allow zero without a real probe.

Alex9779, 2016-07-27:
> Do you need to have both a probe and MBL with zero raise?

I don't need a probe... I told this earlier...
But if someone needs a probe AND uses MBL then it would not work for him.

I will just test what you merged, set the raise value to 0 and see if MBL works then...

The PR you created is fine for me, I just have to adjust my `Z_SAGFE_HOMING` coordinates again, because at the moment they are for the probe offsets and your PR will make that for the nozzle and override the offsets for the probe I might have set because of these new lines https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Conditionals_post.h#L587-L594

I have no idea if someone else needs this, I will test if setting the raise to 0 with a probe will work with MBL.

Alex9779, 2016-07-27:
Ok tested... If I want to get MBL working then I have to set:
```
#define Z_RAISE_PROBE_DEPLOY_STOW 0 // Raise to make room for the probe to deploy / stow
#define Z_RAISE_BETWEEN_PROBINGS 0  // Raise between probing points.
```

Setting either one to a higher value renders the leveling sequence unusable...
After your PR this is now allowed...

thinkyhead, 2016-07-27:
> renders the leveling sequence unusable

How does it become unusable? Does it cause the Z axis to raise and then not lower back down again? If so, that's a bug and not by design, so I should track that down and fix it.

Are you aware of `DEBUG_LEVELING_FEATURE` which allows detailed logging of this stuff, if yu send the command `M111 S255`? If you do more testing, be sure to enable this so we can see exactly what's going on.

Alex9779, 2016-07-28:
That's what I am talking about. You can use a probe but not with a raise or a stow value.
If you do then you are not able to level the bed because the bed raises on each point too far because of those settings...

thinkyhead, 2016-07-28:
> because of those settings

Because of a bug. The settings would be fine if the Z lowered back to the correct height at each point.

Alex9779, 2016-07-28:
Yeah well call it what you want. In RC6 this worked and when those settings appeared it didn't anymore... I know it is the things that changed behind those settings.
But actually when doing MBL that raise should not be done because MBL does that on its own...

thinkyhead, 2016-07-28:
> Yeah well call it what you want.

Thank you, I will.

> when doing MBL that raise should not be done because MBL does that on its own

It raises only to the level of the Z endstop. For my machine (and my preference) I wanted a larger raise.

closed 2016-07-29 by Alex9779