MarlinFirmware/Marlin #4427: UBL and Z offset question

OrbitAVI, 2016-07-27:
What is the best way to determine the correct z offset with the mesh enabled?

I am able to build a mesh using the G29 P1 command. My issue is when I go to test print using G26 my hot end is too close to the bed so it fails. Is it a problem with my offset or an issue with the G26 layer height? I simply used the default .2 layer height.

The G26 I issued was G26 H 215 N .8

My printer: i3 kit with alum bed and a prox sensor. Z offset is set with a shim (-1.1). Nozzel is .8.

Roxy-3D, 2016-07-27:
> The G26 I issued was G26 H 215 N .8

H 215 ?     That is a layer height of 215mm  not 2.15mm        A nozzle size of .8mm is huge!   Let's hope the math is all linear.   If it isn't you may need to scale that number up or down to get the right thing to happen.

But to your question...   I tried to keep my Z-Offset close to zero.  But if your nozzle size is really .8mm probably a layer height of .6mm would be closer to 'right'.

OrbitAVI, 2016-07-27:
OK I will try out the new code tonight if I can. I run a volcano to get through larger prints faster. Typically I print in layer heights of .2 or .3 without issue and still get the advantage of less perimeter shells and faster solid layers. 

How can you keep your z-offset 0 or do you account for the probe offset in the slicer?

Roxy-3D, 2016-07-27:
Well...   What I did was set my Z_PROBE_OFFSET_FROM_EXTRUDER in the Configuration.h file so that I'm perfectly calibrated.     (And that does require a M502, M500 pair to load the EEPROM with the value.)

At that point....    The Mesh offers the correct amount of adjustment everywhere the nozzle moves to on the bed (assuming the Mesh is active).   

And in Slic3r, I just tell it to do .2mm layer heights.  It always starts the first layer at .2mm.   So, I have the correct gap automatically.

OrbitAVI, 2016-07-27:
OK that is the same z offset I am talking about. Sorry I should have said it correctly from the beginning by putting the actual parameter name. I'll try the latest code and double check my Z_PROBE_OFFSET_FROM_EXTRUDER and try the G26 again. Thanks again for your excellent support!

Roxy-3D, 2016-07-27:
Also, when you get the G26 to do it's thing and work...  We may need to add a parameter for to control 'Retract'.  With a .8mm nozzle, my guess is we will need to retract the filament more to keep it from making lines where we don't want them.   A picture will help a lot to figure out how we are going to address this.    It maybe we can just scale up or down the retraction based on the nozzle size.   But my guess it will be more complicated than that.

OrbitAVI, 2016-07-27:
I'll surely post a pic once I can get the validation file to print ok. I do use a high retract with this nozzel / bowden setup. I think its a 6MM retract that I've been running.

Roxy-3D, 2016-07-27:
OK...  And just because I think you are going to need it to get a nice, clean Mesh Validation print...

I've add a 'M' parameter to G26 to set the retraction Multiplier.   If you have problems with Ooooze on the retracted movements, you can start scaling up the amount of retraction until it behaves better.

adamfilip, 2016-07-29:
Can the Defaults for G26 be able to be assigned in the configuraton.h 
easier to do this then remember the Gcode Parameters when sending commands

Being able to change defaults would be handy for
Hotend Temp
Filament Diam
Layer Height
Bed Temp
Nozzle Diam
Ooze
Prime amount

Roxy-3D, 2016-07-29:
Yes.   This can be done.   Let me think about how to do it in the 'cleanest' way.

Roxy-3D, 2016-07-30:
> Being able to change defaults would be handy for

This is done...   I put the values at the very top of the G26 file just because we already have enough clutter in the Configuration.h file.   They are easy to find and change.

closed 2017-10-31 by thinkyhead