MarlinFirmware/Marlin #4429: UBL G29 leveling preheat

OrbitAVI, 2016-07-27:
This is just a suggestion / feature request, but it may be beneficial to be able to pass a preheat param when running a level function like P1. At least for me with an aluminum bed I know it changes a little during warm up.

Roxy-3D, 2016-07-27:
Yes, but can't you just turn on the heaters prior to doing any Mesh work?   I did put in heater code for the G26 but I'm wondering if we should make that an option that can be turned off  to shrink the code foot print.

OrbitAVI, 2016-07-27:
Yes I preheat before leveling, but forgetting to do so just means you have to start over. Not a big deal at all just putting it out there. I'll close this one.

closed 2016-07-27 by OrbitAVI

Roxy-3D, 2016-07-27:
Also...   This is one of the really cool things about using Mesh Leveling as the foundation of the new system:   Once you get the Mesh defined correctly (very accurately), it is loaded automatically when the machine power up.     If you power up your machine and tell it to print, it will pre-heat first at which point the Mesh should be accurate.

OrbitAVI, 2016-07-27:
Exactly!