MarlinFirmware/Marlin #4430: Build broken: floating constant in preprocessor expression

otvald, 2016-07-27:
In commit d6c50671d3550252ff47ce27c8c7924733ec6347, there is a sanity check added for Z_SAFE_HOMING_X_POINT.

However, this will fail if DELTA_PRINTABLE_RADIUS is a float (e.g. 75.0), and Z_SAFE_HOMING is enabled.

This is on the latest RC merge.

From the compiler:
```
In file included from sketch/MarlinConfig.h:29:0,
                 from sketch/SanityCheck.cpp:29:
Configuration.h:424: error: floating constant in preprocessor expression
   #define DELTA_PRINTABLE_RADIUS 75.0
                                  ^
sketch/Configuration.h:652:21: note: in expansion of macro 'DELTA_PRINTABLE_RADIUS'
 #define X_MIN_POS -(DELTA_PRINTABLE_RADIUS)
                     ^
sketch/Configuration.h:819:35: note: in expansion of macro 'X_MIN_POS'
   #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    // X point for Z homing when homing all axis (G28).
                                   ^
sketch/SanityCheck.cpp:366:7: note: in expansion of macro 'Z_SAFE_HOMING_X_POINT'
   #if Z_SAFE_HOMING_X_POINT < MIN_PROBE_X || Z_SAFE_HOMING_X_POINT > MAX_PROBE_X
```

The preprocessor greater than / less than will fail is comparing non-integer.

A) Is this just a failure on my side? (e.g. compiler setup) - How come nobody else get this error?
B) Is DELTA_PRINTABLE_RADIUS not expected to be a float (is i all examples)?

I get around by not setting as float, but is this feasible?

otvald, 2016-07-27:
@thinkyhead I think it was you commit, so I added your name.

Btw, thinkyhead: What is your setup ? delta or ?

Roxy-3D, 2016-07-27:
Z_PROBE_OFFSET_FROM_EXTRUDER is about the only place you should be using a decimal point.   
Please just set it as:
~~~~
#define DELTA_PRINTABLE_RADIUS 75
~~~~

Blue-Marlin, 2016-07-27:
Z_SAFE_HOMING does not make any sense on a DELTA. Switch it off.

otvald, 2016-07-27:
Ok.

May I kindly suggest that SanityCheck.cpp also check if Z_SAFE_HOMING is enabled for DELTA (I know that it may seem obvious to you guys, but didn't to me)

```
/**
 * Make sure Z_SAFE_HOMING point is reachable
 */
#if ENABLED(Z_SAFE_HOMING)

  #if ENABLED(DELTA)
    #error "Please disable Z_SAFE_HOMING in Configuration.h. Not applicable for delta printer."
  #endif

  #if Z_SAFE_HOMING_X_POINT < MIN_PROBE_X || Z_SAFE_HOMING_X_POINT > MAX_PROBE_X
    #error "The given Z_SAFE_HOMING_X_POINT can't be reached by the Z probe."
  #elif Z_SAFE_HOMING_Y_POINT < MIN_PROBE_Y || Z_SAFE_HOMING_Y_POINT > MAX_PROBE_Y
    #error "The given Z_SAFE_HOMING_Y_POINT can't be reached by the Z probe."
  #endif
#endif // Z_SAFE_HOMING
```

thinkyhead, 2016-07-28:
> May I kindly suggest that `SanityCheck.cpp` also check if `Z_SAFE_HOMING` is enabled for `DELTA`?

That's a good idea! I will fix that today.

closed 2016-07-28 by otvald