MarlinFirmware/Marlin #4432: error when compiling

jac224, 2016-07-28:
hello, a few hours ago I download the new version of firmware marlin and I'm having this problem when I compile with Arduinome get this error ....

   Arduino: 1.0.6 (Windows NT (unknown)) Board: "Arduino Mega 2560 or Mega ADK"
_In file included from /nozzle.h:27,
                  from Marlin_main.cpp: 62:
/point_t.h: In constructor 'point_t :: point_t (float, float)':
point_t.h: 49: error: type 'point_t' is not a direct basis of 'point_t'
/point_t.h: In constructor 'point_t :: point_t (float, float, float)':
point_t.h: 59: error: type 'point_t' is not a direct basis of 'point_t'
In file included from Marlin_main.cpp: 63:
/duration_t.h: In constructor 'duration_t :: duration_t ()':
duration_t.h: 36: error: type 'duration_t' is not a direct basis of 'duration_t'_

sincerely hope that is not the error. I would greatly appreciate your help. Thank you

thinkyhead, 2016-07-28:
@jac224 Did you modify your configuration before compiling? If so, post it here and I'll try compiling it, see if I can figure out why it's giving an error. (I don't see error this with my config.)

By the way, we now require Arduino 1.6.0 or higher. See if a newer version of Arduino IDE solves the problem, in case it's compiler-related.

jac224, 2016-07-28:
@thinkyhead at first i modify the configuration.h . But then i tried to compile it without doing any changes to it and gave me the same error. I have arduino 1.0.6 installed

Blue-Marlin, 2016-07-28:
> By the way, we now require Arduino 1.6.0 or higher. See if a newer version of Arduino IDE solves the problem, in case it's compiler-related.

Arduino 1.0.6 is too old. Try a newer one. 1.6.9 works fine. 1.6.10 is to new to have much experience with - but should work.

jbrazio, 2016-07-28:
> Arduino 1.0.6

How is this even compiling without barking ?
Have you void() your warranty by commenting out the following section ?

```cpp
/**
 * Due to the high number of issues related with old versions of Arduino IDE
 * we are now warning our users to update their toolkits. In a future Marlin
 * release we will stop supporting old IDE versions and will require user
 * action to proceed with compilation in such environments.
 */
#if !defined(ARDUINO) || ARDUINO < 10600
  #error "Versions of Arduino IDE prior to 1.6.0 are no longer supported, please update your toolkit."
#endif
```

thinkyhead, 2016-07-28:
@jbrazio I moved the SanityCheck so it would only be done once. So I need to move this test to the `MarlinConfig.h` file.

jbrazio, 2016-07-28:
@thinkyhead `SanityCheck.cpp` ? Aren't the header files `.h` always pre-processed first ? If we have no "code" wouldn't it make sense to keep using the header file for doing all the checks as they are all based on pre-processor conditionals.

I edited the IDE warning comment a moment ago and didn't notice I was editing a `.cpp` file ! Damn, I've been really tired lately.

thinkyhead, 2016-07-29:
@jbrazio The idea (D'oh!) was to get the SanityCheck to occur only once, and had hoped to get it to compile first. (This could be done from the .ino file, perhaps.) But I forgot that if other `.cpp` files compile first all kinds of fun results may occur.

jbrazio, 2016-07-29:
To catch stuff like the Arduino IDE version that's a wonderful idea, `Marlin.ino` is the best place to do such check. But.. if we do plan in the future to support other compilation method outside the IDE, for instance `Makefile` based, then `SanityCheck.h` should be the first thing getting included from `Marlin.h` as our potential entry point would be located on a `Marlin.cpp` file.

closed 2016-07-29 by thinkyhead