MarlinFirmware/Marlin #4433: [DELTA] observing *very* slow movement (random)

otvald, 2016-07-28:
On latest RC.

Doing G28 I observed sometimes that the travel speed (after first contact) was VERY slow when trying to hit contacts.

I have also seen that sometimes a travel (G1) can be VERY slow (guess < 0.1 mm/sec), despite setting f4000.

If I am the only one seeing this, I will assume there is a bug in my config...

thinkyhead, 2016-07-28:
@otvald Please post your configuration, just in case. Meanwhile, I'll look at the `G28` code and make sure the homing speeds are being set correctly.

otvald, 2016-07-28:
My Configuration.h : http://pastebin.com/W5Q6ests
My Configuration_adv.h : http://pastebin.com/aKbNkU5V

I have looked at the code, and I enabled debug and uncommented extra debug in 'prepare_kinematic_move_to'. Strange; The working and non-working commands seem to be initialized the same. (I was expecting corrupt feedrate_mm_m, but then the 'seconds' below should have differed).

_If you would like some extra debug just say so!_

This command (g1 x30 y40 z260 f4000) moves VERY slow:
```
SENDING:G1 X30 Y40 Z260 F4000
echo:G1 X30 Y40 Z260 F4000
mm=50.41 seconds=0.76 steps=75
echo:busy: processing
```
(the busy is because I run from Pronterface, and it polls every N seconds, and in this case the system is still moving).

This (same command) works as expected:
```
SENDING:G1 X30 Y40 Z260 F4000
echo:G1 X30 Y40 Z260 F4000
mm=50.41 seconds=0.76 steps=75
```

otvald, 2016-07-28:
Seem a problem with block->nominal_speed

Added debug in calculate_trapezoid_for_block

Just after
```
  // Calculate the size of Plateau of Nominal Rate.
  int32_t plateau_steps = block->step_event_count - accelerate_steps - decelerate_steps;
```
I added:
```
    SERIAL_ECHO_START;
     SERIAL_ECHOPGM("TOT - steps: ");
     SERIAL_ECHO(block->step_event_count);
     SERIAL_ECHOPGM("block->acceleration: ");
     SERIAL_ECHO(block->acceleration);
     SERIAL_ECHOPGM("block->acceleration_rate: ");
     SERIAL_ECHO(block->acceleration_rate);
     SERIAL_ECHOPGM("block->nominal_speed :");
     SERIAL_ECHOLN(block->nominal_speed);
```

For working  flow:
```
SENDING:G1 X30 Y40 Z260 F4000
mm=50.41 seconds=0.76 steps=75
TOT - steps: 42block->acceleration: 500.01block->acceleration_rate: 369828block->nominal_speed :66.67
TOT - steps: 42block->acceleration: 500.01block->acceleration_rate: 369828block->nominal_speed :66.67
TOT - steps: 43block->acceleration: 500.01block->acceleration_rate: 373645block->nominal_speed :66.67
TOT - steps: 42block->acceleration: 500.01block->acceleration_rate: 369828block->nominal_speed :66.67
TOT - steps: 43block->acceleration: 500.01block->acceleration_rate: 373645block->nominal_speed :66.67
(continues) to 70 steps
```

For **bad** flow:
```
SENDING:G1 X30 Y40 Z260 F4000
mm=50.41 seconds=0.76 steps=75
TOT - steps: 42block->acceleration: 500.01block->acceleration_rate: 369828block->nominal_speed :0.00
TOT - steps: 42block->acceleration: 500.01block->acceleration_rate: 369828block->nominal_speed :0.00
TOT - steps: 43block->acceleration: 500.01block->acceleration_rate: 373645block->nominal_speed :0.00
TOT - steps: 42block->acceleration: 500.01block->acceleration_rate: 369828block->nominal_speed :0.00
TOT - steps: 43block->acceleration: 500.01block->acceleration_rate: 373645block->nominal_speed :0.00
TOT - steps: 43block->acceleration: 500.01block->acceleration_rate: 370399block->nominal_speed :0.00
```

rogueqd, 2016-07-28:
I have a similar problem issuing G28 command. G28 takes minutes. The printer moves normally after Home has been found, but issuing G28 again is still very slow.
Video showing problem: https://www.youtube.com/watch?v=2WcRAywBIRI
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/389350/Configuration.h.txt)

otvald, 2016-07-28:
@thinkyhead I think I got it:

planner.cpp (void Planner::buffer_line)
```

  #else
    float delta_mm[4];
    #if ENABLED(DELTA)
      // On delta all axes (should!) have the same steps-per-mm
      // so calculate distance in steps first, then do one division
      // at the end to get millimeters
    #else
      delta_mm[X_AXIS] = dx * steps_to_mm[X_AXIS];
      delta_mm[Y_AXIS] = dy * steps_to_mm[Y_AXIS];
      delta_mm[Z_AXIS] = dz * steps_to_mm[Z_AXIS];
    #endif
  #endif
  delta_mm[E_AXIS] = 0.01 * (de * steps_to_mm[E_AXIS]) * volumetric_multiplier[extruder] * extruder_multiplier[extruder];
```

On a DELTA, this means that delta_mm[0] , delta_mm[1], delta_mm[2] are **undefined**. Only delta_mm[E_AXIS] is initialized...

Then further down

```
  // Calculate and limit speed in mm/sec for each axis
  float current_speed[NUM_AXIS];
  float speed_factor = 1.0; //factor <=1 do decrease speed
  LOOP_XYZE(i) {
    current_speed[i] = delta_mm[i] * inverse_second;
    float cs = fabs(current_speed[i]), mf = max_feedrate_mm_s[i];    
    if (cs > mf)  speed_factor = min(speed_factor, mf / cs);
  }
```

Here we iterate over delta_mm[0..2], and if these numers are high (in my case 2E17), then speed_factor will get VERY small! (near zero).

**Solution**: Should delta_mm[X Y Z _AXIS] be copied from E_AXIS? Or should the LOOP only address E_AXIS in the evaluation of speed_factor for DELTA printer? E.g.

```
  // Calculate and limit speed in mm/sec for each axis
  float current_speed[NUM_AXIS];
  float speed_factor = 1.0; //factor <=1 do decrease speed
#if ENABLED(DELTA)
  LOOP_E(i) {
#else
  LOOP_XYZE(i) {
#endif
    current_speed[i] = delta_mm[i] * inverse_second;
    float cs = fabs(current_speed[i]), mf = max_feedrate_mm_s[i];    
    if (cs > mf)  speed_factor = min(speed_factor, mf / cs);
  }
```
And #define LOOP_E(VAR) for (uint8_t VAR=E_AXIS; VAR<=E_AXIS; VAR++).

thinkyhead, 2016-07-28:
@otvald Good eye! Actually, I attempted to reduce calculation required for `block->millimeters` by two multiplications and missed that `delta_mm` did in fact need setting for the benefit of the `current_speed` loop you point out.

closed 2016-07-28 by thinkyhead