MarlinFirmware/Marlin #4435: customizing Z (increment and decrements)

pankajbadatia, 2016-07-28:
hello everyone .
I am trying to customize the RAMPS 1.4 for my work. I am trying to read values of pressure sensor.
 
When it reaches a higher threshold the z starts increment 
with lower threshold z stops.

I tried but the z increments  and marlin resets

Following is the output

start
echo:Marlin 1.1.0-RC7
echo: Last Updated: 2016-07-26 12:00 | Author: (none, default config)
Compiled: Jul 28 2016
echo: Free Memory: 4707  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00 C100.00 L20
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:enqueueing "G1 X100 Y200 F500"
G1 X100 Y200 F500
20
Aspirate was detected
Qty was for head200
G1 Z1 F12
G1 Z2 F12
G1 Z3 F12
G1 Z4 F12
start
echo:Marlin 1.1.0-RC7
echo: Last Updated: 2016-07-26 12:00 | Author: (none, default config)
Compiled: Jul 28 2016
echo: Free Memory: 4707  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00 C100.00 L20
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0

pankajbadatia, 2016-07-28:
Can anyone help me for increment z axis in coding. I am stuck because marlin restarts

brainscan, 2016-07-28:
You might need to try and explain in more detail what it is you're trying to achieve and also what you have tried to get these results.
Are you trying to use force sensing resistors for bed levelling?

pankajbadatia, 2016-07-28:
no its completely for different use. I am using it for personal use. 
in this i have used spare analog pin of RAMPS 1.4 and i am fetching values from this  analog pin. 

Created a storage space for higher threshold and lower threshold.

I just want to increment it  within my program
enqueue_and_echo_command()

works fine
 
but as you can see

do{
int value=pressuresenorvalue();
Zmotorincrement()
}while(value<pressuresensor_highthreshold)

void Zmotorincrement(){ 
  b++;
  String emotorinc = "G1 Z" ;
  String gcode = emotorinc + b + " F12";
   int gcode_length = gcode.length() + 1;
char gcode_array[gcode_length];
gcode.toCharArray(gcode_array, gcode_length);
 enqueue_and_echo_command(gcode_array);
MYSERIAL.println(gcode);
emotorinc = "";
}

Blue-Marlin, 2016-07-28:
Don't use `analogeRead()` in Marlin.
Don't use the String class in Marlin.
Don't enqueue_and_echo_command() simple moves. We have a wide variety of much simpler to use moves. (feedrate_mm_m = 12*60; line_to_z(float zPosition);)
If you just want a stepper driven pressure-controler don't use Marlin - it's much to complex for that task.

pankajbadatia, 2016-08-01:
Thanks . But are there any function's for Xaxis and Y axis  positioning  instead of using enqueue_and_echo_command() ; 

i want to move both at a time to my destination

Blue-Marlin, 2016-08-01:
If you have already set up `destination[]`, probably `prepare_move_to_destination()` is the most convenient and universal.
But please look up the definitions for 'prepare_move', 'line_to', `do_blocking_move` (all in MarlinMain.cpp)

Roxy-3D, 2016-08-01:
>  But are there any function's for Xaxis and Y axis positioning instead of using enqueue_and_echo_command() ;

Grab the devel-ubl branch and check out the G26 command.   That command is aimed at Mesh Validation.   If your Mesh definition is wrong, it will show up in the print.  It moves all around the bed as it prints.

closed 2017-11-01 by thinkyhead