MarlinFirmware/Marlin #4436: Compile Error

jpacman, 2016-07-28:
I cant even Compile this into a hex this is my log.

> Arduino: 1.0.5-r2 (Windows NT (unknown)), Board: "[usbtinyisp]Printrboard"
C:\Users\Jorda\Documents\PRINTRBOARD\FILES\arduino-1.0.5-r2-at90usb1286-master\arduino-1.0.5-r2-at90usb1286-master\hardware\tools\avr\bin\avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=at90usb1286 -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -IC:\Users\Jorda\Documents\PRINTRBOARD\FILES\arduino-1.0.5-r2-at90usb1286-master\arduino-1.0.5-r2-at90usb1286-master\hardware\at90usb1286\cores\at90usb1286 -IC:\Users\Jorda\Documents\PRINTRBOARD\FILES\arduino-1.0.5-r2-at90usb1286-master\arduino-1.0.5-r2-at90usb1286-master\libraries\Wire -IC:\Users\Jorda\Documents\PRINTRBOARD\FILES\arduino-1.0.5-r2-at90usb1286-master\arduino-1.0.5-r2-at90usb1286-master\libraries\LiquidCrystal -IC:\Users\Jorda\Documents\PRINTRBOARD\FILES\arduino-1.0.5-r2-at90usb1286-master\arduino-1.0.5-r2-at90usb1286-master\libraries\SPI C:\Users\Jorda\AppData\Local\Temp\build3245090032925994841.tmp\blinkm.cpp -o C:\Users\Jorda\AppData\Local\Temp\build3245090032925994841.tmp\blinkm.cpp.o 

C:\Users\Jorda\Documents\PRINTRBOARD\FILES\arduino-1.0.5-r2-at90usb1286-master\arduino-1.0.5-r2-at90usb1286-master\hardware\tools\avr\bin\avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=at90usb1286 -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -IC:\Users\Jorda\Documents\PRINTRBOARD\FILES\arduino-1.0.5-r2-at90usb1286-master\arduino-1.0.5-r2-at90usb1286-master\hardware\at90usb1286\cores\at90usb1286 -IC:\Users\Jorda\Documents\PRINTRBOARD\FILES\arduino-1.0.5-r2-at90usb1286-master\arduino-1.0.5-r2-at90usb1286-master\libraries\Wire -IC:\Users\Jorda\Documents\PRINTRBOARD\FILES\arduino-1.0.5-r2-at90usb1286-master\arduino-1.0.5-r2-at90usb1286-master\libraries\LiquidCrystal -IC:\Users\Jorda\Documents\PRINTRBOARD\FILES\arduino-1.0.5-r2-at90usb1286-master\arduino-1.0.5-r2-at90usb1286-master\libraries\SPI C:\Users\Jorda\AppData\Local\Temp\build3245090032925994841.tmp\cardreader.cpp -o C:\Users\Jorda\AppData\Local\Temp\build3245090032925994841.tmp\cardreader.cpp.o 

In file included from /stepper.h:47,
                 from cardreader.cpp:26:
/speed_lookuptable.h:30: warning: only initialized variables can be placed into program memory area
/speed_lookuptable.h:65: warning: only initialized variables can be placed into program memory area
In file included from /temperature.h:32,
                 from cardreader.cpp:27:
/thermistortables.h:140: warning: only initialized variables can be placed into program memory area
/thermistortables.h:201: warning: only initialized variables can be placed into program memory area
C:\Users\Jorda\Documents\PRINTRBOARD\FILES\arduino-1.0.5-r2-at90usb1286-master\arduino-1.0.5-r2-at90usb1286-master\hardware\tools\avr\bin\avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=at90usb1286 -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -IC:\Users\Jorda\Documents\PRINTRBOARD\FILES\arduino-1.0.5-r2-at90usb1286-master\arduino-1.0.5-r2-at90usb1286-master\hardware\at90usb1286\cores\at90usb1286 -IC:\Users\Jorda\Documents\PRINTRBOARD\FILES\arduino-1.0.5-r2-at90usb1286-master\arduino-1.0.5-r2-at90usb1286-master\libraries\Wire -IC:\Users\Jorda\Documents\PRINTRBOARD\FILES\arduino-1.0.5-r2-at90usb1286-master\arduino-1.0.5-r2-at90usb1286-master\libraries\LiquidCrystal -IC:\Users\Jorda\Documents\PRINTRBOARD\FILES\arduino-1.0.5-r2-at90usb1286-master\arduino-1.0.5-r2-at90usb1286-master\libraries\SPI C:\Users\Jorda\AppData\Local\Temp\build3245090032925994841.tmp\configuration_store.cpp -o C:\Users\Jorda\AppData\Local\Temp\build3245090032925994841.tmp\configuration_store.cpp.o 

In file included from /temperature.h:32,
                 from configuration_store.cpp:132:
/thermistortables.h:140: warning: only initialized variables can be placed into program memory area
/thermistortables.h:201: warning: only initialized variables can be placed into program memory area
In file included from /stepper.h:47,
                 from /temperature.h:35,
                 from configuration_store.cpp:132:
/speed_lookuptable.h:30: warning: only initialized variables can be placed into program memory area
/speed_lookuptable.h:65: warning: only initialized variables can be placed into program memory area
configuration_store.cpp: In function 'void _EEPROM_writeData(int&, uint8_t*, uint8_t)':
configuration_store.cpp:150: warning: only initialized variables can be placed into program memory area
configuration_store.cpp: In function 'void Config_ResetDefault()':
configuration_store.cpp:689: warning: only initialized variables can be placed into program memory area
configuration_store.cpp: In function 'void Config_PrintSettings(bool)':
configuration_store.cpp:705: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:708: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:709: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:710: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:711: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:729: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:732: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:733: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:734: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:735: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:740: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:743: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:744: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:745: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:746: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:750: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:753: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:754: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:755: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:760: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:763: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:764: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:765: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:766: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:767: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:768: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:773: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:776: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:777: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:778: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:855: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:878: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:879: warning: only initialized variables can be placed into program memory area
configuration_store.cpp:880: warning: only initialized variables can be placed into program memory area
[... cut ...]
Marlin_main.cpp:8105: warning: only initialized variables can be placed into program memory area
Marlin_main.cpp: In function 'void manage_inactivity(bool)':
Marlin_main.cpp:8518: warning: only initialized variables can be placed into program memory area
Marlin_main.cpp: In function 'void kill(const char*)':
Marlin_main.cpp:8674: warning: only initialized variables can be placed into program memory area
Marlin_main.cpp: In function 'void stop()':
Marlin_main.cpp:8779: warning: only initialized variables can be placed into program memory area
>

thinkyhead, 2016-07-28:
What version of Arduino IDE are you using?

jpacman, 2016-07-28:
1.0.5-r2

thinkyhead, 2016-07-29:
@jpacman Try again with `RCBugFix`. It will tell you to upgrade to Arduino 1.6.0 or higher. 1.1.0-RC7 should have told you this, but there was a glitch.

closed 2017-10-31 by thinkyhead