MarlinFirmware/Marlin #4437: Compile Error - Rambo

adamfilip, 2016-07-28:
Trying out the UBL version on my Makerfarm i3V with Rambo
. so far Havent been able to get it to compile

Error below. also attached configuration.h and configuration_adv.h

`Arduino: 1.6.9 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\G29_Unified_Bed_Leveling.cpp.o: In function `MarlinSerial::write(unsigned char)':

sketch/MarlinSerial.h:125: undefined reference to `lcd_implementation_clear()'

sketch\G29_Unified_Bed_Leveling.cpp.o: In function `fine_tune_mesh(float, float, float, bool)':

sketch/G29_Unified_Bed_Leveling.cpp:1626: undefined reference to `lcd_implementation_clear()'

sketch\G29_Unified_Bed_Leveling.cpp.o: In function `gcode_G29()':

sketch/G29_Unified_Bed_Leveling.cpp:446: undefined reference to `lcd_implementation_clear()'

sketch/G29_Unified_Bed_Leveling.cpp:464: undefined reference to `lcd_implementation_clear()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/388742/Configuration.zip)
`

adamfilip, 2016-07-28:
Tried it with 1.6.10 and got this

Arduino: 1.6.10 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\Jen\AppData\Local\Temp\cc6Z1hN1.ltrans1.ltrans.o: In function `gcode_G29() [clone .part.7] [clone .lto_priv.204]':

cc6Z1hN1.ltrans1.o:(.text+0x1676): undefined reference to `lcd_implementation_clear()'

cc6Z1hN1.ltrans1.o:(.text+0x177c): undefined reference to `lcd_implementation_clear()'

cc6Z1hN1.ltrans1.o:(.text+0x19ca): undefined reference to `lcd_implementation_clear()'

cc6Z1hN1.ltrans1.o:(.text+0x1aa8): undefined reference to `lcd_implementation_clear()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

Roxy-3D, 2016-07-28:
@adamfilip   Long term this isn't true.  But right now you need to have an LCD Panel with an Encoder Wheel.     If you don't have one, you can get one on eBay for $10.00 including shipping:

http://www.ebay.com/itm/251242416836?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

adamfilip, 2016-07-28:
My Machine has a LCD with encoder

adamfilip, 2016-07-28:
[image]

adamfilip, 2016-07-28:
if its helpful. here is verbose output during compile
[VERBOSE.txt](https://github.com/MarlinFirmware/Marlin/files/389173/VERBOSE.txt)

Blue-Marlin, 2016-07-28:
dogm_lcd_implementation.h
```diff
@@ -288,11 +288,11 @@ static void lcd_implementation_init() {
     show_bootscreen = false;
 
   #endif
 }
 
-static void lcd_implementation_clear() { } // Automatically cleared by Picture Loop
+void lcd_implementation_clear() { } // Automatically cleared by Picture Loop
 
 FORCE_INLINE void _draw_centered_temp(int temp, int x, int y) {
   int degsize = 6 * (temp >= 100 ? 3 : temp >= 10 ? 2 : 1); // number's pixel width
   u8g.setPrintPos(x - (18 - degsize) / 2, y); // move left if shorter
   lcd_print(itostr3(temp));
```

Roxy-3D, 2016-07-28:
Thanks @Blue-Marlin !    That makes sense.   Nothing ever called lcd_implementation_clear() outside of lcd code before.   So, now that G29 and G26 do call it, it can't be declared as static.

~~~~
-static void lcd_implementation_clear() { } // Automatically cleared by Picture Loop
+void lcd_implementation_clear() { } // Automatically cleared by Picture Loop
~~~~

@adamfilip  I've never compiled the UBL code against the full graphics controller because I don't have one to test.    But I'll try to be as responsive as possible to any issues you run into.

Blue-Marlin, 2016-07-28:
I suggest to activate UBL in all configs, in this baranch, for the time you are working on it, to get some sanity checks with every commit/PR.

adamfilip, 2016-07-29:
Ok I made that change to dogm_lcd_implementation.h and it was able to compile and update the Rambo :)

Next small problem. how do I set the Z Max endstop to be the Z min Probe Endstop
Rambo doesnt have a Z min probe Endstop PIN setting.

thinkyhead, 2016-07-29:
> activate UBL in all configs, in this branch

And/or add specific UBL tests to the `.travis.yml`.

adamfilip, 2016-07-29:
Got it Running.. When I first tried to do a G29 P1 M it asked me to Turn on EEPROM but i have this enabled. so I had to do a M502, M500, M501 and then it started doing the Mesh Probing. it also asked this when i tried the menu selection.

BTW So far its Amazing! Thank you so much for all the time and effort you have put into this!
very excited.. need to keep playing

adamfilip, 2016-07-29:
First Bed Map created. WooHoo! :)
[image]

adamfilip, 2016-07-29:
BTW displays fine in Simplify3D Controller

Roxy-3D, 2016-07-29:
> And/or add specific UBL tests to the .travis.yml.

I need help and guidance to do that.    Let's handle that in the Google Hangout Box.

> First Bed Map created. WooHoo! :)

You don't have to,  but you should fill in all the invalidated cells.  You can either do a G29 P2 R or a G29 P3 R?  C#   (4 or 5 times).    

But once you have the Mesh fully populated, you want to shift gears and do a G26 to see how well it is defined.   Once you have the image on the print bed, you can shift back to the G29 side of the world and do a G29 P4 R to edit all of the 'questionable' Mesh locations.

Please post a picture of your G26 once you have gone through a couple of iterations of refinement.  Its amazing how you get perfect adhesion across the entire bed.

Roxy-3D, 2016-07-29:
> Got it Running.. 

Was the one Blue-Marlin change sufficient to get it to work?  Does the Encoder Wheel work for the editing?

> BTW So far its Amazing! Thank you so much for all the time and effort you have put into this!
very excited.. need to keep playing

You are very welcome.   It was a fun piece of code to write.   Wait until you see how a big print doesn't have any problems getting good adhesion!   If you are going to do Auto Bed Leveling, a high resolution Mesh is the Cat's Meow.

thinkyhead, 2016-07-29:
> I need help and guidance to do that.

Have you looked at the `.travis.yml` file lately? It should be obvious how to do anything you'd want.

adamfilip, 2016-07-29:
Question. Why didnt it capture the Left and bottom row?

adamfilip, 2016-07-29:
> Was the one Blue-Marlin change sufficient to get it to work? Does the Encoder Wheel work for the editing?

So far I dont see how to edit from the menu.. only option is too Prepare > Level Bed

Roxy-3D, 2016-07-29:
> Have you looked at the .travis.yml file lately? It should be obvious how to do anything you'd want.

I have never looked at it.   If it is easy, I'll just dig in.   I assumed it was difficult.

> Question. Why didnt it capture the Left and bottom row?

Because your probe can't reach those bed locations.   Those have to be filled in manually.  Either by manually probing (G29 P2) or (G29 P3).   Take your pick.   And those values don't have to be super accurate.   Because once you have the G26 Mesh Validation Print to guide you,    You can do a G29 P4 R to get everything accurate.

Roxy-3D, 2016-07-29:
The LCD Panel Menu does not have any options (yet).   Everything is done with PronterFace commands.

adamfilip, 2016-07-29:
Can the best only be comprised of areas that the probe can reach so no manual entry is required?

Roxy-3D, 2016-07-29:
Yes, but you will give up part of your Print Envelope if you do that.    You can do that, and at any time in the future, go back and do the manual entry of those places.   Or for that matter, get the G26 print, and continue with the editing of those areas to get them perfect.

adamfilip, 2016-07-29:
I started doing a G29 P2 but since my Probe is 17mm lower then my nozzle. I have to turn the encoder (it works)  alot to get it nozzle to touch.. for the manual. since it should know the offset of the nozzle to the prob can it bring the nozzle down 3mm from the surface instead.

adamfilip, 2016-07-29:
When starting manual probing it starts 5mm above probe height. since my hozzle is -16.62mm below probing height I need to turn the encoder in .025 increments down -21mm before the nozzle touches and for the 2nd manual point it raises back up +5 above probe.. so each time i need to rotate it down -21mm which takes forever LOL..  I gave up once it went to the 2nd one. I dont mind doing it once for the first one.. but even that should be +5 of the offset between the nozzle and probe. and then only +3 from the last point measured to be safe.

Roxy-3D, 2016-07-29:
> I started doing a G29 P2 but since my Probe is 17mm lower then my nozzle. I have to turn the encoder (it works) alot to get it nozzle to touch.. 

and

> When starting manual probing it starts 5mm above probe height. since my hozzle is -16.62mm below probing height I need to turn the encoder in .025 increments down -21mm before the nozzle touches and [...]

Has anybody noticed RepRap'ers don't like to read directions?

      P2    Phase 2   Probe areas of the Mesh that can not be automatically handled.  Phase 2 respects an H
      		      parameter to control the height between Mesh points.  The default height for movement
		      between Mesh points is 5mm.  A smaller number can be used to make this part of the 
		      calibration less time consuming.  You will be running the nozzle down until it just barely 
		      touches the glass.  You should have the nozzle clean with no plastic obstructing your view.  
		      Use caution and move slowly.  It is possible to damage your printer if you are careless. 
		      Note that this command will use the configuration #define SIZE_OF_LITTLE_RAISE if the 
		      nozzle is moving a distance of less than BIG_RAISE_NOT_NEEDED. 

		      The H parameter can be set negative if your Mesh dips in a large area.  You can press
		      and hold the LCD Panel's encoder wheel to terminate the current Phase 2 command.  You
		      can then re-issue the G29 P 2 command with an H parameter that is more suitable for the
		      area you are manually probing.   Note that the command tries to start you in a corner
		      of the bed where movement will be predictable.  You can force the location to be used in
		      the distance calculations by using the X and Y parameters.  You may find it is helpful to
		      print out a Mesh Map (G29 M ) to understand where the mesh is invalidated and where
		      the nozzle will need to move in order to complete the command.    The C parameter is 
		      available on the Phase 2 command also and indicates the search for points to measure should
		      be done based on the current location of the nozzle.

		      A B parameter is also available for this command and described up above.  It places the
		      manual probe subsystem into Business Card mode where the thickness of a business care is
		      measured and then used to accurately set the nozzle height in all manual probing for the 
		      duration of the command.  (S for Shim mode would be a better parameter name, but S is needed 
		      for Save or Store of the Mesh to EEPROM)  A Business card can be used, but you will have
		      better results if you use a flexible Shim that does not compress very much.  That makes it 
		      easier for you to get the nozzle to press with similar amounts of force against the shim so you 
		      can get accurate measurements.  As you are starting to touch the nozzle against the shim try
		      to get it to grasp the shim with the same force as when you measured the thickness of the
		      shim at the start of the command.

		      Phase 2 allows the M (Map) parameter to be specified.  This helps the user see the progression
		      of the Mesh being built.

      P3    Phase 3   Fill the unpopulated regions of the Mesh with a fixed value.  The C parameter is used to
      		      specify the Constant value to fill all invalid areas of the Mesh.  If no C parameter is 
		      specified, a value of 0.0 is assumed.  The R parameter can be given to specify the number 
		      of points to set.  If the R parameter is specified the current nozzle position is used to 
		      find the closest points to alter unless the X and Y parameter are used to specify the fill 
		      location

adamfilip, 2016-07-29:
> Has anybody noticed RepRap'ers don't like to read directions?

Really don't appreciate that comment.

I have read and re-read the instructions. Sorry if i dont know all the little details instantly.
Since you have designed this system you will feel that what needs to be done is obvious but it isnt. 

You have written alot about the parameters for the two new commands. 
but it not always clear and obvious.

Roxy-3D, 2016-07-29:
Adam...   It was a joke....   It is easy to miss small things with all of the complexity and details.   I fully realize that.

adamfilip, 2016-07-29:
I would appreciate if a simple Summary of how the system works (for the end user) was added.
it would help us RepRap'ers understand the process

I wrote up a quick version that isnt complete
-----------------------------------------------------

Once you system is configured correctly
How the UBL System works

Note: Each command may need different parameters to suit your specific system/requirements the below commands are just defaults. Please refer to the Parameter listing for more information.

Stage 1 – A G29 P1 M is issued. The System then begins to probe the bed automatically. Any spots that cannot be reached by the probe will need to be manually measured in future steps.

Stage 2 – A G29 P2 Command is issued. The system proceeds to Points where the prob was not able take a reading. From this point you need to turn the Rotary dial and lower the nozzle until it touched the bed surface. Once at the bed surface press in the rotary dial and the system will move to the next point. Until all points have been captured.

Stage 3 – A G26 Command is issued. The Hotend and Bed will heat up. Then the system will extrude filament in a grid pattern over the surface of the bed to confirm the mesh is accurate.

Stage 4 – Adjust mesh where required, re-run G26 to revalidate

Stage 5 – Save Mesh

Roxy-3D, 2016-07-29:
Yes, a summary or over view would be helpful.    

> Stage 2 – A G29 P2 Command is issued. The system proceeds to Points where the prob was not able take a reading. From this point you need to turn the Rotary dial and lower the nozzle until it touched [...]

My perspective on this is:

Stage 2 - If you have declared any X or Y PROBE_OFFSET_FROM EXTRUDER, you will have uninitialized areas of the Mesh.  You are not required to populate these values, but you will lose part of your Print Envelope (at the bed level) if you do not populate them.   You have 2 ways to populate them.  You can do a manual probing using the nozzle and Encoder Wheel at each point.  (G29 P2 R)    You can also fill the areas with a fixed constant.  (G29 P3 C?.??? R3 for the three uninitialized points closest to the nozzle.)   You can also use a combination of tactics.  You can manually probe a few locations that are spaced apart from each, and then quickly fill in the areas between the manually probed points with reasonable values for that area of the bed.   (I'm working on an easy way to extend the known values outwards which may become the most popular way to complete the Mesh.)

> Stage 5 – Save Mesh

I think I would say:

Stage 5 - Save your Final Mesh in a known storage location.   However, you should be saving your Mesh periodically during the process also.   In most cases, especially when you are just starting out with your first piece of glass, there are plenty of storage locations.   If something goes wrong and you need to reset the machine, or if you miscalculate on what the results of a Mesh building command will do, you want to be able to back up to where things were correct and not lose all of your forward progress.   At the end when your Mesh is well defined and accurate, save it in a location you will be able to find it again.

adamfilip, 2016-07-29:
> Save your Final Mesh in a known storage location.

How is this accomplished? I assumed just a M500

Roxy-3D, 2016-07-29:
M500 will (for now) save the Mesh in the active location.  But only if an Active location has been specified.  This was done just because people are used to saving their settings with M500.    

G29 S3 will save the current Mesh at Slot #3 and until you load or store to a different slot, Slot #3 will be active.  This stays active across reboots and power cycles.      When the machine powers up, it will see that you previously saved a Mesh at Slot 3 and automatically reload it.  It will also come up in the same state with regard to UBL being active or disabled.

adamfilip, 2016-07-31:
Can you explain the BIG_RAISE_NOT_NEEDED

also I realize that I can just fill the unmeasured points with a default value of 0
but if i am doing it manually. can I adjust something so that I dont have to travel Z 17mm for each location.

adamfilip, 2016-07-31:
I reran G29 P1 M after homing
But now all the values are in the -17.000mm range 
intitially when I did the same command the numbers coming back were around 0

adamfilip, 2016-08-02:
@Roxy-3D Any thoughts on why the mesh would now show as starting to much higher then before?

Roxy-3D, 2016-08-02:
> Can you explain the BIG_RAISE_NOT_NEEDED
> also I realize that I can just fill the unmeasured points with a default value of 0 but if i am doing it manually. can I adjust something so that I dont have to travel Z 17mm for each location.

For some reason, I didn't see these messages.  I must be sorting things wrong.  

The thinking behind BIG_RAISE_NOT_NEEDED and SIZE_OF_LITTLE_RAISE has to do with the search order.   The UBL system doesn't do Zig Zag.  It tries to go to points close to the starting or specified starting point.   That lets you move the nozzle to the center of a 'problem area' of your Mesh and keep repeating the specified operation until you have dealt with the problem.    If you are only moving a little bit to the next point, a BIG_RAISE_ISNT_NEEDED.   If the next point being probed is close enough, we don't have to worry about damaging the nozzle or the extruder.   If the next point is close enough, let's not do a big RAISE_EXTRUDER_BETWEEN_PROBES.    The question is...  what SIZE_OF_a_LITTLE_RAISE is safe to not do a big raise?   We don't want to damage the printer.  But we also don't want the user to have to dial down a million clicks to  deal with the point.     (And you know how painful that is....  You gave up after one point!!!)    There has to be a more elegant answer than what I have coded.  But that is what I was trying to accomplish.  I know it is painful to manually probe points!!!     If we can come up with a better answer...  Its only software, we can recode it!   :)

> I reran G29 P1 M after homing
> But now all the values are in the -17.000mm range intitially when I did the same command the numbers coming back were around 0

Once again, my apologies.   I didn't see this message until now.  I'm doing something wrong on searching for new messages.

I don't know the answer.   I have never seen that.   My first guess is the EEPROM has a goofy value for Z_Offset.   My second thought is I'm really tired of all these Z Offset problems where nobody has a clue what is going wrong.   It would be helpful to post the output of G29 W  before you do anything to change your settings.  I need to figure out a way to force the System into a clean state without losing any work that has been done to create valid Mesh data.   

This thought might not be acceptable to you.  But here is one idea:   What ever your Mesh size is declared as, consider bumping it up by one.   If you were doing 10x10, make it 11x11.   Start over.   You now know how to generate an accurate Mesh.  It really isn't that much work, especially if you can brute force fill the  unreachable areas.    And then use G26 to fine tune the Mesh.   (I don't remember hearing that you ever did a G26????   That is important because that lets you do a G29 P4 R)

I think my answer is...  Let's start over.  Probably what you are seeing is a bug and it has to be fixed.   But lets just get your system setup with a really solid and accurate Mesh defined.   I'll find the root cause of your problem soon enough.   Does that work for you?

adamfilip, 2016-08-02:
Ok Thats fine. I will try and change the mesh size and see what happens.

Roxy-3D, 2016-08-02:
Please try the M502 and M500 pair first.

adamfilip, 2016-08-02:
Will do.

adamfilip, 2016-08-03:
I turned off and on the machine.. ran a M502 and M500 and send a G29 P1 M and its back to probing in the 0 Range. no more 17mm offset.

I did a G26 B80 H245 P2
but the nozzle was too close to the bed and no extrusion happened.

So I did lowered my nozzle to the bed (no gap) and did a P29 Z and it make an adjustment but still no extrusion with a  G26 

I tried a G29 Z17.85 which should have increased the gap by .1mm it raised the gantry and did a probing but then said my Probe offset was 17.75 and not the 17.85 as a sent.

If a Number is sent G29 Z17.85 shouldnt it just accept this number.. does it need to do a raise and probing again?

There are some weird movements that ive witnessed.. once after ai moves X 10 over and ran a G29 Z. it did the probing then slammed the gantry left, ignoring the endstop for a second.. makign it skip steps.

Here is a G29 W

SENT: G29 W
RECEIVED: Unified Bed Leveling System Active.
RECEIVED:    Mesh: FFFF Loaded.  
RECEIVED: 
RECEIVED: G29_Correction_Fade_Height : 10.00
RECEIVED: 
RECEIVED: Free EEPROM space starts at: 0x0120
RECEIVED: end of EEPROM              : 0FFF
RECEIVED: z_value[][] size: 278  144
RECEIVED: 
RECEIVED: EEPROM free for UBL: 0x0EDF
RECEIVED: EEPROM can hold 0x001A meshes. 
RECEIVED: sizeof(stat)     :0043
RECEIVED: 
RECEIVED: MESH_NUM_X_POINTS  6
RECEIVED: MESH_NUM_Y_POINTS  6
RECEIVED: MESH_MIN_X         0
RECEIVED: MESH_MIN_Y         0
RECEIVED: MESH_MAX_X         200
RECEIVED: MESH_MAX_Y         240
RECEIVED: MESH_X_DIST        40.00
RECEIVED: MESH_Y_DIST        48.00

adamfilip, 2016-08-03:
After I redid a G29 Z with a .2mm gap to the glass. the G26 I did still didnt extrude anything.. it was too close still.

Roxy-3D, 2016-08-03:
Oooops!  G29 Z has a bug and is non-functional.   There are a number of work arounds.   The easiest is to download and compile the new code.    I think I have the code fixed.   Please note that I changed the documented behaviour of G29 Z.   It is much simpler than before (and hopefully more bug free as a result).   G29 Z #    will just blindly set the Z-Offset  for the UBL system at what ever number you specify.  If you need to raise or lower the nozzle a little bit, this should do it.

G29 Z   (with no number)  will probe the specified location and set the Z-Offset based on what the probe found.     

If you have  MANUAL_BED_LEVELING enabled in your configuration.h you should also be able to set this number from the LCD Panel.    Alternatively, you can also do a G29 P5 C #   which is equivalent to just setting the Z-Offset.  Instead of setting the Z-Offset, it actually edits each and every Mesh point and raises (or lowers) it by the C constant number.   But if you use the P5, it will also Mean adjust your Mesh first. so you may want to do it with no constant first.   And then decide how much you need to raise or lower it.

What I'm wondering about is the G26.    Did the L option not work to raise the nozzle for you so it could print?    It would seem something like G26 C P L .5  would like you get some results?   Can you try that (and maybe a few different L numbers) and see if you can get G26 to print before you reflash your firmware?

adamfilip, 2016-08-04:
L .5 wasnt enough
I had to do a 
G26 B80 F2.85 H245 K C P3 L 1.5
to get it to work.

adamfilip, 2016-08-04:
Updated firmware to current.
G29 Z still acting weird.
my offset is around -17

I lowered the nozzle to touch the bed. then i ran a G29 Z
it measured the probe and said my Z offset was 0.62
which doesnt make sense.

I then manually set my Z offset with a G29 Z -17

Roxy-3D, 2016-08-04:
The G29 Z code is totally untested.   Don't believe what it says   In fact, I don't even use G29 Z.   I didn't know it had a bug in it until you made that post saying it wasn't working.   If you get your Z_PROBE_OFFSET_FROM_EXTRUDER number right, you can keep the Z-Offset at 0.0

adamfilip, 2016-08-04:
Right now I just want to be able to accurately figure out the right offset from the nozzle to probe. Can't do that physically with caliper

adamfilip, 2016-08-04:
How I feel g29 z should work is that once it's runot
 It does a probing of the bed once that is done you then rotate the dial to lower the nozzle manually down until it's .1mm above bed with a paper gap. Once done press in dial to lock in offset

Roxy-3D, 2016-08-04:
> Right now I just want to be able to accurately figure out the right offset from the nozzle to probe. Can't do that physically with caliper.

Yes.  I go through the same thing.  I get the nozzle position close, but I make sure it is 1mm or 1.5mm too high so nothing gets damaged.   I then edit Configuration.h to set the Z_PROBE_OFFSET_FROM_EXTRUDER to be closer to the bed, but I make the changes slowly so nothing gets damaged.   It usually takes me 3 or 4 iterations to be exactly where I want to be.

> It does a probing of the bed once that is done you then rotate the dial to lower the nozzle manually down until it's .1mm above bed with a paper gap. Once done press in dial to lock in offset.   

OK!  That makes sense.   I think what you are saying is you want the G29 Z (with no constant in the command) to 'Baby Step' the nozzle down and when the Encoder Wheel is clicked, that number gets locked into the Z-Offset.   I'll have to look at the code, but I think all of the support routines are in place to do that.  If that is how you want it to behave, that is pretty straight forward.

adamfilip, 2016-08-04:
> OK! That makes sense. I think what you are saying is you want the G29 Z (with no constant in the command) to 'Baby Step' the nozzle down and when the Encoder Wheel is clicked, that number gets [...]

Now for specifics: The Z-Offset is part of the Mesh (and UBL) System. Should it always be active even if UBL is turned off? I think the answer is "Yes!" If so, maybe it should be pulled out of UBL. That is not that hard to do because the LCD Panel already has an option to adjust Z-Offset if you enable MANUAL_BEDLEVELING.

Yes this all makes sense to me. Pull it out of UBL. use a seperate tool to figure out Offset.
then user can update config.h if they like to the value it generates.  This would be super handy

Roxy-3D, 2016-08-04:
> Yes this all makes sense to me. Pull it out of UBL. use a seperate tool to figure out Offset.
then user can update config.h if they like to the value it generates. This would be super handy

I think I have the Z-Offset editor working.   But I'm cleaning up a few other things at the same time.   Later today I'll have it all uploaded to devel_UBL branch.   

I think what is going to happen in practice is people will just get their Z_PROBE_OFFSET_FROM_EXTRUDER close to correct, and not worry about getting exact.   Probably most people just dial in the Z-Offset do compensate for any error in the Z_PROBE_OFFSET_FROM_EXTRUDER number.

adamfilip, 2016-08-04:
Once I have my offset I typically only compensate in the slicer if its off.
I also never know if im suppose to go right off the glass/bed or leave a paper gap or .1mm

Roxy-3D, 2016-08-04:
For the Slicer's....  They assume if you move to Z=0, the nozzle is just barely touching the glass.  They also assume if they move to .2 or .3mm, that is enough of a gap to print the first layer.

adamfilip, 2016-08-04:
Can you add the ability to adjust the G26 Presets for Bed Temp, Hotend Temp, Filament Size, Prime Amount etc. in Config.h so that you dont need to do remember or look up all the conditions each time :)

Roxy-3D, 2016-08-04:
> Can you add the ability to adjust the G26 Presets for Bed Temp, Hotend Temp, Filament Size, Prime Amount etc. in Config.h so that you dont need to do remember or look up all the conditions each time :)

That is already setup for the user to adjust to their own machine.   It is at the top of the G26.cpp file.  I didn't want to add extra clutter to the Configuration.h file.   

Have you gotten a G26 to print a complete  validation image?   Have you used it to help edit the Mesh with a G29 P4 R  ???   I usually do the G26 shifted toward a corner of the bed and give it a G26 P C   That doesn't waist a lot of time moving big distances but still works its way across the bed nicely.

adamfilip, 2016-08-04:
Thanks for the tip about the G26.Cpp i will make some edits. didnt realize

I did manage to get a G26 to print and it looked pretty solid. I didnt feel it necessary to adjust
havent played with editing mesh yet.

Roxy-3D, 2016-08-04:
That's pretty impressive if you got an entire G26 printed and don't see a need to edit anything!

adamfilip, 2016-08-04:
True is was!. but it was way over extruding at the time. and with a 5x5 mesh. 
need to keep testing. looking forward to your updates :)

Roxy-3D, 2016-08-04:
Good point about the over extruding.   I noticed even when I had the Mesh very well defined...  If I dialed back the amount of extrusion, I could still find areas that 'needed' some work to be perfect.  But as long as you can get good adhesion everywhere on the bed, it doesn't really matter.  At least for the things that I print.

adamfilip, 2016-08-05:
Trying to compile todays verson. maybe i missed a setting but im getting this error

sketch\ultralcd.cpp: In function 'void lcd_update()':

ultralcd.cpp:2629: error: 'currentScreen' was not declared in this scope

    if ( currentScreen != _lcd_babystep_z)  {

         ^

ultralcd.cpp:2629: error: '_lcd_babystep_z' was not declared in this scope

    if ( currentScreen != _lcd_babystep_z)  {

                          ^

ultralcd.cpp:2631: error: 'lcd_babystep_z' was not declared in this scope

   lcd_babystep_z();

                  ^

exit status 1
'currentScreen' was not declared in this scope

Roxy-3D, 2016-08-05:
I pulled down the current code and put my Configuration.h and Configuration_adv.h file into it just to make sure it would compile.   I'm actually running that very code right now because I'm trying to resolve if we have a compiler bug in the floating point math.    

Can you try it with my Configuration.h & Configuration_adv.h file and see if you still have a problem?   And what ever you find, I'm interested in hearing the results.   Did you change anything on your settings?  As it turns out...  currentScreen is used by the new Z-Offset editing code I put in place.  It is very believable that I screwed something up.

adamfilip, 2016-08-05:
when i download and use the example config under your name it compiles fine

[10 more replies]

closed 2017-10-31 by thinkyhead