MarlinFirmware/Marlin #4439: Full Grapics Display does not work any more after #4408

floyd871, 2016-07-28:
Hi,
i have build a Delta printer with the Megatronics3 Board with the "REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER". After Pull Request  #4408 the Display remains without any text.  I have identified the problem. It seems that the display is now only serial triggered, but i have wired up mine via D4-D7. So i commented out the new stuff and it worked well again! Is there a configuration option for that?

/*#if ENABLED(DOGLCD)

  #if ENABLED(U8GLIB_ST7920)
    #define LCD_PINS_RS     56 //CS chip select /SS chip slave select
    #define LCD_PINS_ENABLE 51 //SID (MOSI)
    #define LCD_PINS_D4     52 //SCK (CLK) clock
    #define SD_DETECT_PIN 35
  #endif

#else*/

  #define LCD_PINS_RS 32
  #define LCD_PINS_ENABLE 31
  #define LCD_PINS_D4 14
  #define LCD_PINS_D5 30
  #define LCD_PINS_D6 39
  #define LCD_PINS_D7 15
  
  #define SHIFT_CLK 43
  #define SHIFT_LD 35
  #define SHIFT_OUT 34
  #define SHIFT_EN 44

  #define SD_DETECT_PIN 56 // Megatronics v3.1 only

//#endif

Blue-Marlin, 2016-07-28:
Yes!
You have a `REPRAP_DISCOUNT_SMART_CONTROLLER` - not a `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`

floyd871, 2016-07-28:
Are you sure?
I have this one:
http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller

NOT this one:
http://reprap.org/wiki/RepRapDiscount_Smart_Controller

fabtopia, 2016-07-28:
I have the same problem. And i'm sure i have the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

fabtopia, 2016-07-28:
The double enable check make no sense. The comment says "IMPORTANT NOTE: The U8glib library is required for Full Graphic Display!" So if you have DOGLCD is enable U8GLIB_ST7920 is enabled too otherwise you got no Pins defened

I propose:

#if ENABLED(REPRAPWORLD_GRAPHICAL_LCD)
  #define LCD_PINS_RS     56 //CS chip select /SS chip slave select
  #define LCD_PINS_ENABLE 51 //SID (MOSI)
  #define LCD_PINS_D4     52 //SCK (CLK) clock
  #define SD_DETECT_PIN 35
#else
  #define LCD_PINS_RS 32
  #define LCD_PINS_ENABLE 31
  #define LCD_PINS_D4 14
  #define LCD_PINS_D5 30
  #define LCD_PINS_D6 39
  #define LCD_PINS_D7 15
  
  #define SHIFT_CLK 43
  #define SHIFT_LD 35
  #define SHIFT_OUT 34
  #define SHIFT_EN 44

  #define SD_DETECT_PIN 56 // Megatronics v3.1 only
#endif

Maybe @brupje  from reprapworld can check if this works for him too

AnHardt, 2016-07-28:
> So if you have DOGLCD is enable U8GLIB_ST7920 is enabled too otherwise you got no Pins defened

It's the other way around. If `U8GLIB_ST7920` is defined also `DOGLCD` is defined. But there are several possibilities to define `DOGLCD` without `U8GLIB_ST7920`.

The #else seems to be at the wrong #if level.
This could be about right
```diff
@@ -111,28 +111,26 @@
   #if ENABLED(U8GLIB_ST7920)
     #define LCD_PINS_RS     56 //CS chip select /SS chip slave select
     #define LCD_PINS_ENABLE 51 //SID (MOSI)
     #define LCD_PINS_D4     52 //SCK (CLK) clock
     #define SD_DETECT_PIN 35
+  #else
+    #define LCD_PINS_RS 32
+    #define LCD_PINS_ENABLE 31
+    #define LCD_PINS_D4 14
+    #define LCD_PINS_D5 30
+    #define LCD_PINS_D6 39
+    #define LCD_PINS_D7 15
+
+    #define SHIFT_CLK 43
+    #define SHIFT_LD 35
+    #define SHIFT_OUT 34
+    #define SHIFT_EN 44
+
+    #define SD_DETECT_PIN 56 // Megatronics v3.1 only
   #endif
 
-#else
-
-  #define LCD_PINS_RS 32
-  #define LCD_PINS_ENABLE 31
-  #define LCD_PINS_D4 14
-  #define LCD_PINS_D5 30
-  #define LCD_PINS_D6 39
-  #define LCD_PINS_D7 15
-  
-  #define SHIFT_CLK 43
-  #define SHIFT_LD 35
-  #define SHIFT_OUT 34
-  #define SHIFT_EN 44
-
-  #define SD_DETECT_PIN 56 // Megatronics v3.1 only
-
 #endif
 
 // Buttons are directly attached using keypad
 #define BTN_EN1 44
 #define BTN_EN2 45
```

A similar change is needed in `pins_MINITRONICS.h `

fabtopia, 2016-07-28:
This gives me the RC7 boot screen but no menu. So the display works. Any clue @floyd871?

fabtopia, 2016-07-28:
The code @AnHardt does not work iether because the conditions_lcd file give the following code:
#if ENABLED(REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)
    #define DOGLCD
    #define U8GLIB_ST7920
    #define REPRAP_DISCOUNT_SMART_CONTROLLER
  #endif

This way the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER still gets the wrong pinout

fabtopia, 2016-07-28:
With my code i get the bootscreen and then nothing... So the display works

thinkyhead, 2016-07-28:
@fabtopia Does having an SD card inserted have any effect on the screen not showing anything after the bootscreen?

fabtopia, 2016-07-28:
No it does not but i mis the sdcard ok at the end of the serial boot info. When i recomment de SDSUPPORT the serail boot info gets in a loop
start
echo:Marlin 1.1.0-RC7
echo: Last Updated: 2016-07-26 12:00 | Author: (iTopia, Pandora)
Compiled: Jul 28 2016
echo: Free Memory: 4253  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X160.00 Y160.00 Z6400.00 E274.00
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z2.50 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P1000.00 R1500.00 T1500.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P43.91 I1.93 D250.33 C100.00 L20
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z0.00
start
echo:Marlin 1.1.0-RC7
echo: Last Updated: 2016-07-26 12:00 | Author: (iTopia, Pandora)
Compiled: Jul 28 2016
echo: Free Memory: 4253  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X160.00 Y160.00 Z6400.00 E274.00
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z2.50 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P1000.00 R1500.00 T1500.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P43.91 I1.93 D250.33 C100.00 L20
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z0.00

fabtopia, 2016-07-28:
When i recomment the define SDSUPPORT the screen is going on and of and the serial bootinfo gets in a loop:

"echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z0.00
start
echo:Marlin 1.1.0-RC7
echo: Last Updated: 2016-07-26 12:00 | Author: (iTopia, Pandora)
Compiled: Jul 28 2016
echo: Free Memory: 4253  PlannerBufferBytes: 1232
"

thinkyhead, 2016-07-28:
Hmm. So turning off `SDSUPPORT` seems to be causing continuous rebooting. Are you using a Megatronics version 3.0 or 3.1?

floyd871, 2016-07-29:
I have Megetronics 3.0, but i did not checked the SD functionality yet.

brupje, 2016-07-29:
I have tested RC7 against Megatronics, RAMPS and Minitronics with the ReprapWorld Graphical LCD and found it to be working ok. Also disabling SDSUPPORT with or without LCD on megatronics 3 does not seem to cause any problems.

fabtopia, 2016-07-29:
I have a Megatronics 3.0 but used the on board card reader of the R_D_F_G_S_C with RC3 and RC6 with success. I will try to disconnect the SD pins

fabtopia, 2016-07-29:
I will open a new issue. because when i comment the R_D_F_G_S-C and the SDSUPPORT, the Board will still reboot and reboot. Even if i disconnect de display.

thinkyhead, 2016-07-29:
@fabtopia Keep using `RCBugFix` going forward, as we're continuing to patch stuff as we find it.

boelle, 2016-08-11:
So what happend here?

fabtopia, 2016-08-11:
I made a wire failure after some testing without display and connecting it back on. So no new issue will be opened.

closed 2016-08-11 by thinkyhead