MarlinFirmware/Marlin #4440: Unable to upload RCBugFix to Mega 2560 using 1.6.1

funbobby2001, 2016-07-28:
Arduino: 1.6.10 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Sketch uses 125,432 bytes (49%) of program storage space. Maximum is 253,952 bytes.
Global variables use 5,288 bytes (64%) of dynamic memory, leaving 2,904 bytes for local variables. Maximum is 8,192 bytes.
C:\Program Files\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega2560 -cwiring -PCOM3 -b115200 -D -Uflash:w:C:\Users\Bobby\AppData\Local\Temp\buildada0f219018d72bfcf61d33f47621a94.tmp/Marlin.ino.hex:i 
An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

funbobby2001, 2016-07-28:
Seems to work with 1.6.9

jbrazio, 2016-07-28:
Could you please enable verbose compilation and upload mode, try again with 1.6.10 and post here the log file ? Thanks.

thinkyhead, 2016-07-29:
1.6.10 was acting up for me also. So I went back to 1.6.9.

Blue-Marlin, 2016-07-29:
There is no 'general' error with Arduino IDE 1.6.10. I can upload to a Mega2560. Win7 64.

jbrazio, 2016-07-29:
Exactly, a verbose log report is important.
@funbobby2001

funbobby2001, 2016-07-29:
That _was_ the verbose upload log, since uploading was the issue, but I can redo it with verbose compile as well if you think it would help.

Roxy-3D, 2016-07-29:
Can somebody explain things to me?   I'm worried I'm going to need to reload my BootLoader.   My Atmega2560 board has a 6 pin ISP connector on it.   There are very low cost AVR ISP Programmers on eBay.  For example, this one:  

http://www.ebay.com/itm/USBASP-USBISP-AVR-Programmer-Adapter-Cable-USB-ATMEGA8-ATMEGA128-Arduino-NEW-/322165885246?hash=item4b0295493e:g:N~MAAOSwMNxXbBKZ

But all of the AVR ISP programmers have 10 pin connectors.    Obviously, that won't work.   Can somebody educate me on this topic?

jbrazio, 2016-07-29:
When I bought my USBASP it came with two cables, a 6 pin and a 10 pin.
It seems you can now [buy adapters](http://www.ebay.com/itm/10Pin-Convert-To-6Pin-Adapter-Board-Converter-For-ATMEL-STK500-AVRISP-USBASP-/222185215992?hash=item33bb4557f8:g:iFgAAOSw7s5XhiGJ) or a [USBASP with the 6pin ribbon cable](http://www.ebay.com/itm/ISP-USB-5V-USBasp-AVR-Programmer-ATMEGA8-ATMEGA128-6PIN-Wire-Support-Win7-64Bit-/191411946654?hash=item2c910a849e:g:-ZgAAOxyIPNTcZHK).

As you can see, all the ICSP signals are on the 10 pin header also.
[image]

Roxy-3D, 2016-07-29:
I bet the AtMega2560 designer decided to save a little board space and connector cost and chopped the extra GND pins.    I was going to order the USBASP you linked to up above, but it says they don't want any USA orders.  :(      I'll see if I can hunt down another one.

jbrazio, 2016-07-29:
I find you another [one](http://www.ebay.com/itm/USBasp-USB-5V-ISP-AVR-Programmer-ATMEGA8-ATMEGA128-6PIN-Wire-Support-Win7-64Bit-/351071017800?hash=item51bd769b48:g:-ZgAAOxyIPNTcZHK), [two](http://www.ebay.com/itm/ISP-USB-5V-USBasp-AVR-Programmer-ATMEGA8-ATMEGA128-6PIN-Wire-Support-Win7-64Bit-/191411946654?hash=item2c910a849e:g:-ZgAAOxyIPNTcZHK), [three](http://www.ebay.com/itm/USB-ISP-5V-USBasp-AVR-Programmer-ATMEGA8-ATMEGA128-Win7-64Bit-6PIN-Wire-Support-/181434655946?hash=item2a3e5924ca:g:-boAAOxyIPNTcZHM), [four](http://www.ebay.com/itm/New-USB-ISP-5V-USBasp-AVR-Programmer-ATMEGA8-ATMEGA128-6PIN-Wire-Support-Win7-/111377757219?hash=item19eea1b423:g:-boAAOxyIPNTcZHM) and [five](http://www.ebay.com/itm/5V-AVR-Programmer-USBasp-USB-ISP-ATMEGA8-ATMEGA128-6PIN-Wire-Support-Win7-64Bit-/271489740923?hash=item3f360cb07b:g:-ZgAAOxyIPNTcZHK). :-D

Actually the 6pin ICSP it's quite common as it the the Arduino standard. What I am now seeing is that people selling USBASP or USBtiny are cutting down an only ship one cable type so they can sell the adapter PCB boards.

Roxy-3D, 2016-07-29:
Thanks!   I soon will be able to recover from a wiped out bootstrap.   Can I leave this connected so I don't have to discount and reconnect PronterFace all the time?  That would be real nice!

closed 2016-07-29 by funbobby2001

jbrazio, 2016-07-29:
> Can I leave this connected so I don't have to discount and reconnect PronterFace all the time

Actually uploading using the ICSP is slower than using the bootloader sketch upload method I believe.
There is no way around disconnecting the serial, reconnecting the serial..

@Roxy-3D If the debug you're doing does not imply G-Code submission, then I would suggest to use the Arduino's IDE builtin serial console, you can leave it open, when you click upload it will temporary disconnect it for you and connect it back when the upload is complete which is quite convenient.

Roxy-3D, 2016-07-30:
@jbrazio  I checked out the serial console.   That is nice.   But every once in a while, it won't let go of the serial port.   I'll have to play with it some more.   One bad thing is it doesn't use proportional space fonts like PronterFace and Repetier Host, so that screws up all the formatting on things. 

Thanks for the tip!

esenapaj, 2016-08-13:
I had a similar uploading problem, but it seems that newest hourly build has resolved those.