MarlinFirmware/Marlin #4442: Z probe servo not working with RCBugFix

funbobby2001, 2016-07-28:
Servo was working with 1.0.2 
With RCBugFix I just get "Servo <angle> out of range" from Repetier Host with M280 no matter what angle I try.
(Have not tried RC, just the BugFix)
Far as I can tell I've duplicated all the appropriate settings from my working Marlin 1.0.2 config.
Board is a Ramps 1.4A and I've tried both "BOARD_RAMPS_14_EFB      43" and "BOARD_RAMPS_13_EFB      33" since my working config in 1.0.2 was 33.
Z probe is also used for homing.

Roxy-3D, 2016-07-28:
The servo's used to be defined as a set.   Because Z (where the servo is typically used) is the 3rd part of the coordinate, the Z Servo was usually #2 (starting from 0).     You might try doing your M280 with that in mind.    M280 P0 S90 and see what happens.

Blue-Marlin, 2016-07-28:
Servo numbering begins with 0. Try M280 P0 S45
If it worked for you in Marlin 1.0.2 with board = 33 its now 43!

funbobby2001, 2016-07-29:
Got it working, thanks! Not even sure what I did other than using "43" instead of "BOARD_RAMPS_14_EFB"

closed 2016-07-29 by funbobby2001