MarlinFirmware/Marlin #4443: Configuration confusion on my part, axes movement direction for corexy

MichaelBrock, 2016-07-28:
I apologize for advance in taking up the groups attention with what is probably an inane question.

I have a corexy.  The bed rests at the bottom and when homing raises, until a inductive sensor triggers (so a typical corexy arrangement).   I upgraded to the RCBugFix a few days ago and since have had issues. I reported elsewhere about G28 attempting to drive the bed even further downwards but I have worked around that.    After homing X and Y  the head moves to the safe_homing position and the position is reported as  current_position=(85.00, 120.00, 0.00).  The bed then raises correctly to the sensor and when the sensor is triggered it reports:  current_position=(85.00, 120.00, -4.00) : > TRIGGER ENDSTOP).   

After G28 is completed the current position is reported as current_position=(85.00, 120.00, -16.55).  I am still working out why that is adjusted  (current_position[axis] = LOGICAL_POSITION(base_home_pos(axis), axis); changes the z offset to -20 which is then adjusted by the probe offset to -16.55)  but my question at this point is much simpler:

Shouldn't the z position be a positive number?   I suspect that something in my configuration is incorrect but I can't find anything that has changed since before upgrading.

Blue-Marlin, 2016-07-28:
Please append Configuration.h and Configuration_adv.h (simply rename .h to .txt and draw into the browser)

MichaelBrock, 2016-07-28:
I should have provide those originally.  But here they are now. 

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/389242/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/389243/Configuration_adv.txt)

Blue-Marlin, 2016-07-28:
Do you have a special reason for 
#define Z_MIN_POS -20

MichaelBrock, 2016-07-28:
No particular reason.   In my initial testing I found that I could not move the bed to close the gap to measure the probe offset and so adjusted that.  Just a "leftover" from a prior attempt.

Blue-Marlin, 2016-07-28:
Please set back to 0.
The next time you want to find the z-probe-offset just set it too large

MichaelBrock, 2016-07-28:
I will set it back to 0.  I was operating on the suggestion made [here](https://github.com/MarlinFirmware/Marlin/issues/3647#issuecomment-216077543) by thinkyhead.  In that same thread you recommended "Better than fooling about the z-height with G92 is, to simply do the obvious. For going beyond the endstop position - switch off all kinds of endstops".  Can you point out how to switch off all of the endstops and still permit homing & ABL.  

Doesn't Z_MIN_POS simply serve as "// Travel limits after homing (units are in mm)"  so simply prevents the printer from traveling outside of 300  to -20?     

Thanks for the replies and the help!

MichaelBrock, 2016-07-28:
To answer my own question regarding Z_MIN_POS:  it is not simply a travel limit as described.  It is used to define:    

 #define Z_HOME_POS (Z_HOME_DIR < 0 ? Z_MIN_POS : Z_MAX_POS)

and Z_HOME_POS is in turn used to set the current_position after homing.  So that's where the -20 is coming from that I'm seeing.  I was headed in the right direction!   So all of the _MIN_POS and _MAX_POS (if the axis is inverted) variables in configuration.h  are used to set the current position after homing and shouldn't be used to simply permit movement past the "limit" for calibration.

Blue-Marlin, 2016-07-28:
> Doesn't Z_MIN_POS simply serve as "// Travel limits after homing (units are in mm)" so simply prevents the printer from traveling outside of 300 to -20? 
Not as long the software endstops are off.

>  Can you point out how to switch off all of the endstops and still permit homing & ABL. 
Switch off software endstops in the config. Switch on hardware endstops/probe in the config.
Home.
Use M120/M121 to switch the hardware endstops on/off by will.

But currently the way to begin with a too large z-probe-distance seems to be easier.

Changes to switch on/off the software endstops by G/M-code are still under development.

MichaelBrock, 2016-07-28:
Your suggestion for a fix was the right one (naturally).  It is working correctly now, and even more importantly, working how I would expect it to.   I.e. it's making sense!     I agree that the "too large z-probe offset" is the way to go and was simple and straight-forward.

thinkyhead, 2016-07-29:
Sorry! Did I forget to mention `MANUAL_Z_HOME_POS` when I recommended setting `Z_MIN_POS` to a lower value? Yeah, if you mess with `Z_MIN_POS` it will affect `Z_HOME_POS` unless `MANUAL_HOME_POSITIONS` is used to explicitly set your home positions.

closed 2016-07-29 by thinkyhead