MarlinFirmware/Marlin #4446: '>' disapear when going to far in menu

spilz87, 2016-07-28:
It's not really important, but sometimes I'm looking for the '>' which disapears when I navigate in menu with coder and when I go too far.
I don't know if it's the good location to report this "issue".

sorry for my English

Spilz

jbrazio, 2016-07-29:
Thank you @spilz87 for reporting the issue, we're looking into it.

thinkyhead, 2016-07-29:
Probably a simple bug. I'll mess with my LCD tonight and come up with a patch.

thinkyhead, 2016-08-02:
@spilz87 I believe this might fix it for you. Add the following new line in `ultralcd.cpp`:
```diff
// Scroll through menu items, scrolling as-needed to stay in view
#define END_MENU() \
  } \
  if (encoderLine >= _thisItemNr) { \
    encoderLine = _thisItemNr - 1; \
    encoderPosition = encoderLine * (ENCODER_STEPS_PER_MENU_ITEM); \
+   lcdDrawUpdate = LCDVIEW_CALL_REDRAW_NEXT; \
  } \
  if (encoderLine >= encoderTopLine + LCD_HEIGHT) { \
    encoderTopLine = encoderLine - (LCD_HEIGHT - 1); \
    lcdDrawUpdate = LCDVIEW_CALL_REDRAW_NEXT; \
  } \
  UNUSED(_skipStatic)
```

spilz87, 2016-08-03:
Thanks,
I will test it when I come back home, end of week

boelle, 2016-08-11:
@spilz87 did you test this?

jbrazio, 2016-08-11:
Closed by #4512.
Pleas reopen this issue if the problem still persists.

closed 2016-08-11 by jbrazio

spilz87, 2016-08-11:
I just test it, it seems ok. 
I didn't try all menu

Thanks for your help