MarlinFirmware/Marlin #4454: After first movement, printer stuck in "busy: processing" (RC7, delta)

oxivanisher, 2016-07-29:
I have uploaded RC7 to my Kossel delta printer. I used the default configuration which I uploaded some time ago. I had to tewak the filament diameter (which was somehow wrong) and the homing speed was way to high. This was no problem.

**BUT:** After connecting the printer, I can talk to it and it responds correctly. But as soon as I issue some movement command (even G28), it gets executed but after that the printer only responds `Recv: echo:busy: processing`. I can get it only out of this state by reconnecting.

I have tried connecting it to Octoprint and Simplify3D with the same result.

thinkyhead, 2016-07-29:
To help narrow down the cause…

- Do you get the same behavior if you skip the `T0` command?
- Do you see similar behavior (no busy messages) with `HOST_KEEPALIVE_FEATURE` disabled?
- What is your Baud Rate? HAve you tried a lower baud rate (<= 115000)?

oxivanisher, 2016-07-29:
* I do not know how to skip `T0`. But it happens only after i initiated any move, not at the point of `T0`.
* Disabling `HOST_KEEPALIVE_FEATURE` did not help
* I forgot to mention this, but I tried to lower the baud rate to 115000 already. Although it worked with "full speed" before.

thinkyhead, 2016-07-29:
> after i initiated any move

Hmm, very peculiar. I did some delta movement fixes yesterday and pushed them into `RCBugFix`. Can you test `RCBugFix` to see if I managed to fix the problem?

oxivanisher, 2016-07-29:
Ok, it even got stranger... I used the `RCBugFix`, applied the changes to the example config file mentioned above (it looks as at least one config option got renamed again ^^) and successfully uploaded it. I not do not get the "processing" messages, but it behaves the same as before (not moving after `G28`. (`HOST_KEEPALIVE_FEATURE` is enabled)

```
Attempting connection at \\.\COM6...
RECEIVED: start
  Connected to machine!
RECEIVED: echo:Marlin 1.1.0-RCBugFix
Marlin 1.1.0-RCBugFix
SENT: T0
RECEIVED: echo: Last Updated: 2016-04-27 12:00 | Author: (oxivanisher)
Last Updated: 2016-04-27 12:00 | Author: (oxivanisher)
RECEIVED: Compiled: Jul 30 2016
RECEIVED: echo: Free Memory: 3386  PlannerBufferBytes: 1232
Free Memory: 3386  PlannerBufferBytes: 1232
RECEIVED: echo:Hardcoded Default Settings Loaded
Hardcoded Default Settings Loaded
RECEIVED: echo:Steps per unit:
Steps per unit:
RECEIVED: echo:  M92 X100.00 Y100.00 Z100.00 E158.00
M92 X100.00 Y100.00 Z100.00 E158.00
RECEIVED: echo:Maximum feedrates (mm/s):
Maximum feedrates (mm/s):
RECEIVED: echo:  M203 X200.00 Y200.00 Z200.00 E25.00
M203 X200.00 Y200.00 Z200.00 E25.00
RECEIVED: echo:Maximum Acceleration (mm/s2):
Maximum Acceleration (mm/s2):
RECEIVED: echo:  M201 X9000 Y9000 Z9000 E10000
M201 X9000 Y9000 Z9000 E10000
RECEIVED: echo:Accelerations: P=printing, R=retract and T=travel
Accelerations: P=printing, R=retract and T=travel
RECEIVED: echo:  M204 P2000.00 R3000.00 T3000.00
M204 P2000.00 R3000.00 T3000.00
RECEIVED: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
RECEIVED: echo:  M205 S0.00 T0.00 B20000 X20.00 Z20.00 E20.00
M205 S0.00 T0.00 B20000 X20.00 Z20.00 E20.00
RECEIVED: echo:Home offset (mm)
Home offset (mm)
RECEIVED: echo:  M206 X0.00 Y0.00 Z0.00
M206 X0.00 Y0.00 Z0.00
RECEIVED: echo:Endstop adjustment (mm):
Endstop adjustment (mm):
RECEIVED: echo:  M666 X0.00 Y0.00 Z0.00
M666 X0.00 Y0.00 Z0.00
RECEIVED: echo:Delta settings: L=diagonal_rod, R=radius, S=segments_per_second, ABC=diagonal_rod_trim_tower_[123]
Delta settings: L=diagonal_rod, R=radius, S=segments_per_second, ABC=diagonal_rod_trim_tower_[123]
RECEIVED: echo:  M665 L319.80 R175.10 S160.00 A0.00 B0.00 C0.00
M665 L319.80 R175.10 S160.00 A0.00 B0.00 C0.00
RECEIVED: echo:Material heatup parameters:
Material heatup parameters:
RECEIVED: echo:  M145 S0 H180 B70 F255
M145 S0 H180 B70 F255
RECEIVED: echo:  M145 S1 H240 B100 F255
M145 S1 H240 B100 F255
RECEIVED: echo:PID settings:
PID settings:
RECEIVED: echo:  M301 P22.04 I1.65 D73.67 C100.00 L20
M301 P22.04 I1.65 D73.67 C100.00 L20
RECEIVED: echo:Filament settings: Disabled
Filament settings: Disabled
RECEIVED: echo:  M200 D1.75
M200 D1.75
RECEIVED: echo:  M200 D0
M200 D0
RECEIVED: echo:Z-Probe Offset (mm):
Z-Probe Offset (mm):
RECEIVED: echo:  M851 Z0.30
M851 Z0.30
SENT: M105
RECEIVED: echo:SD card ok
RECEIVED: echo:Active Extruder: 0
Active Extruder: 0
RECEIVED: ok
RECEIVED: echo:SD card ok
RECEIVED: ok T:20.5 /0.0 B:20.4 /0.0 B@:0 @:0
RECEIVED: 
SENT: M105
RECEIVED: ok T:20.5 /0.0 B:20.5 /0.0 B@:0 @:0
RECEIVED: 
SENT: M105
RECEIVED: ok T:20.5 /0.0 B:20.4 /0.0 B@:0 @:0
RECEIVED: 
SENT: G28
RECEIVED: X:140.00 Y:140.00 Z:331.14 E:0.00 Count X: -2147483648 Y:-2147483648 Z:-2147483648
RECEIVED: ok
SENT: M105
RECEIVED: ok T:20.7 /0.0 B:20.7 /0.0 B@:0 @:0
RECEIVED: 
SENT: M105
RECEIVED: ok T:20.7 /0.0 B:20.7 /0.0 B@:0 @:0
RECEIVED: 
SENT: M105
RECEIVED: ok T:20.8 /0.0 B:20.7 /0.0 B@:0 @:0
RECEIVED: 
SENT: M105
RECEIVED: ok T:20.7 /0.0 B:20.6 /0.0 B@:0 @:0
RECEIVED: 
SENT: M105
RECEIVED: ok T:20.7 /0.0 B:20.6 /0.0 B@:0 @:0
RECEIVED: 
SENT: M105
RECEIVED: ok T:20.7 /0.0 B:20.7 /0.0 B@:0 @:0
RECEIVED: 
SENT: M105
RECEIVED: ok T:20.7 /0.0 B:20.7 /0.0 B@:0 @:0
RECEIVED: 
SENT: M105
RECEIVED: ok T:20.9 /0.0 B:20.7 /0.0 B@:0 @:0
RECEIVED: 
SENT: M105
RECEIVED: ok T:20.7 /0.0 B:20.7 /0.0 B@:0 @:0
RECEIVED: 
SENT: G1 Z100 F5000
RECEIVED: ok
SENT: M105
RECEIVED: ok T:20.7 /0.0 B:20.6 /0.0 B@:0 @:0
```

Blue-Marlin, 2016-07-29:
Ups???
> RECEIVED: X:140.00 Y:140.00 Z:331.14 E:0.00 Count X: -2147483648 Y:-2147483648 Z:-2147483648

M502
M500
retest
and if not solved - Config please.

thinkyhead, 2016-07-30:
> Config please.

https://github.com/MarlinFirmware/Marlin/files/390010/Configuration.txt

thinkyhead, 2016-07-30:
Just for fun, try building with a different version of Arduino IDE. Try 1.6.8 and earlier.

oxivanisher, 2016-07-30:
> M502
> M500
> retest
 I did this when I went from RC6 to RC7. Also i tried it just now and it did not help.

Also i tried Arduino IDE 1.6.8 without any success. 

And with more logging, it still looks almost the same
```
RECEIVED: ok T:20.9 /0.0 B:20.7 /0.0 B@:0 @:0
RECEIVED: 
SENT: G28
RECEIVED: echo:G28
G28
RECEIVED: X:140.00 Y:140.00 Z:331.14 E:0.00 Count X: -2147483648 Y:-2147483648 Z:-2147483648
RECEIVED: ok
SENT: M105
RECEIVED: echo:M105
M105
RECEIVED: ok T:21.0 /0.0 B:20.7 /0.0 B@:0 @:0
RECEIVED: 
[..]
SENT: G1 Z100
RECEIVED: echo:G1 Z100
G1 Z100
RECEIVED: ok
SENT: M105
RECEIVED: echo:M105
M105
RECEIVED: ok T:20.8 /0.0 B:20.7 /0.0 B@:0 @:0
RECEIVED: 
```
But nothing happens

oxivanisher, 2016-07-30:
I realized something. I am able to do `G28` as many times as I like without disconnecting and it seems to work. Just "normal" `G1`'s do not seem to work.

oxivanisher, 2016-07-30:
With the current version of the RcBugFix it is printing again! :+1:

closed 2016-07-30 by oxivanisher

thinkyhead, 2016-07-31:
There were some recent changes specific to a delta movement bug that surely did the trick.