MarlinFirmware/Marlin #4460: Micro pauses in Delta X/Y moves

rogueqd, 2016-07-29:
The print head does a micro pause at the end of each segment in a delta X/Y move. The pause is short enough that it doesn't really affect the print quality, but it makes the printer movement look very jerky.

thinkyhead, 2016-07-30:
- What kind of board are you using?
- What kind of LCD controller do you have?
- What is your `DELTA_SEGMENTS_PER_SECOND` setting?

rogueqd, 2016-07-30:
Sorry, forgot to attach config.
BOARD_RAMPS_13_EFB
PANEL_ONE LCD
DELTA_SEGMENTS_PER_SECOND 200
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/391592/Configuration.h.txt)

rogueqd, 2016-07-30:
Maybe segments in the wrong term. Here's a short video.
https://www.youtube.com/watch?v=QxcED-5MVwA

thinkyhead, 2016-07-30:
Yeah, on `DELTA` every move is split into very small "segments" to prevent dipping of the effector. This calculation does take some extra CPU. Try lowering `DELTA_SEGMENTS_PER_SECOND` by 20 and re-test. Keep lowering by 20 until the pauses go away. If the pauses don't go away, then we should look closer.

rogueqd, 2016-07-30:
Lowering the DELTA_SEGMENTS_PER_SECOND didn't have any effect. I went down to 100.
Lowering the F parameter in the G1 commands smoothed out the end of the move. I reduced F down to 4000 (from 8000) and the second half of the move was smooth.
At 2000 I only got one pause at the end of the first segment, then the rest of the move was smooth.

thinkyhead, 2016-07-30:
You can also try lowering your `DEFAULT_MAX_FEEDRATE`. The default for most deltas is pretty high. 500mm/s (30000mm/m). See if you can find a sweet spot with that setting.

rogueqd, 2016-07-30:
Ok, so I tried reducing the numbers. I lowered my DEFAULT_MAX_FEEDRATE is down to 10 on all axes.
All the acceleration variables down to 0.5 and the jerk variables to 0.2.
I didn't seen any change in the movement speed and the judders were still occurring.

I tried lower the F values in G1 moves, the judders stop at 1300, with slight juddering noticeable at 1400.

I wasn't getting any juddering on the old firmware, but I'm having trouble getting it to compile since I upgraded my arduino software to compile the new firmware. (it was really old, lol)

Blue-Marlin, 2016-07-31:
How are these moves initiated? By LCD-panel, g-code via USB, SD-Card? Does it make a difference?

rogueqd, 2016-07-31:
Both g-code via usb and sd-card. I didn't try via LCD panel.

I don't remember noticing it before issue #4433 was fixed. Not sure if that's relevant.

Blue-Marlin, 2016-07-31:
Did you try after #4450

rogueqd, 2016-08-01:
I downloaded the latest RCBugFix juyst now and it's fixed. Thanks.

closed 2016-08-01 by thinkyhead