MarlinFirmware/Marlin #4461: Changing Esteps/mm during print job causes print head to raise.

rogueqd, 2016-07-29:
Sometimes I change Esteps/mm during a print job because it's a finer adjustment than changing the Flow %.

When I tried it today it caused the print head to raise to about 180mm high. The printer kept printing. The Z height on the LCD displayed as the correct Z height for the print job, but the print head was ~180mm above the print bed.

Kossel Mini printer.
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/391546/Configuration.h.txt)

thinkyhead, 2016-07-30:
I recently added code to take account when steps-per-mm changes for any axis and re-figure the current position. Clearly this can be skipped for the E axis, so I will roll a patch for that shortly.

Obviously it's a strange thing to change steps/mm values for XYZ (i.e., "ABC" on Delta) during a print, but outside of that it's allowable. The simple idea is to make sure the `planner.position` and `stepper.count` values have parity with the `current_position`. I think probably I forgot to account for Delta when re-figuring these values.

thinkyhead, 2016-07-30:
I've applied a patch to address this. Would you mind testing the latest `RCBugFix` to make sure it's not still broken?

rogueqd, 2016-07-30:
Yep, I can confirm it's resolved in the latest RCBugFix.

closed 2016-07-30 by thinkyhead