MarlinFirmware/Marlin #4462: Heatbed/Fan reversed with hotend in RAMPS 1.4 EFF config

moo1426, 2016-07-29:
When I enable my hotend with my board defined as a RAMPS 1.4 EFF the firmware enables the fan connected to the heatbed/fan2 terminal. The error is fixed if the board is defined as RAMPS 1.4 EFB. I am using singlenozzle if it makes a difference on 1.10-RC7.

thinkyhead, 2016-07-30:
Hmm, that's odd. Both EFF and EFB define `FAN_PIN` as pin 9, which I believe is the middle connector.
```cpp
#if MB(RAMPS_14_EFF) || MB(RAMPS_13_EFF) || ENABLED(IS_RAMPS_EFB)
  #define FAN_PIN           9 // (Sprinter config)
  . . .
#endif
```

thinkyhead, 2016-07-30:
You must be referring to `FAN1_PIN` and not `FAN_PIN`. I ran into this just now as I was putting together #4463. Since this is a pre-release and in previous versions of Marlin you had to self-assign the FAN1_PIN, I think this is probably safe to patch. I had been concerned that setups wired backward would be a problem, but this is new to 1.1.0.

boelle, 2016-08-11:
if its safe to patch why not fix and close this one to get the list shorter?

thinkyhead, 2016-08-11:
@boelle See the above linked merged PR #4463.

closed 2016-08-11 by thinkyhead