MarlinFirmware/Marlin #4478: Can't get SD card to work with RCBugFix

funbobby2001, 2016-07-30:
-Ramps 1.4A; using motherboard 43
-Reprap discount full graphic smart controller
Worked fine in Marlin 1.0.2 which is the last version I had.
Controller recognizes that the card has been inserted but cannot see any of the files on the card.
Repetier host cannot see the files either, but Windows explorer can through my laptop card reader.

Blue-Marlin, 2016-07-30:
#define SDSUPPORT

funbobby2001, 2016-08-01:
It is

Blue-Marlin, 2016-08-01:
Sorry i can't reproduce the problem.
Please try frormating - low level formatting the card.
If that does not help, please append the complete Configuration.h - renamed to *.txt.

jbrazio, 2016-08-01:
Also check the card's filesystem, we only support `FAT`.

funbobby2001, 2016-08-01:
It was FAT32. 
Low level formatting did not resolve the issue.

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/394573/Configuration.h.txt)

Blue-Marlin, 2016-08-01:
Your config. works with minimal changes in today's RCBugFix on my tst system. (MANUAL_HOME_POSITIONS)

Check socket and card again.

funbobby2001, 2016-08-02:
Just tried today's RCBugFix and that didn't work, so I guess it must be my board/socket.

thinkyhead, 2016-08-02:
@funbobby2001 If you revert back to 1.0.2 the SD card works okay?

funbobby2001, 2016-08-02:
I know I should try that, just been dragging my feet because I have to uninstall Arduino and reinstall an older version to upload it.

thinkyhead, 2016-08-03:
@funbobby2001 How odd. As far as I know older versions of Marlin should compile with the current toolkit.

Blue-Marlin, 2016-08-03:
@thinkyhead 
There are several configs for 1.0.2 who do not compile with Arduino versions above 1.0.6.
Search for https://github.com/MarlinFirmware/Marlin/issues?q=is%3Aissue+POINTER_REGS+is%3Aclosed

funbobby2001, 2016-08-03:
Now I can't get my old 1.0.2 to compile in Arduino 1.0.5

In file included from /dogm_lcd_implementation.h:38,
                 from ultralcd.cpp:39:
dogm_font_data_marlin.h:13: error: 'u8g_fntpgm_uint8_t' does not name a type
dogm_font_data_marlin.h:172: error: 'u8g_fntpgm_uint8_t' does not name a type
In file included from /dogm_lcd_implementation.h:40,
                 from ultralcd.cpp:39:
ultralcd_st7920_u8glib_rrd.h:43: error: 'u8g_t' was not declared in this scope
ultralcd_st7920_u8glib_rrd.h:43: error: 'u8g' was not declared in this scope
ultralcd_st7920_u8glib_rrd.h:43: error: 'u8g_dev_t' was not declared in this scope
ultralcd_st7920_u8glib_rrd.h:43: error: 'dev' was not declared in this scope
ultralcd_st7920_u8glib_rrd.h:43: error: expected primary-expression before 'msg'
ultralcd_st7920_u8glib_rrd.h:43: error: expected primary-expression before 'void'
ultralcd_st7920_u8glib_rrd.h:43: error: initializer expression list treated as compound expression
ultralcd_st7920_u8glib_rrd.h:44: error: expected ',' or ';' before '{' token

Blue-Marlin, 2016-08-03:
Did you reinstall u8glib in a matching version? (https://github.com/MarlinFirmware/Marlin/tree/1.0.x/ArduinoAddons/Arduino_1.0.x/libraries/U8glib v1.1 then)

funbobby2001, 2016-08-03:
Almost certainly not. LOL 
As soon as my print finishes I will try again. Thx

funbobby2001, 2016-08-03:
Progress has been made.
1.0.2 now compiles and uploads with the older u8glib using Arduino 1.6.9, so that's nice
I've discovered that all the files on my SD card, which I've successfully printed from before, now have no file extension.
Why trying to upload files with Repetier Host the file extensions seem to be striped.
If I manually copy files to the card with the file extension then everything is good.
How is it that the files on the card worked in the past and now they do not? Weird.
Either way I will not use the host to upload anymore because it takes FOREVER.

thinkyhead, 2016-08-03:
I haven't used the file-writing commands from a host in a very long tim. We'll have to make sure we have no bugs in that code. Otherwise it could be a bug in Repetier-Host.

closed 2017-10-31 by thinkyhead