MarlinFirmware/Marlin #4480: Stange behaviour on heater

mjmeans, 2016-07-31:
I'm using Arduino Mega, RAMPS 1.4, Marlin RC6, on a Kossel with 360mm horizontals. The heat bed is a sandwich of 1/16" cork, TriDPrinting's 240mm 12V Kapton heater, TriDPrinting 3mm x 250mm aluminum heat spreader, and 3mm x 250mm borosilicate glass.

The RAMPS board has A4988 drivers with Arctic Alumina attached aluminum heat sinks and a 40mm cooling fan. The D8 MOSFET also has a clip on heat sink I got from digikey.

(FYI, in an earlier build I did not have an aluminum heat spreader and the bed temperature varied wildly across the surface according to an IR thermometer. After adding the heat spreader, the bed was within 2 degrees across its entire surface.)

Anyway, I've noticed an strange thing a couple of times now. Sometimes it causes a runaway trigger and halt and other times it just recovers on its own. Look at the attached image. At about 30:15 you see a sharp drop then rise in bed temp and a cut of bed power then you see the bed temps rises for several seconds while no power is applied, then stabilizes and PID resumes as before. Yes, I'm using PID on the bed. During normal operation, since PID is enabled on the bed, I see the D8 led flicker. But during that drop, for several seconds, the D8 LED was on solid even though Repetier reported that it was not driving bed power.

Oh, I should point out that I had an runaway trigger several times on the hot end also so this is not limited to the heat bed. But I was not watching at those times and I can't say whether the D10 led held on during that time, but the graph looked the same.

Should I suspect an issue with Marlin, or Arduino Mega, or RAMPS?

[image]

Blue-Marlin, 2016-07-31:
About thermal errors:
```
//===========================================================================
//=============================Thermal Settings  ============================
//===========================================================================

#if DISABLED(PIDTEMPBED)
  #define BED_CHECK_INTERVAL 5000 // ms between checks in bang-bang control
  #if ENABLED(BED_LIMIT_SWITCHING)
    #define BED_HYSTERESIS 2 // Only disable heating if T>target+BED_HYSTERESIS and enable heating if T>target-BED_HYSTERESIS
  #endif
#endif

/**
 * Thermal Protection protects your printer from damage and fire if a
 * thermistor falls out or temperature sensors fail in any way.
 *
 * The issue: If a thermistor falls out or a temperature sensor fails,
 * Marlin can no longer sense the actual temperature. Since a disconnected
 * thermistor reads as a low temperature, the firmware will keep the heater on.
 *
 * The solution: Once the temperature reaches the target, start observing.
 * If the temperature stays too far below the target (hysteresis) for too long (period),
 * the firmware will halt the machine as a safety precaution.
 *
 * If you get false positives for "Thermal Runaway" increase THERMAL_PROTECTION_HYSTERESIS and/or THERMAL_PROTECTION_PERIOD
 */
#if ENABLED(THERMAL_PROTECTION_HOTENDS)
  #define THERMAL_PROTECTION_PERIOD 40        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 4     // Degrees Celsius

  /**
   * Whenever an M104 or M109 increases the target temperature the firmware will wait for the
   * WATCH_TEMP_PERIOD to expire, and if the temperature hasn't increased by WATCH_TEMP_INCREASE
   * degrees, the machine is halted, requiring a hard reset. This test restarts with any M104/M109,
   * but only if the current temperature is far enough below the target for a reliable test.
   *
   * If you get false positives for "Heating failed" increase WATCH_TEMP_PERIOD and/or decrease WATCH_TEMP_INCREASE
   * WATCH_TEMP_INCREASE should not be below 2.
   */
  #define WATCH_TEMP_PERIOD 20                // Seconds
  #define WATCH_TEMP_INCREASE 2               // Degrees Celsius
#endif

/**
 * Thermal Protection parameters for the bed are just as above for hotends.
 */
#if ENABLED(THERMAL_PROTECTION_BED)
  #define THERMAL_PROTECTION_BED_PERIOD 20    // Seconds
  #define THERMAL_PROTECTION_BED_HYSTERESIS 2 // Degrees Celsius

  /**
   * Whenever an M140 or M190 increases the target temperature the firmware will wait for the
   * WATCH_BED_TEMP_PERIOD to expire, and if the temperature hasn't increased by WATCH_BED_TEMP_INCREASE
   * degrees, the machine is halted, requiring a hard reset. This test restarts with any M140/M190,
   * but only if the current temperature is far enough below the target for a reliable test.
   *
   * If you get too many "Heating failed" errors, increase WATCH_BED_TEMP_PERIOD and/or decrease
   * WATCH_BED_TEMP_INCREASE. (WATCH_BED_TEMP_INCREASE should not be below 2.)
   */
  #define WATCH_BED_TEMP_PERIOD 60                // Seconds
  #define WATCH_BED_TEMP_INCREASE 2               // Degrees Celsius
#endif
```
If you get thermal errors - relax the values.

The moment you got the temperature falling (30:15) - was there a fan switched on or off?

mjmeans, 2016-07-31:
Not to my knowledge. The hot end fan is always on. I'm only printing ABS so there is no part cooling fan even installed. None of the electronics are mounted under the bed. Two sides of the printer are covered with Plexiglass to shield it from room AC drafts. The RAMPS is mounted on the external side of the frame with its cooling fan pointed down into the RAMPS board (spreading stepper driver heat laterally), but its cooling fan is always on.

Only the PSU (12V 30A LED type) has a fan that turns on or off by itself. But it is sitting about 6 inches away from the printer and its fan exhausts up, not toward the bed. Its airflow could create a horizontal cylindrival eddy current of air, but its 6 inches away from the bottom of the printer which is about 400mm from the center of the bed. I think the distance makes it unlikely that that fan could effect the thermistor reading.

thinkyhead, 2016-07-31:
Are you using `BOARD_RAMPS_14_EEF`?

mjmeans, 2016-08-01:
Yes. That's correct. However, I'm printing only ABS and there is not even a fan connected to D9 (part cooling) and no part cooling is enabled in repetier-host, so the D9 MOSFET should not even be activating.

AnHardt, 2016-08-03:
[image]
The red line has the expected slope for a bed not heated anymore.
The green line has much more slope. So the bed is ether actively cooled, or something is irritating the ADC. The most likely cause is a change in the supply voltage like when a big load is switched on or off.
The heater-power curve follows exactly my expectations what it should do acording to the measured temperatures. Even the temperature curve follows my expectations related to the heater power - except the fast temperature drop.

boelle, 2016-08-11:
maybe we can close this one?

closed 2016-08-11 by thinkyhead