MarlinFirmware/Marlin #4483: RC7 , run G29,Servo jitter?why?

jackshi618, 2016-07-31:
RC7 , run G29,Servo jitter?why?

Blue-Marlin, 2016-07-31:
#define DEACTIVATE_SERVOS_AFTER_MOVE ?

jackshi618, 2016-07-31:
disable!

Roxy-3D, 2016-07-31:
> #define DEACTIVATE_SERVOS_AFTER_MOVE ?

That is a good thing to try!   However, I can tell you the cheap Tower Pro 9g servo's off of eBay still jitter even with them deactivated.   I've put big capacitors on their power.  I've buffered their control signal (for the case where they are activated.)   

The one thing that helped the most (but did not fix the problem) was to move the control signal to a pin where nothing else on the port is switching.   Somehow, the Fast I/O seems to glitch all of the pins on a port when any of them change state.  

@jackshi618 I don't think this is an RC7 issue.  If you go back to RC3, it will happen there too under the right conditions.

thinkyhead, 2016-07-31:
Some points to consider towards documentation, no doubt. @MarlinFirmware/documentation-maintainers

sarf2k4, 2016-08-03:
It is most likely due to high current interference/leak from the motor's wire, wrapping them in an aluminum foil followed by an insulation tape might help eliminating it

orangeudav, 2016-08-30:
@jackshi618 Do you have oscilloscope? Interference with motor wire and floating length of impulse due error in firmware should visible difference imho

Roxy-3D, 2016-08-30:
This Servo from eBay does not seem to jitter.  It is almost exactly the same size and dimensions as the cheap Blue TowerPro Servos.   (I had to re-print with a slightly different shape for it to fit perfectly.)   

So far...   no jitter at all!

http://www.ebay.com/itm/262564045761?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

thinkyhead, 2016-08-30:
I'm sure we'll cover this in the Servos section of the documentation.

closed 2016-08-30 by thinkyhead