MarlinFirmware/Marlin #4484: Machine without any change,RC3 to RC7,Can't move Z axis?

jackshi618, 2016-07-31:
why?

Blue-Marlin, 2016-07-31:
Too complicated to explain. Try RCBugFic. Likely fixed be #4476.

Blue-Marlin, 2016-07-31:
Be a bit more verbose. At least the type of machine would help to find a better answer than: "You are wrong - it works for me". Ideally append your Configuration.h.

otvald, 2016-07-31:
@jackshi618 
1. What is your setup ? Delta, Cartesian, ?
2. Check endstops to see if triggered (M119)
3. Does G28 do anything? It does not home? (Assume DELTA)
4. No noise or hard-stopping noise from steppers?
5. You did not change anything in config? If not, did you see the changes from RC3 to RC7 in git? Did you apply these changes?

jackshi618, 2016-08-05:
CoreXY，Send G28，Z motor vibration.

thinkyhead, 2016-08-07:
@jackshi618 Have you tried adjusting your Z max speed, Z homing speed, Z acceleration…?

boelle, 2016-08-11:
since there is not much response in general and not very usefull feedback maybe close this one?

closed 2016-08-11 by thinkyhead