MarlinFirmware/Marlin #4485: 1.1.0-RC7 ignores maximum feed rates

warriordog, 2016-07-31:
In 1.1.0-RC7, any movement seems to be able to go at any speed irregardless of the maximum feed rate.  The feed rates show correctly with M503, and they load properly from both Configuration.h and M203.  But any G0/1 commands will move at their specified F speed or the last movement speed even if it is far greater than the max feed rate for that axis.  This is an issue when trying to print with Slic3r, which always assumes that the printer will limit Z movements to a safe speed.  By default it moves the Z-axis at as much as 7800 mm  / m, which is way too fast for most setups.  Normally Marlin would limit this to 5 mm / s, but in RC7 it seems to just blindly obey the specified speed.

This is on a RepRap Mendel 2, with a Melzi 2 (sanguino 1.1 compatible) board.

thinkyhead, 2016-08-01:
Please try `RCBugFix`. The issue you are reporting should now be fixed, thanks to #4450.

warriordog, 2016-08-01:
@thinkyhead 

This no longer occurs in RCBugFix, thanks for the tip!

closed 2016-08-01 by warriordog

sanyi, 2018-01-02:
This issue is still present in Marlin-bugfix-1.1.x. A command like ```G0 F4800 X87.008 Y87.008 Z0.200``` will not respect Z federate and will cause the motor to skip.