MarlinFirmware/Marlin #4488: Compile errors when using VIKI 2

mperdue, 2016-08-01:
When I try to enable the use of my VIKI 2 LCD I get a compiler error in ultralcd.cpp at line 2695

'u8g' was not declared in this scope

I've checked my configuration.h file and I think I have everything set as it was in the file I used in RC6. Any suggestions would be appreciated. It builds and runs without the VIKI 2.

Cartesian system with an Azteeg X3 Pro controller.

thinkyhead, 2016-08-01:
Are you seeing this with the current `RCBugFix` also? Double-check to make sure VIKI2 is the only display that you have enabled, also.

mperdue, 2016-08-01:
I just tried it with the current RCBugFix and got the same results. I'l looked through the configuration.h file and don't see any other LCD enabled.

mperdue, 2016-08-01:
A bit more information. I've tried building with a couple of the other LCD devices enabled instead of the VIKI2 and that seems to work OK. With VIKI2 or miniVIKI I get the error

mperdue, 2016-08-02:
Here are the error messages...

Arduino: 1.6.9 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch/ultralcd_impl_DOGM.h:187:0,
                 from sketch/ultralcd.cpp:47:
sketch/utf_mapper.h: In function 'char charset_mapper(char)':
utf_mapper.h:29: error: 'u8g' was not declared in this scope
   #define HARDWARE_CHAR_OUT u8g.print
                             ^
sketch/utf_mapper.h:352:5: note: in expansion of macro 'HARDWARE_CHAR_OUT'
     HARDWARE_CHAR_OUT( c );
     ^
In file included from sketch/ultralcd.cpp:47:0:
sketch/ultralcd_impl_DOGM.h: In function 'void lcd_setFont(char)':
ultralcd_impl_DOGM.h:194: error: 'u8g' was not declared in this scope
     case FONT_STATUSMENU : {u8g.setFont(FONT_STATUSMENU_NAME); currentfont = FONT_STATUSMENU;}; break;
                             ^
ultralcd_impl_DOGM.h:195: error: 'u8g' was not declared in this scope
     case FONT_MENU       : {u8g.setFont(FONT_MENU_NAME); currentfont = FONT_MENU;}; break;
                             ^
ultralcd_impl_DOGM.h:196: error: 'u8g' was not declared in this scope
     case FONT_SPECIAL    : {u8g.setFont(FONT_SPECIAL_NAME); currentfont = FONT_SPECIAL;}; break;
                             ^
ultralcd_impl_DOGM.h:197: error: 'u8g' was not declared in this scope
     case FONT_MENU_EDIT  : {u8g.setFont(FONT_MENU_EDIT_NAME); currentfont = FONT_MENU_EDIT;}; break;
                             ^
sketch/ultralcd_impl_DOGM.h: In function 'char lcd_print(char)':
ultralcd_impl_DOGM.h:204: error: 'u8g' was not declared in this scope
     u8g.setFont(FONT_SPECIAL_NAME);
     ^
sketch/ultralcd_impl_DOGM.h: In function 'void lcd_implementation_init()':
ultralcd_impl_DOGM.h:247: error: 'u8g' was not declared in this scope
     u8g.setContrast(lcd_contrast);
     ^
sketch/ultralcd_impl_DOGM.h: In function 'void lcd_kill_screen()':
ultralcd_impl_DOGM.h:306: error: 'u8g' was not declared in this scope
   u8g.setPrintPos(0, u8g.getHeight()/4*1);
   ^
sketch/ultralcd_impl_DOGM.h: In function 'void _draw_centered_temp(int, int, int)':
ultralcd_impl_DOGM.h:318: error: 'u8g' was not declared in this scope
   u8g.setPrintPos(x - (18 - degsize) / 2, y); // move left if shorter
   ^
sketch/ultralcd_impl_DOGM.h: In function 'void _draw_heater_status(int, int)':
ultralcd_impl_DOGM.h:337: error: 'u8g' was not declared in this scope
     u8g.setColorIndex(0); // white on black
     ^
ultralcd_impl_DOGM.h:342: error: 'u8g' was not declared in this scope
     u8g.drawBox(x + h, y, 2, 2);
     ^
sketch/ultralcd_impl_DOGM.h: In function 'void lcd_implementation_status_screen()':
ultralcd_impl_DOGM.h:364: error: 'u8g' was not declared in this scope
   u8g.setColorIndex(1); // black on white
   ^
In file included from sketch/ultralcd.cpp:47:0:
sketch/ultralcd_impl_DOGM.h: In function 'void lcd_implementation_mark_as_selected(uint8_t, bool)':
ultralcd_impl_DOGM.h:487: error: 'u8g' was not declared in this scope
     u8g.setColorIndex(1);  // black on white
     ^
ultralcd_impl_DOGM.h:492: error: 'u8g' was not declared in this scope
     u8g.setColorIndex(1); // unmarked text is black on white
     ^
ultralcd_impl_DOGM.h:494: error: 'u8g' was not declared in this scope
   u8g.setPrintPos((START_COL) * (DOG_CHAR_WIDTH), (row + 1) * (DOG_CHAR_HEIGHT));
   ^
sketch/ultralcd_impl_DOGM.h: In function 'void lcd_implementation_drawmenu_generic(bool, uint8_t, const char*, char, char)':
ultralcd_impl_DOGM.h:536: error: 'u8g' was not declared in this scope
   u8g.setPrintPos(LCD_PIXEL_WIDTH - (DOG_CHAR_WIDTH), (row + 1) * (DOG_CHAR_HEIGHT));
   ^
sketch/ultralcd_impl_DOGM.h: In function 'void _drawmenu_setting_edit_generic(bool, uint8_t, const char*, const char*, bool)':
ultralcd_impl_DOGM.h:554: error: 'u8g' was not declared in this scope
   u8g.setPrintPos(LCD_PIXEL_WIDTH - (DOG_CHAR_WIDTH) * vallen, (row + 1) * (DOG_CHAR_HEIGHT));
   ^
sketch/ultralcd_impl_DOGM.h: In function 'void lcd_implementation_drawedit(const char*, const char*)':
ultralcd_impl_DOGM.h:602: error: 'u8g' was not declared in this scope
   float rowHeight = u8g.getHeight() / (rows + 1); // 1/(rows+1) = 1/2 or 1/3
                     ^
sketch/ultralcd.cpp: In function 'void kill_screen(const char*)':
ultralcd.cpp:554: error: 'u8g' was not declared in this scope
     u8g.firstPage();
     ^
sketch/ultralcd.cpp: In function 'void lcd_update()':
ultralcd.cpp:2694: error: 'u8g' was not declared in this scope
         u8g.firstPage(); 
         ^
sketch/ultralcd.cpp: In function 'void set_lcd_contrast(int)':
ultralcd.cpp:2800: error: 'u8g' was not declared in this scope
     u8g.setContrast(lcd_contrast);
     ^
exit status 1
'u8g' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Blue-Marlin, 2016-08-02:
Are you sure you have installed u8glib?

mperdue, 2016-08-02:
Yes, I'm sure. It works great with RC6

thinkyhead, 2016-08-02:
I believe #4506 will fix the issue. It will be merged shortly.

thinkyhead, 2016-08-02:
@mperdue Could you post your `Configuration.h` file? I just enabled `VIKI2` with the default configuration and it compiles without any error for me.

mperdue, 2016-08-02:
I'll post it in a few. I figure there must be an error on my part.

mperdue, 2016-08-03:
OK, I downloaded everything a third time and added my changes one at a time with a build for each change. Third time seems to be a charm. Thanks and sorry for the false alarm.

closed 2016-08-03 by mperdue