MarlinFirmware/Marlin #4489: LCD not responding

checkerz4456, 2016-08-01:
The firmwear compiles and uploads just fine, just my LCD is only lit up, and does not show and characters. I am using a Reprap discount smart controller with 20x4 character spaces and SD support. I have all of that commented in config.h, as well as ULTRA LCD because I do not have a full graphics display. The is all plugged into my Arduino Mega 2650/RAMPS 1.4 EFB What am I doing wrong?

thinkyhead, 2016-08-01:
Can you check to see if the problem exists in the [current `RCBugFix` branch](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix)? I suspect the problem does still exist, as this is not an issue I have seen before. There have been some changes to low-level LCD support since RC6, for sure, but mostly related to graphical displays.

zenmetsu, 2016-08-13:
I also cannot get a 20x4 display to work with RCBugFix.  I'm using a VFD (cu20025ecpb-w1j) which is a drop-in replacement for an LCD.  It works fine with other Arduino code using the LiquidCrystal library.  

Do we just have to plug in the display and uncomment `#define ULTRA_LCD`?  I went as far as to define LCD_WIDTH and LCD_HEIGHT and still got no joy.  I was planning on extending the code to allow for software control of the brightness, one of the benefits of using a VFD in addition to the awesome contrast.

zenmetsu, 2016-08-13:
Just tried with a standard LCD, same issue.  No response at all when unit is powered up.

zenmetsu, 2016-08-13:
My bad, I had read suggestion to enable the reprapworld graphical LCD and that was keeping the display from working.  Tested in `RC7` and `RCBugFix` without issue.

[https://imgur.com/y6RPyuq.jpg]

thinkyhead, 2016-08-14:
I'm glad to hear you worked it out. Note that `ULTRA_LCD` is automatically enabled for all character-based displays, so if you use one of the named displays there's no need to also enable `ULTRA_LCD`.

closed 2016-08-14 by thinkyhead