MarlinFirmware/Marlin #4492: Delta Printer Bed leveling

cblomart, 2016-08-01:
Hello,

I hope someone can help me...

I try to configure bed leveling on my delta printer (kossel mini).
I use a capacitive probe fixed to the effector.
The z probe triggers arround 1 mm above the bed (can be ajusted a bit).

So as said the z probe seems properly recognized and triggers when close to the bed (otherwize "open").

When issuing G29 the auto bed leveling begins... the printing head hits the bed serveral times as well as doesn't touch it several times... On "repetier" logs i see a bunch of time the same number that seems to be the probe to extruder z offset plus 1cm (no variation).

Afterwards the coordinates of the printer is all screwed Z0 doesn't touch the surface and X0 Y0 is not in the center...

Don't know what to do :-( ...

Would realy appreciate any help..

PS: cant post the Configuration.h nor the repetier logs right now but no problem if necessary.

Blue-Marlin, 2016-08-01:
We need to now what brach you are using. If RCBugFix we need the date.
Please append your Configuratin.h renamed to *.txt.
A at least approximately calibrated printer may help.
The last change related to probing was making optional the double touch. Ether try to reactivate double touch (PROBE_DOUBLE_TOUCH) or try to increase `Z_HOME_BUMP_MM`.

Roxy-3D, 2016-08-01:
> Don't know what to do :-( ...

Delta's are very ornery.   Getting them configured well is difficult.   Here is another discussion that you might want read and jump into:   https://github.com/MarlinFirmware/Marlin/issues/2159#issuecomment-236457036

cblomart, 2016-08-01:
As i downloaded the zip from release, i would suppose the branch is the RC one.

Here is the [Configuration](https://gist.github.com/cblomart/f908a94e525c8ec015cb66f93924dba0).

I increased the Z_HOME_BUMP as well as X and Y as the config seems to say that for the deltas the value should be the same for all axes.

This said... i can print correctly, but on large surface i do have isses as only the center of the bed seem level. On the border i either am over or bellow... i currently correct this by adjusting X, Y and Z max endstops... It takes a few iterations :( I tough i could get rid of this tedious calibration by having ALB.

I dont find the PROBE_DOUBLE_TOUCH variable...

rogueqd, 2016-08-03:
It sounds like your Delta Radius isn't flat. Have you adjusted DELTA_SMOOTH_ROD_OFFSET?

thinkyhead, 2016-08-03:
For testing, please use the current `RCBugFix` code so we're on the same page. Then you can try the `PROBE_DOUBLE_TOUCH` option, which is only recently added. I also recommend that you enable `DEBUG_LEVELING_FEATURE` and issue the `M111 S255` command to enable the most detailed logging possible. We've made a lot of changes to bed leveling in recent weeks, so I'm sure there are more areas where we can fix things.

cblomart, 2016-08-05:
Ok i tested on RCBugFix... 
Good news: it does double touch.
Bad news: Leveling doesn't work

I need to stop the printer because now it goes out of the print area when leveling...
I have the [repetier logs](https://gist.github.com/cblomart/f707340c2b430503e77092e8fe896056) as well as the new [Configuration.h](https://gist.github.com/cblomart/78899f3bab12ebdab36792f06ec32dd6)

thinkyhead, 2016-08-07:
I have a delta to mess with. It doesn't extrude yet, but I can use it to test leveling!

aljscott, 2016-08-07:
#4520 I've having a similar issue with double touch. Its going out of the print area when levelling.
Only when its turned on.

cblomart, 2016-08-09:
I removed double touch and it works better.
Everything stays inside the print area.
The printing head doesn't hit the bed aggressively all the time. More a gentle touch altought it doesnt touch the bed everytime: on some part of the bed it just stops ahead of the bed and goes to the following point. 
Leveling naturaly doesn't work, in the center it seems ok but on the "Y" edge traveling at 1mm up, the head presses the bed...

thinkyhead, 2016-08-09:
I didn't get much chance to play with delta leveling today, as I was working on the calibration. Hopefully tomorrow I will be able to start debugging the probing behavior.

boelle, 2016-08-14:
@thinkyhead any news?

thinkyhead, 2016-08-19:
I have seen no problems with leveling on my delta. Also there have been some changes to the code, so perhaps this issue is gone. In any case, @cblomart should try `RCBugFix` and see where it stands today.

cblomart, 2016-08-19:
As promissed i will test with the aluminium bed asap with an updated RCBugFix.
Did you test with doubletouch @thinkyhead ? As it seems that is what caused the head to go off printing volume. Without it auto bed leveling seems to work (i need a better zprobe reactivity)

thinkyhead, 2016-08-28:
Be sure to re-download `RCBugFix` and test again soon, as there have been a few tweaks in this area. And if you still see issues, please test one or both of my branches, which are attempting to produce the most accurate Z position at the end of leveling.

This one adjusts Z based on the homing position:
https://github.com/thinkyhead/Marlin/tree/rc_fix_leveling_maths

And this one adjusts Z based on the final probe position:
https://github.com/thinkyhead/Marlin/tree/rc_final_z_correction

boelle, 2016-09-03:
@cblomart have you retested?

boelle, 2016-09-20:
@thinkyhead @jbrazio @Roxy-3D think we can close this one due to inactivity?

closed 2016-10-26 by thinkyhead