MarlinFirmware/Marlin #4494: Mesh bed leveling problem

Dreyer1416, 2016-08-01:
Hi,

I am having some problems with the mesh leveling. Here is the problem:
I use a Z max endstop and have a `Z_MAX_POS 47` in `configuration.h`. However when I input a` G29 S1` command and the printer homes itself, the Z height given by a`M114` command is equal to `1` which is what I have in `MESH_HOME_SEARCH_Z`. 

Am i am doing something wrong?

Marlin Firmware version: 1.1.0-RC6

Roxy-3D, 2016-08-01:
You only have 47mm above the print bed?    Do you have a Z-Probe to measure the position of the bed?

Dreyer1416, 2016-08-01:
Yes, only 47mm. I do have a Z-probe but I deactivated it since that according to the wiki* there is no automatic way to probe the bed. So I was employing the manual method, using a piece of paper to attest when the nozzle touched the printer bed.

*https://github.com/MarlinFirmware/Marlin/wiki/Mesh-Bed-Leveling)

epatel, 2016-08-01:
This commit addressed max Z endstop with MBL. Could you make sure you have this in the version you are trying? 

https://github.com/MarlinFirmware/Marlin/pull/4146

Roxy-3D, 2016-08-01:
Well, the whole purpose of the Z-Probe is to measure how high the nozzle is above the bed at a given coordinate.  If you are willing to try a 'Development' branch, the Mesh Bed Leveling concept has mostly been automated (to use a Z-Probe).  That branch has the tools in place to very quickly and accurately tune your Mesh.    

And if you re-read that first paragraph of the Wiki, it has changed.

@epatel   You will have a smile on your face when you get the time to bring up: https://github.com/MarlinFirmware/Marlin/tree/devel-ubl

epatel, 2016-08-01:
See this comment https://github.com/MarlinFirmware/Marlin/issues/4088#issuecomment-228568198

Dreyer1416, 2016-08-01:
I was not using the updated commit by @epatel . I am getting it now. Lets see if it works.

Regarding the UBL branch, it says in its documentation that I need to have a LCD panel, which sadly I don't. I can have a look at the code to see if I can get it to work without one though. I will update this thread with any news. Thanks

Roxy-3D, 2016-08-02:
> I can have a look at the code to see if I can get it to work without one though. 

Don't even try!   The system was designed to be able to be used without an LCD Panel.  But right now, the code to do that isn't there.  It won't work.    Soon (in 1 or 2 months) it will work without an LCD Panel.     My suggestion is if you want to use UBL is spend $10 and order this:   http://www.ebay.com/itm/251242416836?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT    Otherwise, in a 6 weeks or so, you can do it without an LCD Panel.

closed 2017-10-31 by thinkyhead