MarlinFirmware/Marlin #4495: too few arguments to function 'void _lcd_move_e(int8_t)'

fabtopia, 2016-08-01:
With the following settings in configuration.h

// This defines the number of extruders
// :[1,2,3,4]
#define EXTRUDERS 2

// For Cyclops or any "multi-extruder" that shares a single nozzle.
#define SINGLENOZZLE

I get an error "too few arguments to function 'void _lcd_move_e(int8_t)'"

changing it to 
// This defines the number of extruders
// :[1,2,3,4]
#define EXTRUDERS 1

// For Cyclops or any "multi-extruder" that shares a single nozzle.
//#define SINGLENOZZLE

there is no problem.

esenapaj, 2016-08-01:
Probably PR #4491 (Use a default argument in _lcd_move_e) will fix it.

lavato, 2016-08-01:
> Probably PR #4491 (Use a default argument in _lcd_move_e) will fix it.

for me, it did not.

closed 2016-08-01 by thinkyhead

fabtopia, 2016-08-02:
After the last RCBugFix with #4491 there is at the repaire menu the move "extruder," "extruder 1" and "extruder 2". I think there is one to many ("Extruder")

esenapaj, 2016-08-04:
>there is at the repaire menu the move "extruder," "extruder 1" and "extruder 2".

It seems that it's intentional.
https://github.com/MarlinFirmware/Marlin/pull/4491#discussion_r73245220

fabtopia, 2016-08-04:
Oke, I get the idea of @thinkyhead behind it, but then the text "Extruder" could better be changed in "Extruder Actief" or "Extruder Current" of something like that.