MarlinFirmware/Marlin #4498: Long Delay every Layer Change after RC7

manianac, 2016-08-02:
I've just upgraded to RC7, and now RCBugFix, and I find my printer pausing every layer change for about 30 - 40 seconds.

It is a Cartesian printer with a RUMBA board.  No Z probe or anything unusual.  Before I start digging in, is there anywhere specific I should start looking or placing logging commands?

Blue-Marlin, 2016-08-02:
```
G0 X50 Y50 Z5
G0 X80 Y80 Z5
G0 X80 Y80 Z5.2
G0 X50 Y50 Z5.2
G0 X50 Y50 Z5.4
G0 X80 Y80 Z5.4
G0 X80 Y80 Z5.6
G0 X50 Y50 Z5.6
```
Not a long g-code program, with extremely small layers. Runs completely unremarkable.

manianac, 2016-08-02:
That gcode runs without issue.  I did find my gcode runs fine without the heaters on, but once the extruder temp is above 170degC, it locks on every layer.  Here's the gcode that currently is failing.

`G28
G92 E0
; layer 2, Z = 0.6
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z0.600 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 3, Z = 0.9
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z0.900 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 4, Z = 1.2
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z1.200 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 5, Z = 1.5
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z1.500 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 6, Z = 1.8
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z1.800 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 7, Z = 2.1
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z2.100 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 8, Z = 2.4
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z2.400 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 9, Z = 2.7
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z2.700 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 10, Z = 3
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z3.000 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 11, Z = 3.3
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z3.300 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 12, Z = 3.6
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z3.600 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 13, Z = 3.9
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z3.900 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 14, Z = 4.2
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z4.200 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 15, Z = 4.5
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z4.500 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 16, Z = 4.8
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z4.800 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 17, Z = 5.1
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z5.100 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 18, Z = 5.4
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z5.400 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 19, Z = 5.7
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z5.700 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10
; layer 20, Z = 6
; outer perimeter
G1 X142.260 Y144.760 F15000
G1 Z6.000 F120
G11
G92 E0
G1 X122.740 Y144.760 E2.8108 F540
G1 X122.740 Y125.240 E5.6216
G1 X142.260 Y125.240 E8.4323
G1 X142.260 Y144.760 E11.2431
G92 E0
G10`

Blue-Marlin, 2016-08-02:
What are your M208 settings?

manianac, 2016-08-02:
`
Recv: echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
Recv: echo:  M207 S2.00 F900.00 Z0.00
Recv: echo:Recover: S=Extra length (mm) F:Speed (mm/m)
Recv: echo:  M208 S0.00 F660.00
Recv: echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveries
Recv: echo:  M209 S0
`

Blue-Marlin, 2016-08-02:
Hmmm.
What happens if you remove the G10/G11 statements?

manianac, 2016-08-02:
I will check tomorrow. Were there any changes to firmware retraction going into rc7?

Blue-Marlin, 2016-08-02:
As far as i know, no direct changes to firmware retraction, but extruder handling in general.
If the problem does or not occur without firmware retract we have narrowed down the area to search.
Up to now pure moves seem to be unsuspisious, but if its not the firmware retraction we probably have to search there, or in G92 (that's very unlikely).

margaale, 2016-08-02:
I'm having the exact same problem plus it takes around a minute to start extruding. Had to roll back to RC6, I have firmware retraction enabled and I use slic3r. My printer is a p3stell (prusa i3 like) with RAMPS.
If i have tonight i'll flash RC7 again with firmware retraction disabled to see if the error is still there. If there is something else i can do to help i'll be glad to.

manianac, 2016-08-02:
I've swapped out numerous commands, and have confirmed the problem is `G11` gcode.  Everything else is working fine currently.

Blue-Marlin, 2016-08-02:
Comfirmed. As soon i #define FWRETRACT i see, not pauses, but extremely slow unretracts.

margaale, 2016-08-02:
Great, if a patch is submited give me a heads up if you need some more testing. Thanks

manianac, 2016-08-02:
I may not have noticed the slow retracts, but thank you

Blue-Marlin, 2016-08-02:
Got it.
```diff
@@ -2518,11 +2518,11 @@ static void homeaxis(AxisEnum axis) {
       if (retract_zlift > 0.01) {
         current_position[Z_AXIS] += retract_zlift;
         SYNC_PLAN_POSITION_KINEMATIC();
       }
 
-      feedrate_mm_m = MMM_TO_MMS(retract_recover_feedrate_mm_s);
+      feedrate_mm_m = MMS_TO_MMM(retract_recover_feedrate_mm_s);
       float move_e = swapping ? retract_length_swap + retract_recover_length_swap : retract_length + retract_recover_length;
       current_position[E_AXIS] -= move_e / volumetric_multiplier[active_extruder];
       sync_plan_position_e();
       prepare_move_to_destination();
     }
```

manianac, 2016-08-02:
Just recompiled, confirmed that did the trick

closed 2016-08-02 by manianac