MarlinFirmware/Marlin #4499: RC7 flow rate adjustment causes zero extrusion

toby4548, 2016-08-02:
o-lampe reported this issue on RC1 before.

When I set  flow rate (any value below 100% ) the extrusion stops completely. Only retraction moves are done.

My config: Kossel mini/Ramps/Reprap discount smart dispaly.

Blue-Marlin, 2016-08-02:
Just to garantee this calculation is done inm float you could try:
```diff
@@ -806,11 +807,11 @@ void Planner::check_axes_activity() {
     float delta_mm[4];
     delta_mm[X_AXIS] = dx * steps_to_mm[X_AXIS];
     delta_mm[Y_AXIS] = dy * steps_to_mm[Y_AXIS];
     delta_mm[Z_AXIS] = dz * steps_to_mm[Z_AXIS];
   #endif
-  delta_mm[E_AXIS] = 0.01 * (de * steps_to_mm[E_AXIS]) * volumetric_multiplier[extruder] * extruder_multiplier[extruder];
+  delta_mm[E_AXIS] = 0.01 * (de * steps_to_mm[E_AXIS]) * volumetric_multiplier[extruder] * (float)extruder_multiplier[extruder];
 
   if (block->steps[X_AXIS] <= dropsegments && block->steps[Y_AXIS] <= dropsegments && block->steps[Z_AXIS] <= dropsegments) {
     block->millimeters = fabs(delta_mm[E_AXIS]);
   }
   else {
```
Maybe related to #4498

toby4548, 2016-08-02:
Thanks for your help : )
I tried it, but it's still not working.
The firmware is able to read the integer value from host program or LCD controler(eg. 101,102...), but any value lower than 100 (eg. 99.98...) will cause this problem.

thinkyhead, 2016-08-02:
> Just to guarantee this calculation is done in float

Including a `float` (0.01) as any part of the calculation causes all values to be coerced to `float`.

thinkyhead, 2016-08-02:
@toby4548 Are you referring to the "Flow Rate" as applies to the E axis only, and buried in the LCD menus, or the "Feed Rate" as applies to all movement, as adjusted from the Info Screen?

Blue-Marlin, 2016-08-02:
@thinkyhead 
Off corse that is the way it should work. But
a we changed that line recently.
b dividing 99/100 versus 100/100 in integer could explain the problem.
So, just to be sure, i asked for an experiment.

thinkyhead, 2016-08-02:
> dividing 99/100 versus 100/100 in integer could explain the problem

Yes, it would. But the code is clearly doing float calculation, so you're implying a compiler bug.

Blue-Marlin, 2016-08-02:
Yes. This experiment was ought to exclude a possible compiler bug at this particular place as the source of the problem. I admit this would have been a very lucky shot.

Blue-Marlin, 2016-08-03:
I'm absolutely unable to reproduce the problem.
Please be sure to use the newest RCBugFix. Please provide your Configuration.h. Please provide a sequence to reproduce the problem.

toby4548, 2016-08-03:
Hello,
thank you for helping me.
After disassemble the printer, which from my friend, I found that he use some costomized board from a Chinese company called Micromake. 
[https://goo.gl/yRov6S](url)
The company claim that it's compatible with RAMP 1.4, but I still think that this problem is because of this board.
Here is my Configuration file
[https://www.dropbox.com/s/1fzlwdo5qtmxnwg/Configuration.h?dl=0](url)

thinkyhead, 2016-08-07:
@toby4548 I just got a Delta printer from Micromake and have this same board. So far it seems to be 100% compatible with RAMPS pins. I believe it's actually an MKS board with a Micromake silkscreen.

boelle, 2016-08-11:
@toby4548 "Please provide a sequence to reproduce the problem."

thinkyhead, 2016-08-11:
@toby4548 If the printer is still disassembled, then no rush. As mentioned, try `RCBugFix` because we're patching regularly.

thinkyhead, 2016-08-11:
This probably won't make a bit of difference, but just for fun, modify this part of `planner.cpp` and see if it makes any difference.
```cpp
-  block->steps[E_AXIS] = labs(de);
-  block->steps[E_AXIS] *= volumetric_multiplier[extruder];
-  block->steps[E_AXIS] *= extruder_multiplier[extruder];
-  block->steps[E_AXIS] /= 100;
+  block->steps[E_AXIS] = (labs(de) * volumetric_multiplier[extruder] * extruder_multiplier[extruder]) / 100L;
```

toby4548, 2016-08-12:
@thinkyhead Thanks for helping
I've tried RCBugFix but it has the same problem as RC7 on my printer.
I've also tried to modify the code in planner.cpp and it will just make the extruder stop all the time.

thinkyhead, 2016-08-19:
> dividing 99/100 versus 100/100 in integer could explain the problem

My earlier comment about `float` coercion might be jumping the gun. I ran into something interesting recently, which was if I did this…
```cpp
long mylong = labs(myfloat * 10000);
```
…the result was different than if I did this…
```cpp
long mylong = labs(myfloat * 10000.0);
```
So @toby4548 try this instead. Make the change as above, but use this line as replacement:
```cpp
block->steps[E_AXIS] = labs(de) * volumetric_multiplier[extruder] * extruder_multiplier[extruder] * 0.1;
```

thinkyhead, 2016-08-21:
Seeing any different result in today's `RCBugFix`? There continue to be changes almost every day.

Roxy-3D, 2016-08-21:
```cpp
long mylong = labs(myfloat * 10000);
```
…the result was different than if I did this…
```cpp
long mylong = labs(myfloat * 10000.0);
```
That doesn't seem right.  Isn't the integer supposed to converted to a float to continue the calculation?   I've noticed some unexpected behavior and compensated by going totally paranoid:
```cpp
  #define MESH_X_DIST ((float) ((((float) MESH_MAX_X)-((float) MESH_MIN_X)) / (((float) MESH_NUM_X_POINTS)-1.0)))
  #define MESH_Y_DIST ((float) ((((float) MESH_MAX_Y)-((float) MESH_MIN_Y)) / (((float) MESH_NUM_Y_POINTS)-1.0)))
```
Do we have some option turned on in the GCC compile or something else to cause this unexpected behavior?

thinkyhead, 2016-08-22:
> Isn't the integer (in the first example) supposed to be converted to a float to continue the calculation?

Yes, absolutely. But evidently 10000, when converted to `float` by the 8-bit AVR, ends up being stored as 9999.99999 or something similar.

boelle, 2016-08-28:
@thinkyhead i think we can close this one due to no response...

Roxy-3D, 2016-08-28:
> Yes, absolutely. But evidently 10000, when converted to float by the 8-bit AVR, ends up being stored as 9999.99999 or something similar.

With 4-Byte floats, it might be something more like 9999.99    We only get about 6 digits of precision.

>> I've noticed some unexpected behavior and compensated by going totally paranoid
> I would so much prefer to see the hard data in cases like this, rather than just the anecdote.

I don't have hard data and an example of where my problem is coming from yet.   But shifting to the 'paranoid' declaration of #define MESH_X_DIST  and #define MESH_Y_DIST did fix things and let me continue.    I expect I will run into problems again soon, and I'll provide more detail then.   The reason is I'm going to be shifting the code to allow MESH_X_MIN MESH_Y_MIN to be negative.

> i think we can close this one due to no response...

I must have missed the question.   But I'm obviously trying to be responsive.

thinkyhead, 2016-08-29:
Jo Prusa mentioned today that his fork has been rewritten to use fixed-point in performant areas. I presume it's using 16 bits for the whole part and 16 for the fractional part. That's enough for most purposes, with up to 65535mm length, plus a resolution down to 1/65535 of a mm. If the E axis needed to run continually for more than 65 meters we could deal with that. Of course most slicers reset E to zero on every layer, and sometimes on every line segment.

boelle, 2016-08-29:
> I must have missed the request. But I'm obviously trying to be responsive.

i was thinking more on @toby4548

thinkyhead, 2016-08-30:
I'm going to do a scrub-through of configured values and mixed integer-float maths throughout the three big ones: main, planner, and stepper, and make sure that everything that needs to be interpreted as float is explicitly cast as float. This should remove any anomalies, of which this issue may be only one instance.

boelle, 2016-09-03:
@thinkyhead @jbrazio @Roxy-3D what do you think?

23 days since @toby4548 last replied... think its time to close this one

closed 2016-09-03 by Roxy-3D