MarlinFirmware/Marlin #4507: Z Probe Endstop Support for Rambo

adamfilip, 2016-08-02:
Currently Rambo does not have a Endstop available for the Z probe
I would like to be able to use the Z max endstop as a Z Probe endstop

Can this be set in the PINS Rambo area.

Roxy-3D, 2016-08-02:
ThinkyHead is probably the right one to answer this...  But I'll try to help just so you can get started on the problem.   The Pins files, pretty much, documents what the board has silk screened on it.   It is a little bit complicated to use both a Z-Min-Endstop and a Z-Probe.    

I think you want to comment out the Z_MAX_PIN definition in your Rambo Pins file.  And replace it with Z_MIN_PROBE_PIN being defined on the same pin number that Z_MAX_PIN used to be on.

Once that is done, you need to very carefully setup your Configuration.h  to enable: Z_MIN_PROBE_ENDSTOP.       

Do you really want to have Z-Min endstop for homing and the Z-Probe for probing?

adamfilip, 2016-08-03:
with the Z endstop on the probe. if the probe is up.. there is no endstop that would stop the gantry from plunging.. I just assumed that having it separate would be safer.

thinkyhead, 2016-08-03:
Once #4517 is merged you should be able to get the pin you want by using the following in your `Configuration.h`:
```cpp
#define Z_MIN_PROBE_ENDSTOP
#define Z_MIN_PROBE_PIN 30
```

adamfilip, 2016-08-03:
Great Thanks

dvseliteus, 2016-10-04:
@thinkyhead Marlin-1.1.0-RC7 I'm having trouble getting my FIX_MOUNTED_PROBE to Trigger using the M119 command in Pronterface?
[Configuration.h](http://pastebin.com/RTwDu9B4)
[RAMBO.h](http://pastebin.com/L4geMMES)
I have tried the suggestions above and about everything else I can think of.
The probe was working at some point during my trials of the RCBUGFIX Marlin Version.
Any suggestions on how I can further test?

thinkyhead, 2016-10-06:
@dvseliteus Endstop testing is quite straightforward. They're either open or closed, and this registers as either open or TRIGGERED. Thus if an endstop isn't triggering at all, you can try setting the `ENDSTOPPULLUP` for that particular endstop. If it's triggering in reverse, then you just set (or unset) the `ENDSTOP_INVERTING` for that endstop. Just test them one at a time until you isolate the cause of the failure.

dvseliteus, 2016-10-07:
@thinkyhead Thanks for this info it really helped me dial it in. Here is what ended up working for me.

// coarse Endstop Settings
//#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  #define ENDSTOPPULLUP_ZMAX
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  #define ENDSTOPPULLUP_ZMIN_PROBE
#endif

closed 2016-10-09 by thinkyhead