MarlinFirmware/Marlin #4508: LCD does see's sd files, SD dectect enabled

kjdkeller, 2016-08-02:
This is my 3rd Marlin install on this Cartesian printer the last was marlin rc6 bug fix.  Never had this issue.  All the lcd menus scroll up and down ok except the sd card menu.  I can see the files, I just cant scroll down to select them.  I am using the rep rap discount full graphic smart controller.

I also loaded this rc7 to a delta printer with a reprap discount smart controller and the sd menu works fine.

thinkyhead, 2016-08-03:
@kjdkeller I revamped the menu code today, and it's now merged into `RCBugFix`. Can you check to see if the SD files now scroll properly? The changes seem to have fixed all scrolling issues for me.

kjdkeller, 2016-08-03:
Hi, thanks for that.  I loaded the newest bugfix, it works now.  However, the SD menu scrolls slowly like its delayed compared to the other menus.

Blue-Marlin, 2016-08-04:
We can't buffer the complete directory. We do only store the index to a directory entry. For every directory entry we have to scan the the complete directory to this file or its processor. 
If you have only a few directory entrys it's much faster then if you have a lot. Formating your sd-card may help.

thinkyhead, 2016-08-04:
> We can't buffer the complete directory

This is true. I have experimented with caching the sort order in https://github.com/thinkyhead/Marlin/tree/sd_alpha_sort and it includes some options to cache the listing (if you happen to have enough extra RAM). But since it uses dynamic memory allocation, it cannot be included in release versions of Marlin.

boelle, 2016-08-11:
So this is not a bug as such? can we close this one?

thinkyhead, 2016-08-11:
@kjdkeller A "low-level format" of the SD card is known to have helped with read/write issues. Try that, or another SD card, and see if the speed improves.

kjdkeller, 2016-08-11:
The issue is resolved with the SD card scrolling. I think it's okay to close the issue.

closed 2016-08-12 by thinkyhead