MarlinFirmware/Marlin #4513: Sometimes Buzzer gets stuck

thinkyhead, 2016-08-03:
For testing I'm using `M428` from Pronterface console. This command plays two beep sounds. Sometimes there seems to be queue corruption, as it will play 2 incorrect tones (the same tones in reverse, it seems) followed by a high-pitched tone that doesn't stop. Playing an LCD click or sending another `M428` will stop the tone from playing.

In my testing, I found that adding `SERIAL_ECHO` commands to `Buzzer::off` and `Buzzer::reset` prevented the stuck beep from occurring. So I thought possibly it has to do with timing. But that wouldn't explain why it plays the wrong tones. So now I'm adding logging to `CircularQueue` to see if it's getting confused.

jbrazio, 2016-08-03:
Could it be `FastIO` messing something ? This could explain somehow the stuck tone but not the reverse order, for the reverse order could the serial ISR be breaking `CircularQueue`, in other words do we have to use `cli()` and `sei()` when handling the queue ?

thinkyhead, 2016-08-03:
@jbrazio When the problem occurs, it's neither a problem with the `CircularQueue` nor the `::tone` function call. Both of those seem to be working, with `::tone(f,d)` receiving the correct pitch and duration.

However, the pitch is clearly wrong when this occurs, and then the extra high-pitched tone at the end follows the last tone finishing. So the problem may be with this LCD specifically, or something in the say the tone is being applied.

I'm using the `SPEAKER` option, so the `::tone(f,d)` function does all the work. So, perhaps communication with the LCD is being interrupted during the call to `::tone()`.

thinkyhead, 2016-08-03:
And sure enough, this fixes it:
```cpp
cli();
::tone(BEEPER_PIN, this->state.tone.frequency, this->state.tone.duration);
sei();
```

jbrazio, 2016-08-03:
This is wierd, how about ?
```cpp
cli();
this->state.tone = this->buffer.dequeue();
sei();
```

thinkyhead, 2016-08-03:
That makes no difference. And as already stated, the correct `frequency` and `duration` are being given to the `::tone(f,d)` function. It's a hardware issue, apparently. It seems that bits need to be sent to the LCD with critical timing, and if this is interrupted it gets the wrong data and plays the wrong tones.

lavato, 2016-08-06:
I know this may not help but for me buzzer never worked better as of 5347f39

closed 2016-08-07 by thinkyhead

jbrazio, 2016-08-11:
@lavato Now it has full priority when beeping :-P