MarlinFirmware/Marlin #4518: RC7-RC6    muving  zerro  point   when printing on coreXY

PeetKos, 2016-08-03:
Hello!   
when  I print   on   RC5     all  work ok,  but I  need  new version to use some future.  

on the foto    from  left to  right    printed 1" cube         1- RC5 single perimetr 0-infill     2- RC7 single perimetr 0-ifill       3 -  RC7 single  20%-infill   

after  printed    i   use  G1 X0   comand  and see    that    10mm   gap  from  endstop

I  use hand made   printer   with  external 6600    drivers    NEMA23  motor,   on  RC5   all   work ok.
I try    to   lower  speed and acceleration  -     same result  
also i try  to   chenge     adv.config 
/ By default pololu step drivers require an active high signal. However, some high power drivers require an active low signal as step.
#define INVERT_X_STEP_PIN false
#define INVERT_Y_STEP_PIN false
 in result     zerro  point      move     in other   direction     (before  its move  to   X- Y+      and after   its move  to y+ only)

and   again   in  RC5   I have't   that problem  at all     only  in RC6  and RC7
  

[image]

PeetKos, 2016-08-03:
I'm   conneting external drivers  with   common cathode   configuration.

PeetKos, 2016-08-03:
Traying With  common anode  have same result ;(

thinkyhead, 2016-08-03:
So it looks like RC7 did the second cube correctly, but then the third cube is consistently shifting on every layer. Are you able to find some pattern or common factor that causes this shifting to occur?

Please post your `Configuration.h` so we can see all the options you have enabled.

PeetKos, 2016-08-04:
second  cube  incorrect to    but      shifting    not so big.     i  see    that  shifting bigger if   head have more  different dirrection movment    in   layer   ...        also   some time     shifting  rise with  high

yestoday  I  come back   to RC5     to   mechaincs  testing     and    see  that    cube   is correct  also  its correct   when  I duble  speed  and  acceleration movment  
[Marlin_conf.zip](https://github.com/MarlinFirmware/Marlin/files/400976/Marlin_conf.zip)

PeetKos, 2016-08-04:
its  coommon anode    configuration            and      some  test  
 #define INVERT_X_STEP_PIN false
#define INVERT_Y_STEP_PIN false
atributes         -        in  result  shfting      change dirrection      but still   persist  

if   you    say  i  can  post  working   RC5   configuration     to compare.

thinkyhead, 2016-08-04:
Your settings that work with RC5 should work with `RCBugFix`. (Only change the items the compiler complains about.)

To help to isolate the cause of the layer shifting issue, I will make some branches for you to test. I may also suggest some things you can try, to help us isolate the cause. I have a busy day tomorrow, but we can look at this more closely over the next few days.

PeetKos, 2016-08-04:
ok  I will wait for  You help

about  sttings RC5   -    yes   its look like same for movment sittings...    but  I  can't simple    copy  file   because other options   is  different   in     RC5  and RC7

thinkyhead, 2016-08-04:
> I can't simple copy file because other options is different in RC5 and RC7

Yes you can. The code in `SanityCheck.h` will tell you what to change. If it forgets to inform you about some setting, please let me know.

PeetKos, 2016-08-05:
no  i  just wait  for You pach    -    i  think    problem in   syncronnisation stepping both motors....  maybe  you  may look in difference in   stepping  algoritm   in  RC5  and  RC7

thinkyhead, 2016-08-07:
> i just wait for You pach

Might be a while I need to find more CoreXY testers to help diagnose issues.

PeetKos, 2016-08-07:
maybe on  other drivers     all will be ok.    yestoday   I changed   driver that seems  to hot ...   I will test  soon.

PeetKos, 2016-08-07:
no   test  also  fail  -  RC5 -work whell     rc7-shift layers

thinkyhead, 2016-08-08:
@PeetKos Please post the GCode you're printing. I can see no difference between RC5 and RC6 that should be specifically affecting CoreXY. In particular, neither the stepper or planner code was altered very much. While the `st_set_position` function did gain CoreXY support, it is not usually something that gets called in the middle of a print job.

The fact that you get a normal looking print with no infill, but a slanted print with infill leads me to think there might be a slipping belt, lost steps, or other mechanical failure due to changing direction at high speed. To eliminate this possibility, try setting your `DEFAULT_XYJERK` to a lower value, such as 10, and see if it has any positive effect on printing with `RC7` or `RCBugFix`. Or, just try printing very slowly, and see if it still produces a slanted print or screwed-up X position.

PeetKos, 2016-08-08:
no no no      no infill    -   didn't look  good     ...   shifting  much less but  present  every layer   

belt  is mxl 10mm        spools  with lock on motor...       yesterday  I try  print cube on  RC5 with ACC3000  and   signed speed   200   -     no    shifting layer anyway   
gcode - I will send soon

PeetKos, 2016-08-08:
[1in Cube.zip](https://github.com/MarlinFirmware/Marlin/files/405986/1in.Cube.zip)
I didn't  see any  problem in  file  ...   on rc5    i use   same patameter ...    also  I print from USB and Flash  same result

PeetKos, 2016-08-08:
maybe  You need  video  of  test   fast  print on RC5   and  slow  on RC7     for  enshure  that RC5 didnt lost steps?    
I  didnt   use  rc_bugFix yet.

thinkyhead, 2016-08-08:
`RCBugFix` and `RC7` are similar. But you might as well try `RCBugFix` and see if it exhibits the same issue.

So far I still can find nothing differing between `RC5` and `RC6` that should produce such behavior. I wonder if you would "indulge me" and try building with a different version of Arduino IDE. I know it's a very long long-shot, but we might as well eliminate a quirk of the compiler right off.

PeetKos, 2016-08-09:
heh    first of all    sorry   for my  english.

yesterday  -  I have long day and   night for   testing (about 30-40  test start), and found posible bug .
 Im   use   single compiler under  Linux at all   - version i check  later. 
  now  long  step by  step   story   with foto  and  video
Im using  NEMA23 like motor  and 6600  or 6500 driver   at  1/8 microstep   and 2,2-2,5A current, 24volt 
1)  As  You say   I  put DEFAULT_XYJERK  to 4  and    start   printing   infilled  file  ...   its  vork   good about 10 minut       then     apear shifting :    
https://drive.google.com/open?id=0B3Kf1XzedkKxSEwwaGJ1XzlLR0k
2) I  put    DEFAULT_XYJERK  to 1      and     shifting apear   right after start    (in both prints printing speed about 30-60 mms(slow-first layer and perimetr? fast -  infill),  nonprinting speed about 180)
https://drive.google.com/open?id=0B3Kf1XzedkKxSjZRYk9hbHl3ZTA
3)  Then  I asume  that termal-speed    problem    and  lower  printing speed   to 10,  and acceleration  to  500  -  same result
4)  I'm go back  to   rc5    to   test   mechanics speed  and step loss  (250 -350mms)  ignoring  plastic overheating,   inertia trembling  of   direct  extruder  at corner ...      (if  step-loss   wall   will be shifting  )
1'' cube    cant   gain  that speed   during   acceleration     and  Im also print  circle  -    look at video:  

https://drive.google.com/open?id=0B3Kf1XzedkKxOV9ROWEtcjFRakU
https://drive.google.com/open?id=0B3Kf1XzedkKxQ1lyMi1zeU55Zzg
result  cube    have   stright  wall   at  middle  (no shifting and step loss)
circle  also good:
https://drive.google.com/open?id=0B3Kf1XzedkKxNWFOT3JJV2F3VU0

5)   After   that    im  go   to    rc7   -   and    test test  test       -   some   time  it work good on   speed  80     but     after   starting    step-loss -    it  lost step    on   any  printing speed   till  full reset of  arduino...    

6)   Tests test  -   and  found   -   that   if   global (include printing and non printing max speed)   limited   to  50mms      -    printing    normal    ...    if   60 or   higher  -   after some layer   starting step-loss
look  at     foto  wheree   I  game   with  speed  on the fly (no  infill) :
https://drive.google.com/open?id=0B3Kf1XzedkKxeGJXNkhKTTVMMnc
also   i  fround  that  acceleration     dont affect  on  this problem   (if  speed higer than 60  at any acceleration (100-3000) during some time  problem appered),    if  speed  50   -  3000  -   no  problem

7)     i  try   to  print    3 times     with  speed  50  and fast-honeycomb infill 10-30%  and   result normal : 
https://drive.google.com/open?id=0B3Kf1XzedkKxU1NfM0U1eGpMb28
https://drive.google.com/open?id=0B3Kf1XzedkKxRkxqNmh4NFk0a3c

8)    I lower microstep  to   1/4   and  try   -    working   speed   rices to   80 ...     bul   slow printing is  fear  becouse  resonance   problem   on big printer  (look with  sound):
https://drive.google.com/open?id=0B3Kf1XzedkKxWnJjN2dBZmc5RnM

9)   ok    I  go   buck   to   1/8    set  speed  higher   than  50  (60 or 80  not remember)  ..    and  try to found   problem  where     I     loos-step   on     single perimetr ,    and   found  that is in     at   end of layer   when head  do   small  buck-forward   movment  (or any other fast chenge direction - on other model)

10)   Ok    now    to    find   who   generate   wrong  step   Im  connect   Logic analizer   between  RAMPS and   external drivers 
https://drive.google.com/open?id=0B3Kf1XzedkKxekppMTBrWi1xNG8
(en,  step,  dir,  on   both driver)

now   lool   at    macro  view  on  corner     with  problem  buck forward  -   3 and   5  row   its  direction...    first   and last     its  steps ,   looking normal 
https://drive.google.com/open?id=0B3Kf1XzedkKxSDF5Z1dhRklkTW8

then    zoom  it   :
https://drive.google.com/open?id=0B3Kf1XzedkKxU0RkZDQ1d2c0SFE

You see  the    missed  step  signal   on  de-acceleration   ,  and  its  appear on both motor  -  its  mean that is  not electrical problem...    
(at  lower  speed  small  motor  with  small  rotor interia  can   do   this  stop-and-go  without  overstepping...    but  on bigger motor and bigger speed  motor   will do  uncounting step  over this lost signal)

    Ok    Now    look  ahead  to   the    next   corner on the    same layer on the   same   speed  (same log, some msec after)
https://drive.google.com/open?id=0B3Kf1XzedkKxNTJjM0ZnUnVJdlU

You    see     perfect    corner    (de-acceleration, change dirrection,  acceleration )

Thats     all      now    You  see  I  realy   need You    help  to  slow  that problem..    i  can  do  some  test  for You   and log it   on   analyzer.

PeetKos, 2016-08-10:
Hello !    any   information  for me ? (what version  compiler to test?)
yesterday  I have two print   on RC7  50speed -   one was good (1:20 hour)  second  fail   after 1:47 hour,
second  reprinted   on rc5 - normal

thinkyhead, 2016-08-11:
Sorry, I had a funeral to attend yesterday. I'll get back to this issue soon. Meanwhile, please move on to `RCBugFix` instead of `RC7`.

PeetKos, 2016-08-11:
Сondolences.          You mean that   You  something changed?  and   I can test   it now?

thinkyhead, 2016-08-11:
@PeetKos I mean: stop testing with `RC7` because our reference-point needs to be `RCBugFix`. I haven't specifically addressed this issue, because I have no idea of the cause. But we might correct it in the course of patching other things.

boelle, 2016-08-14:
@PeetKos Please try and test with RCBugFix NOT  RC7

PeetKos, 2016-08-15:
ok   will  try it  tomorrow

thinkyhead, 2016-08-23:
Be sure to tighten those stepper pulleys before testing. Looking forward to your results.

boelle, 2016-08-24:
@PeetKos what did the test say?

closed 2016-08-24 by thinkyhead