MarlinFirmware/Marlin #4519: Look ahead

Ram-Alsabti, 2016-08-03:
Hi,

I was wondering if anyone maybe has knowledge (and is willing to share it :) ) about look ahead. I am still studying planer code. I would like to modify/optimize movements depending on angle of print, that is why I think look ahead would help (to modify parameters in real time).

Will be happy for any help, thank you.
Ram

thinkyhead, 2016-08-03:
Marlin does lookahead planning in order to determine its acceleration behavior, allowing for more continuous movement. The functions to study are:

- `Planner::calculate_trapezoid_for_block`
- `Planner::recalculate`
  - `Planner::reverse_pass`
    - `Planner::reverse_pass_kernel`
  - `Planner::forward_pass`
    - `Planner::forward_pass_kernel`
  - `Planner::recalculate_trapezoids`

Ram-Alsabti, 2016-08-04:
Thank you for hints, I will look into it. Just to make sure, on pictures below that "curves" is because of look ahead function? I did make measurement (steps of motor with magnetic encoder) on on Y-axis and Extruder of regular 20mm cube. 1st pic is measurement on Extruder (you can see print cycle - 3 walls and infill), 2nd is Y-axis just infill part (you can see on picture 3 how it looks 5% infill geometry) and last 4th picture is Y- axis printing outer wall (before it stops it start making longer pauses... after stopping X-axis starts moving). I hope what i described is because of look ahead, to know if I am working in right direction, thank you.

[image]
[image]
[image]
[image]

Ram-Alsabti, 2016-08-04:
Specially on 4th picture I am not sure if it is look ahead or just "nature" of motor, although it is a stepper motor.

AnHardt, 2016-12-10:
Would be extremely helpful if the axes of the diagram wouldn't only have numbers but units.

I expect to see steps over time. Then we see a move with, more or less, constant speed, at the end decelerating, the stopping. That's neither a effect of look ahead, nor "nature" of a stepper motor, but nature of motion. The smallest movement we can make is a step. So the time has to vary when decelerating. The varying times during the constant speed phase are caused by the 'bresenham' algorithm. What we see is not the 'leading' axis. This wold be the only one what has a constant timing.

Issue can be closed - i think.

closed 2016-12-11 by thinkyhead

Ram-Alsabti, 2016-12-11:
Ok if this (picture) is not a look-ahead, then someone explain (without copy pasting what all are doing) what and where it is.

AnHardt, 2016-12-11:
In your model (https://cloud.githubusercontent.com/assets/6893581/17405913/aa509b20-5a61-11e6-8259-24522a98a81a.png) there is no combination of moves where look-ahed can have an effect. In every transition at least one axis has a full stop.

Imagine a subdivided circle. The angular difference between the segments becomes small when the number of segments is high. Now let's assume a low jerk, low acceleration, no speed limit and no look-ahead. 
For the first move, from standing still, the motion would accelerate up to the middle of the move, the deaccelerate to the end of the segment. All following moves would behave the same. 
With look-ahead it would look like:
Plan the firs segment like above. Begin to move. Plan the second segment. Calculate, using the angular difference between the segments, jerk and acceleration, the maximum speed for the transition between the moves. Alter the first move so that it ends at the junction speed. for that shift the point for the change from accelerating to deaccelerating further to the back. The maximum speed will increase a bit because the time for acceleration is increased. The second move now begins with the maximum junction speed, and accelerates up to the point where it needs to break, to reach a full stop at the end of the segment. If you now draw speed/way the two planed moves look mirrored at their junction. 
When planing the third move again at first the junction speed is determined. Now we can raise the speed for the end of move nr. 2. This shifts the acceleration-deacceleration-point to the middle of the second move. The third move is now looking like the second before.
[image]
When now the number of segments in our circle rises, the angular difference between the segments decreases, the junction speed rises, the middle moves top speed can be higher and higher. 
If we have a segmented straight line the junction speed is only limited by the over all maximum speed (what we defined as unlimited above). The speed is than only limited by the way we need for decelerating to zero, when all planed moves have been done.