MarlinFirmware/Marlin #4520: Delta Auto Bed Grid Leveling RCBugFix - weird coordinates

aljscott, 2016-08-03:
Hi,

I've been trying to configure a Delta printer with dual extruder using RC6 and RCBugFix.
Both are doing really weird things in terms of the auto bed level grid coordinates that it is probing.
If I execute G29 V4 and look at the coordinates being output to the console they appear to make sense.
Consistent with a grid covering the defined probe-able area.
However the X-Y positions the extruder moves to do not match the actual coordinates, the further it gets from the starting position the more it is out. In fact the LEFT and BACK area are not being covered and even through the output on the console is a perfect grid, the extruder is probing some pretty random points.
Its really confusing and changing the grid coordinates to compensate seems to sometimes send the extruder disappearing on an upwards trajectory where it starts probing the air.

The sensors is a IR probe and its not being erroneously triggered.
I assume the offset is from extruder one?
Settings:

#define X_PROBE_OFFSET_FROM_EXTRUDER -9.0     // Z probe to nozzle X offset: -left  +right
#define Y_PROBE_OFFSET_FROM_EXTRUDER 26.0     // Z probe to nozzle Y offset: -front +behind
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.0     // Z probe to nozzle Z offset: -below (always!)

    #define DELTA_PROBEABLE_RADIUS 80
    #define LEFT_PROBE_BED_POSITION -(DELTA_PROBEABLE_RADIUS)
    #define RIGHT_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS
    #define FRONT_PROBE_BED_POSITION - (DELTA_PROBEABLE_RADIUS-20)
    #define BACK_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS

aljscott, 2016-08-03:
```
`SENDING:G29 L-40  R40 B40 F-40 P10 V4
G29 Auto Bed Leveling
Bed X: -40.000 Y: -40.000 Z: -0.352
echo:busy: processing
Bed X: -20.000 Y: -40.000 Z: -0.610
Bed X: 0.000 Y: -40.000 Z: -0.668
echo:busy: processing
Bed X: 20.000 Y: -40.000 Z: -0.799
Bed X: 40.000 Y: -40.000 Z: -0.622
echo:busy: processing
Bed X: 40.000 Y: -20.000 Z: -0.822
Bed X: 20.000 Y: -20.000 Z: -0.918
Bed X: 0.000 Y: -20.000 Z: -0.838
echo:busy: processing
Bed X: -20.000 Y: -20.000 Z: -0.720
Bed X: -40.000 Y: -20.000 Z: -0.508
echo:busy: processing
Bed X: -40.000 Y: 0.000 Z: -0.630
Bed X: -20.000 Y: 0.000 Z: -0.758
Bed X: 0.000 Y: 0.000 Z: -0.899
echo:busy: processing
Bed X: 20.000 Y: 0.000 Z: -0.871
Bed X: 40.000 Y: 0.000 Z: -0.799
echo:busy: processing
Bed X: 40.000 Y: 20.000 Z: -1.008
Bed X: 20.000 Y: 20.000 Z: -0.988
echo:busy: processing
Bed X: 0.000 Y: 20.000 Z: -1.081
Bed X: -20.000 Y: 20.000 Z: -1.029
Bed X: -40.000 Y: 20.000 Z: -0.927
echo:busy: processing
Bed X: -40.000 Y: 40.000 Z: -1.223
Bed X: -20.000 Y: 40.000 Z: -1.367
echo:busy: processing
Bed X: 0.000 Y: 40.000 Z: -1.429
Bed X: 20.000 Y: 40.000 Z: -1.398
echo:busy: processing
Bed X: 40.000 Y: 40.000 Z: -1.218
-0.35 -0.61 -0.67 -0.80 -0.62
-0.51 -0.72 -0.84 -0.92 -0.82
-0.63 -0.76 -0.90 -0.87 -0.80
-0.93 -1.03 -1.08 -0.99 -1.01
-1.22 -1.37 -1.43 -1.40 -1.22
>>>G0 X0 Y0
SENDING:G0 X0 Y0
>>>G0 X40 `Y40``
```

thinkyhead, 2016-08-03:
We need to get feedback from more Delta owners to see what's happening here. I have a Delta under construction, but it will be a while before I can test with it.

@MarlinFirmware/testers-delta-team

superlou, 2016-08-03:
Can they algorithm be tested without automatic bed leveling on a delta?  Or does it require some sort of bed probe?

thinkyhead, 2016-08-03:
We don't currently have any hooks to do "faux" probing. So any test of `G29` currently requires some kind of probe.

Blue-Marlin, 2016-08-04:
> SENDING:G29 L-40  R40 B40 F-40 P10 V4

The strange part of tis is the __P10__.
P does not work for deltas
```
      int auto_bed_leveling_grid_points = AUTO_BED_LEVELING_GRID_POINTS;

      #if DISABLED(DELTA)
        if (code_seen('P')) auto_bed_leveling_grid_points = code_value_int();
        if (auto_bed_leveling_grid_points < 2) {
          SERIAL_PROTOCOLLNPGM("?Number of probed (P)oints is implausible (2 minimum).");
          return;
        }
      #endif
```
And be happy about that. I tried that on a i3 and got a memory overflow. (9 was the max. for me)
We should ether drop the P-parameter or make at least an upper limit.

This strange moves occur on delta machines when the probe/endstop is triggered unplanned. 
Try to increase Z_RAISE_BETWEEN_PROBINGS, Z_RAISE_PROBE_DEPLOY_STOW and the *_HOME_BUMP_MM values.
What kind of probe do you use?

aljscott, 2016-08-04:
I have an IR probe, mounted on a E3D chimera hotend.
https://miscsolutions.wordpress.com/mini-height-sensor-board/
It's fixed on the fan, and triggers about 3mm below the sensor.
What I've done is adjust the extruder tip to coincide with this point.
I've done some more analysis, certain points on the grid appear to be skipped.
Its apparent if you look at the following output, you can see the 0.0 readings.
Those positions in the matrix where never probed and they aren't out of bounds.
There was no endstop trigger. Will give your suggestions a try and pull the latest RCBugFix.
The weird thing is that X never probes left of -56, no matter how I adjust the LEFT_PROBE_BED_POSITION. I just don't seem to be able to get a grid that cover the heat bed. It's always off to the right, and towards the front.

SENDING:G29 V4
echo:Active Extruder: 0
G29 Auto Bed Leveling
Bed X: 76.000 Y: -40.000 Z: -1.998
Bed X: 54.000 Y: -40.000 Z: -0.448
Bed X: 32.000 Y: -40.000 Z: -0.683
Bed X: 10.000 Y: -40.000 Z: -0.701
Bed X: -12.000 Y: -40.000 Z: -0.618
Bed X: -34.000 Y: -40.000 Z: -0.429
Bed X: -56.000 Y: -40.000 Z: -0.149
Bed X: -78.000 Y: -40.000 Z: 0.152
Bed X: -78.000 Y: -25.000 Z: -0.007
Bed X: -56.000 Y: -25.000 Z: -0.292
Bed X: -34.000 Y: -25.000 Z: -0.549
Bed X: -12.000 Y: -25.000 Z: -0.730
Bed X: 10.000 Y: -25.000 Z: -0.838
Bed X: 32.000 Y: -25.000 Z: -0.831
Bed X: 54.000 Y: -25.000 Z: -0.621
Bed X: 76.000 Y: -25.000 Z: -0.401
Bed X: 98.000 Y: -10.000 Z: -0.364
Bed X: 76.000 Y: -10.000 Z: -0.551
Bed X: 54.000 Y: -10.000 Z: -0.720
Bed X: 32.000 Y: -10.000 Z: -0.798
Bed X: 10.000 Y: -10.000 Z: -0.931
Bed X: -12.000 Y: -10.000 Z: -0.800
Bed X: -34.000 Y: -10.000 Z: -0.632
Bed X: -56.000 Y: -10.000 Z: -0.421
Bed X: -78.000 Y: -10.000 Z: -0.014
Bed X: -78.000 Y: 5.000 Z: -0.277
Bed X: -56.000 Y: 5.000 Z: -0.541
Bed X: -34.000 Y: 5.000 Z: -0.709
Bed X: -12.000 Y: 5.000 Z: -0.902
Bed X: 10.000 Y: 5.000 Z: -0.918
Bed X: 32.000 Y: 5.000 Z: -0.871
Bed X: 54.000 Y: 5.000 Z: -0.710
Bed X: 76.000 Y: 5.000 Z: -0.510
Bed X: 98.000 Y: 5.000 Z: -0.423
Bed X: 76.000 Y: 20.000 Z: -0.578
Bed X: 54.000 Y: 20.000 Z: -0.679
Bed X: 32.000 Y: 20.000 Z: -0.928
Bed X: 10.000 Y: 20.000 Z: -0.969
Bed X: -12.000 Y: 20.000 Z: -1.008
Bed X: -34.000 Y: 20.000 Z: -0.952
Bed X: -56.000 Y: 20.000 Z: -0.851
Bed X: -78.000 Y: 20.000 Z: -0.590
Bed X: -78.000 Y: 35.000 Z: -0.702
Bed X: -56.000 Y: 35.000 Z: -0.993
Bed X: -34.000 Y: 35.000 Z: -1.200
Bed X: -12.000 Y: 35.000 Z: -1.239
Bed X: 10.000 Y: 35.000 Z: -1.209
Bed X: 32.000 Y: 35.000 Z: -1.112
Bed X: 54.000 Y: 35.000 Z: -0.957
Bed X: 76.000 Y: 35.000 Z: -0.733
Bed X: 76.000 Y: 50.000 Z: -0.747
Bed X: 54.000 Y: 50.000 Z: -1.159
Bed X: 32.000 Y: 50.000 Z: -1.228
Bed X: 10.000 Y: 50.000 Z: -1.400
Bed X: -12.000 Y: 50.000 Z: -1.359
Bed X: -34.000 Y: 50.000 Z: -1.313
Bed X: -56.000 Y: 50.000 Z: -0.969
Bed X: -78.000 Y: 50.000 Z: -0.760
Bed X: -56.000 Y: 65.000 Z: -1.019
Bed X: -34.000 Y: 65.000 Z: -1.221
Bed X: -12.000 Y: 65.000 Z: -1.461
Bed X: 10.000 Y: 65.000 Z: -1.470
Bed X: 32.000 Y: 65.000 Z: -1.387
Bed X: 54.000 Y: 65.000 Z: -1.071
Bed X: 54.000 Y: 80.000 Z: -1.157
Bed X: 32.000 Y: 80.000 Z: -1.345
Bed X: 10.000 Y: 80.000 Z: -1.360
Bed X: -12.000 Y: 80.000 Z: -1.300
Bed X: -34.000 Y: 80.000 Z: -1.079
Bed X: -56.000 Y: 80.000 Z: -0.907
Bed X: -12.000 Y: 95.000 Z: -1.140
Bed X: 10.000 Y: 95.000 Z: -1.188

0.45 0.15 -0.15 -0.43 -0.62 -0.70 -0.68 -0.45 -2.00 **0.00**
0.51 -0.01 -0.29 -0.55 -0.73 -0.84 -0.83 -0.62 -0.40 **0.00**
0.30 -0.01 -0.42 -0.63 -0.80 -0.93 -0.80 -0.72 -0.55 -0.36
-0.01 -0.28 -0.54 -0.71 -0.90 -0.92 -0.87 -0.71 -0.51 -0.42
-0.33 -0.59 -0.85 -0.95 -1.01 -0.97 -0.93 -0.68 -0.58 **0.00**
-0.41 -0.70 -0.99 -1.20 -1.24 -1.21 -1.11 -0.96 -0.73 **0.00**
-0.55 -0.76 -0.97 -1.31 -1.36 -1.40 -1.23 -1.16 -0.75 **0.00**
-0.61 -0.82 -1.02 -1.22 -1.46 -1.47 -1.39 -1.07 -0.76 **0.00**
-0.66 -0.73 -0.91 -1.08 -1.30 -1.36 -1.35 -1.16 -0.91 **0.00**
**0.00 0.00 0.00 0.00** -1.14 -1.19 **0.00 0.00 0.00 0.00**

tiagom62, 2016-08-05:
Will try either tomorrow night or Saturday.

AnHardt, 2016-08-05:
What is reachable?
[image]

red -> square bed
green -> DELTA_PRINTABLE_RADIUS
light gray -> DELTA_PRINTABLE_RADIUS displaced by X_PROBE_OFFSET_FROM_EXTRUDER, Y_PROBE_OFFSET_FROM_EXTRUDER
dark gray -> DELTA_PROBEABLE_RADIUS when no tricks are used.

DELTA_PRINTABLE_RADIUS is limited by three factors.
* The diagonal rods have theyr farest reach when they are horizontal.
* The bed diameter, or if square max((Xmax-Xmin)/2, (Ymax-Ymin)/2)
* Collisions: effektor-tower, diagonal rod-cariage,  ...

If we want to use the DELTA_PROBEABLE_RADIUS it is smaller than DELTA_PRINTABLE_RADIUS by sqrt(sq(X_PROBE_OFFSET_FROM_EXTRUDER)+sq(Y_PROBE_OFFSET_FROM_EXTRUDER)).
That means on the right side of the picture there are a lot of points what could be probed, but are not because we subtracted our probe distance from everywhere.

```
    #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 10)
    #define LEFT_PROBE_BED_POSITION max(DELTA_PROBEABLE_RADIUS,(-DELTA_PRINTABLE_RADIUS + X_PROBE_OFFSET_FROM_EXTRUDER))
    #define RIGHT_PROBE_BED_POSITION min(DELTA_PROBEABLE_RADIUS,(DELTA_PRINTABLE_RADIUS + X_PROBE_OFFSET_FROM_EXTRUDER))
    #define FRONT_PROBE_BED_POSITION max(DELTA_PROBEABLE_RADIUS,(-DELTA_PRINTABLE_RADIUS + Y_PROBE_OFFSET_FROM_EXTRUDER))
    #define BACK_PROBE_BED_POSITION min(DELTA_PROBEABLE_RADIUS,(DELTA_PRINTABLE_RADIUS + Y_PROBE_OFFSET_FROM_EXTRUDER))
```
with -10 being a safety margin (could be reduced until probing directly on the beds border)
This rectangle describes an area where all probable points are in, covering the biggest possible area. On the other side here the corner points are not probable.

That does not matter.
```
        for (int xCount = xStart; xCount != xStop; xCount += xInc) {
          double xProbe = left_probe_bed_position + xGridSpacing * xCount;

          #if ENABLED(DELTA)
            // Avoid probing the corners (outside the round or hexagon print surface) on a delta printer.
            float distance_from_center = HYPOT(xProbe, yProbe);
            if (distance_from_center > DELTA_PROBEABLE_RADIUS) continue;
          #endif //DELTA

          float measured_z = probe_pt(xProbe, yProbe, stow_probe_after_each, verbose_level);
```
simply skips the unprobable points. (Could be discussed if DELTA_PRINTABLE_RADIUS is the better margin here.)

However
```
      #if ENABLED(DELTA)

        if (!dryrun) extrapolate_unprobed_bed_level();
        print_bed_level();

      #else // !DELTA

...

  #if ENABLED(DELTA)

    /**
     * All DELTA leveling in the Marlin uses NONLINEAR_BED_LEVELING
     */
    static void extrapolate_one_point(int x, int y, int xdir, int ydir) {
      if (bed_level[x][y] != 0.0) {
        return;  // Don't overwrite good values.
      }
      float a = 2 * bed_level[x + xdir][y] - bed_level[x + xdir * 2][y]; // Left to right.
      float b = 2 * bed_level[x][y + ydir] - bed_level[x][y + ydir * 2]; // Front to back.
      float c = 2 * bed_level[x + xdir][y + ydir] - bed_level[x + xdir * 2][y + ydir * 2]; // Diagonal.
      float median = c;  // Median is robust (ignores outliers).
      if (a < b) {
        if (b < c) median = b;
        if (c < a) median = a;
      }
      else {  // b <= a
[... 55 lines cut ...]
````
    // Adjust print surface height by linear interpolation over the bed_level array.
    void adjust_delta(float cartesian[3]) {
      if (delta_grid_spacing[0] == 0 || delta_grid_spacing[1] == 0) return; // G29 not done!

      int half = (AUTO_BED_LEVELING_GRID_POINTS - 1) / 2;
      float h1 = 0.001 - half, h2 = half - 0.001,
            grid_x = max(h1, min(h2, RAW_X_POSITION(cartesian[X_AXIS]) / delta_grid_spacing[0])),
            grid_y = max(h1, min(h2, RAW_Y_POSITION(cartesian[Y_AXIS]) / delta_grid_spacing[1]));
      int floor_x = floor(grid_x), floor_y = floor(grid_y);
      float ratio_x = grid_x - floor_x, ratio_y = grid_y - floor_y,
            z1 = bed_level[floor_x + half][floor_y + half],
            z2 = bed_level[floor_x + half][floor_y + half + 1],
            z3 = bed_level[floor_x + half + 1][floor_y + half],
            z4 = bed_level[floor_x + half + 1][floor_y + half + 1],
            left = (1 - ratio_y) * z1 + ratio_y * z2,
            right = (1 - ratio_y) * z3 + ratio_y * z4,
            offset = (1 - ratio_x) * left + ratio_x * right;

      delta[X_AXIS] += offset;
      delta[Y_AXIS] += offset;
      delta[Z_AXIS] += offset;
}
```
hopefully extrapolates points outside the measured grid from the outmost points in the grid.

AnHardt, 2016-08-05:
For the las provided data that means.
Not probing all points in the rectangle is not really a problem but by design. Finding a meaningful rectangle can for sure be improved. 
Seeing the 0.0 values in the result __is a problem__ because it indicates that `extrapolate_unprobed_bed_level()` is not doing its job (if you have not been in dryrun mode).

aljscott, 2016-08-07:
Some more information.
Turning on PROBE_DOUBLE_TOUCH make things worse.
I assume its meamt to raise and probes the same position twice?
With it enabled it goes to the first point and probes, raises, then takes off in the +X direction out of bounds and tries to probe off the bed. Then it continues to the next point going +X again at which point I hit the kill button before it crashes into the vertical tower.
If it is not enabled, it probes the first point and then heads -X direction for the second point.
The grid is heavily off in the +X and -Y direction it barely even gets into the -X and +Y territory to probe.
Seems a bit weird that we can have a nicely distributed grid around 0,0?

So the other weird thing is that it came out with +1.90 value for around the centre of the bed.
If I try to print something the nozzle is sitting at a consistent 1.9mm above the bed at 0,0.
Manually moving the head the probe is not misfiring and the hotend / probe are adjusted so they're at the same level. After probing I go G0 X0 Y0 Z0 and its floating above the bed, if I manual move to -Z the probe will trigger when I am ~ -1.9.

I have also had a few situations where moving the head in a linear X or Y direction it will go up or down in the Z axis.. this is always after G29. It gets into some weird state.

To me it seems like the probe offset from extruder isn't behaving as it should and it is doing something weird to the grid.
 

#define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 10)
    #define LEFT_PROBE_BED_POSITION max(DELTA_PROBEABLE_RADIUS,(-DELTA_PRINTABLE_RADIUS + X_PROBE_OFFSET_FROM_EXTRUDER))
    #define RIGHT_PROBE_BED_POSITION min(DELTA_PROBEABLE_RADIUS,(DELTA_PRINTABLE_RADIUS + X_PROBE_OFFSET_FROM_EXTRUDER))
    #define FRONT_PROBE_BED_POSITION max(DELTA_PROBEABLE_RADIUS,(-DELTA_PRINTABLE_RADIUS + Y_PROBE_OFFSET_FROM_EXTRUDER))
    #define BACK_PROBE_BED_POSITION min(DELTA_PROBEABLE_RADIUS,(DELTA_PRINTABLE_RADIUS + Y_PROBE_OFFSET_FROM_EXTRUDER))

^ With this in mind, I'll have a bit of a think tomorrow.

thinkyhead, 2016-08-08:
Please enable `DEBUG_LEVELING_FEATURE` and use `M111 S32` to enable logging of homing and leveling. Perform the `G29` procedure with a host connected, and post the log output for us to examine. It will give us the best picture of what is happening.

aljscott, 2016-08-08:
So after all of that I put the AUTO_BED_LEVELING_GRID_POINTS back to 5 and everything started behaving a bit more like I'd expect.
The probeable area actually covers most of the print bed now, still a bit constrained on the left side.
Actually managed to print a cube now.

boelle, 2016-08-11:
So this one can be closed?

aljscott, 2016-08-11:
If AUTO_BED_LEVELING_GRID_POINTS > 5 the area probed doesn't seem to make much sense in relation to the probeable area.
Either needs a comment in the code or an work out what is going on.
Increasing the grid points shrinks the area probes towards the right, and front.

zenmetsu, 2016-08-11:
Somewhat related as it was brought up in this discussion and affects the probed points...
`#define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 10)`
In all honesty, that '10' shouldn't be hard-coded.  My probe offset is 25mm behind the nozzle.  If any of the X or Y offsets exceed this static value of 10, then G29 will fail with a 'probe out of bounds' error.  I think that this value should be calculated based upon the user-supplied probe X/Y offsets.

Blue-Marlin, 2016-08-11:
It's in the Configuratin.h file - so it's not hard coded.
how about:
```
#define SAFETY_MARGIN 10
#define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - sqrt(sq(X_PROBE_OFFSET_FROM_EXTRUDER)+sq(Y_PROBE_OFFSET_FROM_EXTRUDER)) - SAFETY_MARGIN )
```
https://github.com/MarlinFirmware/Marlin/issues/4520#issuecomment-237950042

zenmetsu, 2016-08-11:
That's kinda what I meant.  Granted, the value only needs to be changed in one place, but the implications of changing this value from '10' might not be understood as it isn't documented...

```

  #define AUTO_BED_LEVELING_GRID  // Deltas only support grid mode.

  #if ENABLED(AUTO_BED_LEVELING_GRID)

    // Set the rectangle in which to probe
    #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 10)
    #define LEFT_PROBE_BED_POSITION -(DELTA_PROBEABLE_RADIUS)
    #define RIGHT_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS
    #define FRONT_PROBE_BED_POSITION -(DELTA_PROBEABLE_RADIUS)
    #define BACK_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS
```

Adding `#define SAFETY_MARGIN 10` makes it a bit more clear, but a short description in a code comment would be a good idea.  There should be a fallback to check if SAFETY_MARGIN is defined, and not, calculate it as `#define SAFETY_MARGIN sqrt(sq(X_PROBE_OFFSET_FROM_EXTRUDER)+sq(Y_PROBE_OFFSET_FROM_EXTRUDER))`

thinkyhead, 2016-08-11:
> calculate it as `sqrt(sq(X_PROBE_OFFSET_FROM_EXTRUDER)+sq(Y_PROBE_OFFSET_FROM_EXTRUDER))`

I believe that would break the sanity check and cause a compiler error.

zenmetsu, 2016-08-11:
You lost me here... XD

I don't see where this would break anything within SanityCheck.h, and the compiler had no issues with it... and it executes on the printer just fine.

For reference, my printable radius is 80mm, and my probe offset is X0 Y25 Z-6.8

What I have noticed is that my probeable radius should be 80-25 = 55mm.  The actual value appears to be 49mm, a difference of 6mm.  I set my MIN_PROBE_EDGE to 5mm, so perhaps it is insetting the actual probe area by this amount.  I'll have to look at the code.
```
SENT: G28
READ: ok
SENT: G29 V4
READ: G29 Auto Bed Leveling
READ: Bed X: 23.000 Y: -42.000 Z: 4.982
READ: Bed X: 10.000 Y: -42.000 Z: 4.978
READ: Bed X: -3.000 Y: -42.000 Z: 5.002
READ: Bed X: -16.000 Y: -42.000 Z: 4.934
READ: Bed X: -29.000 Y: -42.000 Z: 4.907
READ: Bed X: -42.000 Y: -29.000 Z: 4.949
READ: Bed X: -29.000 Y: -29.000 Z: 4.892
READ: Bed X: -16.000 Y: -29.000 Z: 4.985
READ: Bed X: -3.000 Y: -29.000 Z: 5.016
READ: Bed X: 10.000 Y: -29.000 Z: 5.115
READ: Bed X: 23.000 Y: -29.000 Z: 5.009
READ: Bed X: 36.000 Y: -29.000 Z: 5.134
READ: Bed X: 49.000 Y: -16.000 Z: 5.116
READ: Bed X: 36.000 Y: -16.000 Z: 5.089
READ: Bed X: 23.000 Y: -16.000 Z: 5.043
READ: Bed X: 10.000 Y: -16.000 Z: 4.995
READ: Bed X: -3.000 Y: -16.000 Z: 5.000
READ: Bed X: -16.000 Y: -16.000 Z: 5.036
READ: Bed X: -29.000 Y: -16.000 Z: 5.018
READ: Bed X: -42.000 Y: -16.000 Z: 5.037
READ: Bed X: -42.000 Y: -3.000 Z: 4.955
READ: Bed X: -29.000 Y: -3.000 Z: 4.947
READ: Bed X: -16.000 Y: -3.000 Z: 4.976
READ: Bed X: -3.000 Y: -3.000 Z: 4.995
READ: Bed X: 10.000 Y: -3.000 Z: 5.137
READ: Bed X: 23.000 Y: -3.000 Z: 5.202
READ: Bed X: 36.000 Y: -3.000 Z: 5.187
READ: Bed X: 49.000 Y: -3.000 Z: 5.211
READ: Bed X: 49.000 Y: 10.000 Z: 5.050
READ: Bed X: 36.000 Y: 10.000 Z: 5.067
READ: Bed X: 23.000 Y: 10.000 Z: 5.179
READ: Bed X: 10.000 Y: 10.000 Z: 5.019
READ: Bed X: -3.000 Y: 10.000 Z: 5.029
READ: Bed X: -16.000 Y: 10.000 Z: 5.042
READ: Bed X: -29.000 Y: 10.000 Z: 5.042
READ: Bed X: -42.000 Y: 10.000 Z: 5.004
READ: Bed X: -42.000 Y: 23.000 Z: 4.914
READ: Bed X: -29.000 Y: 23.000 Z: 4.949
READ: Bed X: -16.000 Y: 23.000 Z: 4.956
READ: Bed X: -3.000 Y: 23.000 Z: 4.995
READ: Bed X: 10.000 Y: 23.000 Z: 5.136
READ: Bed X: 23.000 Y: 23.000 Z: 5.164
READ: Bed X: 36.000 Y: 23.000 Z: 5.154
READ: Bed X: 49.000 Y: 23.000 Z: 5.131
READ: Bed X: 36.000 Y: 36.000 Z: 5.017
READ: Bed X: 23.000 Y: 36.000 Z: 5.068
READ: Bed X: 10.000 Y: 36.000 Z: 5.015
READ: Bed X: -3.000 Y: 36.000 Z: 4.999
READ: Bed X: -16.000 Y: 36.000 Z: 5.067
READ: Bed X: -29.000 Y: 36.000 Z: 5.015
READ: Bed X: -16.000 Y: 49.000 Z: 4.956
READ: Bed X: -3.000 Y: 49.000 Z: 5.000
READ: Bed X: 10.000 Y: 49.000 Z: 5.108
READ: Bed X: 23.000 Y: 49.000 Z: 5.140
READ: -0.31 -0.28 -0.25 -0.22 -0.11 -0.26 -0.16 -0.19 -0.07
READ: -0.22 -0.24 -0.19 -0.17 -0.10 -0.12 -0.12 -0.08 -0.04
READ: 0.03 -0.15 -0.21 -0.11 -0.08 0.02 -0.09 0.03 -0.08
READ: -0.04 -0.06 -0.08 -0.06 -0.10 -0.10 -0.06 -0.01 0.02
READ: -0.14 -0.15 -0.15 -0.12 -0.10 0.04 0.10 0.09 0.11
READ: -0.13 -0.10 -0.06 -0.06 -0.07 -0.08 0.08 -0.03 -0.05
READ: -0.13 -0.19 -0.15 -0.14 -0.10 0.04 0.06 0.05 0.03
READ: -0.31 -0.24 -0.09 -0.03 -0.10 -0.08 -0.03 -0.08 0.03
READ: -0.34 -0.03 -0.02 -0.14 -0.10 0.01 0.04 -0.10 -0.23
READ: X:23.00 Y:24.00 Z:10.00 E:0.00 Count X: 6986 Y:7941 Z:8294
```

zenmetsu, 2016-08-12:
@aljscott Regarding the whole "probing in the air" bit, I run into that if I don't use `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`  The other option, `#define Z_MIN_PROBE_ENDSTOP` causes the probe to not reach the bed, and as the points continue, the Z value continues to climb moving further from the bed until the effector crashes into the Z tower.

This also happened until using the Z_MIN_ENDSTOP_PIN until I changed the value '#define Z_PROBE_DEPLOY_HEIGHT` from 15 to 50.  Once that was changed, probing continued as expected.

zenmetsu, 2016-08-12:
I figured out how to get the grid to be centered and using as much of the bed as possible. 
I reduced AUTO_BED_LEVELING_GRID_POINTS to 7 and re-executed G29 V4

Notice how I get a single point now for the center rows, and how the grid is centered at X0 Y0.

```
SENT: G28
READ: ok
SENT: G29 V4
READ: G29 Auto Bed Leveling
READ: Bed X: 0.000 Y: -54.000 Z: 4.979
READ: Bed X: 36.000 Y: -36.000 Z: 5.003
READ: Bed X: 18.000 Y: -36.000 Z: 5.018
READ: Bed X: 0.000 Y: -36.000 Z: 5.003
READ: Bed X: -18.000 Y: -36.000 Z: 5.018
READ: Bed X: -36.000 Y: -36.000 Z: 4.868
READ: Bed X: -36.000 Y: -18.000 Z: 4.789
READ: Bed X: -18.000 Y: -18.000 Z: 4.958
READ: Bed X: 0.000 Y: -18.000 Z: 4.974
READ: Bed X: 18.000 Y: -18.000 Z: 5.116
READ: Bed X: 36.000 Y: -18.000 Z: 5.104
READ: Bed X: 54.000 Y: 0.000 Z: 5.119
READ: Bed X: 36.000 Y: 0.000 Z: 5.024
READ: Bed X: 18.000 Y: 0.000 Z: 4.998
READ: Bed X: 0.000 Y: 0.000 Z: 4.973
READ: Bed X: -18.000 Y: 0.000 Z: 4.998
READ: Bed X: -36.000 Y: 0.000 Z: 4.979
READ: Bed X: -54.000 Y: 0.000 Z: 4.759
READ: Bed X: -36.000 Y: 18.000 Z: 4.864
READ: Bed X: -18.000 Y: 18.000 Z: 5.015
READ: Bed X: 0.000 Y: 18.000 Z: 5.095
READ: Bed X: 18.000 Y: 18.000 Z: 5.172
READ: Bed X: 36.000 Y: 18.000 Z: 5.179
READ: Bed X: 36.000 Y: 36.000 Z: 5.062
READ: Bed X: 18.000 Y: 36.000 Z: 5.217
READ: Bed X: 0.000 Y: 36.000 Z: 5.178
READ: Bed X: -18.000 Y: 36.000 Z: 5.060
READ: Bed X: -36.000 Y: 36.000 Z: 4.950
READ: Bed X: 0.000 Y: 54.000 Z: 4.982
READ: -0.32 -0.04 -0.02 -0.12 -0.18 -0.20 -0.21
READ: -0.52 -0.23 -0.08 -0.10 -0.08 -0.10 -0.03
READ: -0.48 -0.31 -0.14 -0.13 0.02 0.00 -0.01
READ: -0.34 -0.12 -0.10 -0.13 -0.10 -0.08 0.02
READ: -0.39 -0.24 -0.09 -0.00 0.07 0.08 0.09
READ: -0.37 -0.15 -0.04 0.08 0.12 -0.04 0.15
READ: -0.22 -0.07 0.00 -0.12 0.16 0.24 0.22
READ: X:0.00 Y:29.00 Z:10.00 E:0.00 Count X: 7457 Y:7457 Z:8433
```

I found out that when the autoprobing was calculating the distance from center, around line 3626 in Marlin_main.cpp, the hypotenuse function could return a float that was just a bit larger than the actual input probe offset... 
```
            float distance_from_center = HYPOT(xProbe, yProbe);
            if (distance_from_center > DELTA_PROBEABLE_RADIUS) continue;
```
This was like 24.998 vs. my input value of 25.0 for total probe offset.  This would cause the front and left rows of points to be dropped entirely and result in an offset grid.  

I realized that I could not change the probeable radius by making it larger because that would trigger the "probe out of bounds" error when G29 was executed.  I then realized that the G29 output shows float values for probed X and Y, but they all end with .00, telling me that the value is actually an int being cast as a float and the two decimal points should probably be omitted in the verbose output.  This caused me to realize that the rounding was causing the dropped points mentioned above.  The fix was to add 1 to yProbe and xProbe calculations...

```
#Marlin_main.cpp
@3593
double yProbe = front_probe_bed_position + yGridSpacing * yCount  + 1.0;
...
@3610
double xProbe = left_probe_bed_position + xGridSpacing * xCount + 1.0;
```

thinkyhead, 2016-08-18:
Ooooh. How I love floats. Thanks for catching that one!

zenmetsu, 2016-08-18:
I made a pull request that has the results of my testing.  There were some additional questions in there regarding another function, but I am still learning my way through Marlin code and perhaps you could chime in.

thinkyhead, 2016-08-19:
@zenmetsu I've incorporated various changes from our conversation into the above PR. I think it achieves a good balance by (1) using whole-number probe points, (2) enforcing a minimum and odd number of points, and (3) providing a more usable default for all machines (3x3 instead of 2x2 points).

thinkyhead, 2016-08-21:
The `HYPOT` function isn't being used any more. It now just compares the sum of squares to the squared radius. If needed, we can correct it a bit by adding 0.001 in the comparison, as in:
```cpp
if (sq(xProbe) + sq(yProbe) > sq(DELTA_PROBEABLE_RADIUS) + 0.01) continue;
```

zenmetsu, 2016-08-21:
That might be the best choice.

thinkyhead, 2016-08-23:
> That might be the best choice.

Done.

closed 2016-08-23 by thinkyhead