MarlinFirmware/Marlin #4521: Marlin freezes during print after 3-4hours

guerradr, 2016-08-03:
After 3 to 4 hours into the print the printer has a total freeze. Printerhead just stops moving and the power to Extruder and heated bed turns off. It does not react on any input by graphic display controller or usb. I had this phenomeno repeatetly during three prints with different slicer Software. And I did not have any power surge whatsoever since the screen was blocked on the print info display still indicating the printing temperatures.
I use:

• Marlin 1.1.0-RC7 
• rigidbot big with custom modifications
• RUMBA Board with 1/32 microstepping
• e3d hotend
• printing via sd card
• slicer cura and simplify3d

I will upload the config files asap.

adamfilip, 2016-08-03:
Ive run into this before.. but my issue was overheating of the steppers. once i added a fan blowing on my board the issue went away.. just a thought.

did this happen before RC7?

guerradr, 2016-08-03:
No it did not happen before. I was using RC6, RC5 and v1.1 and did not have any problems during long prints. My driver board is already cooled by a fan with a nicely made ducting ;-).
Here are my config files:

guerradr, 2016-08-03:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/400159/Configuration.txt)

manianac, 2016-08-04:
Can you post a serial log of when it fails?

QoooQ, 2016-08-04:
freeze me too?!  when i printing after 1-2 hours   Then i into menu is freeze     must   push resert busston

lavato, 2016-08-04:
I can confirm that this did happened to me too.

guerradr, 2016-08-05:
@manianac : I print from the sd Card without pc attached, how can i record a Serial log?
I moved back to rc6 where I don't expirience this Problem, but I will certainly assist in testing.

thinkyhead, 2016-08-07:
> I print from the sd Card without pc attached, how can i record a Serial log?

You will have to connect a PC, and then there's no guarantee that the log will actually reveal anything.

boelle, 2016-08-11:
I never had this issue when i used marlin... i even did 8 hours print back then

but yes connect to a PC... and then have a sniffer app log the serial traffic

thinkyhead, 2016-08-11:
> then have a sniffer app log the serial traffic

@guerradr If you don't have a "sniffer app" handy, you can just use any 3D print host, such as Repetier, Printrun/Pronterface, etc., etc.

zenmetsu, 2016-08-11:
Along these lines, you could enable the free memory check command and issue `M100` as part of your layer change scripts, then observe the logs from Pronterface as thinkyhead suggested... that is if `M100` works properly.  For me, it reports once, and then subsequent invocations do nothing.  I am logging that as a separate issue.

thinkyhead, 2016-08-11:
@zenmetsu Today we both learned that `M100 F` is required or `M100` doesn't do anything on subsequent invocations.

Anyway, I'm pretty sure `M100` will be useless to debug this issue, but have at it.

Roxy-3D, 2016-08-11:
> Anyway, I'm pretty sure M100 will be useless to debug this issue, but have at it.

Agreed.   There is plenty of room between the stack and heap.   If memory is getting corrupted, my bet is something else is causing it.

zenmetsu, 2016-08-11:
Well, it could help rule out memory leaks, and I thought I'd throw it out since putting it into a layer change script would allow for periodic polling.  I've started a print job since learning about the proper invocation.  So far, there appears to be no chance of a stack collision, but you never know if features or modifications to the files could be causing something bad to happen.  I'll keep this one in my back pocket should I ever run into a problem like this where there are unexplained crashes. 

The only times I ever ran into problems with a mid-print failure like this was when I had power issues (brownouts) or I was using Repetier and had my baud rate set to 115200.  I suspect there was a buffering issue since bumping the baud rate to 250000 completely eliminated that problem.  If the printer is unresponsive via the LCD/controller, I would be inclined to rule this out.

thinkyhead, 2016-08-13:
> memory leaks

We can't have those because we don't do any dynamic allocation. All allocation happens at startup time. The only "exception," if you want to call it that, is when we allocate things on the stack within functions — automatic variables. Almost all of those allocations are for fixed amounts, and of course when the function exits the stack is reset. But there are a couple of places where we allocate space on the stack based on a `strlen` or other calculated-at-runtime value. If a string were not properly terminated, it could overflow the stack. (But it would have to be pretty large, as on larger boards there's ~3K of space on the stack/heap.)

The kind of thing that (in software) might cause a crash, freeze, or serious glitch is "buffer overrun." That is when we write data past the end of a buffer, whether that buffer was allocated at boot time in the heap or on the stack doesn't matter.

zenmetsu, 2016-08-13:
ok, my bad. I saw a malloc in the M100 code and just made the assumption that memory was being allocated elsewhere, but it isn't.  And I agree, the M100 code sure as heck isn't going to help in the event of a buffer overrun.  :)

Roxy-3D, 2016-08-13:
And just to be clear, that malloc() in the M100 is new.  It is only there because nothing else ever does a malloc() and we need __brkval to be properly set to get statistics:
~~~~

void gcode_M100() {
  static int m100_not_initialized = 1;
  unsigned char* sp, *ptr;
  int i, j, n;

  if ( __brkval == 0 ) {		// If so, nobody has done a malloc() and we have 
    ptr = (unsigned char *) malloc(1);	// no idea where the heap starts.  We need to force
    *ptr++;				// __brkval to be initialized.  But we also have to 
    free(ptr);				// work around some compiler omptimizations.  GCC 
   					// won't even do the call to malloc() if we don't use
					// the results of the call.  So, we bump the memory
					// at the pointer and then give back the memory.  This 
					// will give us a valid idea of where the Global Variables
					// end and the free memory (above the top of the
  }					// stack) starts.
~~~~

zenmetsu, 2016-08-13:
Fair enough.  Back to the freezing issue, I can see from the posted config that the baudrate issue I was referring to is not going to be the problem.  Short of thermal issues, I'm at a loss.  I ran `RCBugFix` on 3 batches of parts, each being a 385min print job and I did not see a problem with hanging.

boelle, 2016-08-14:
can @anyone reproduce this one?

boelle, 2016-08-24:
@thinkyhead @jbrazio maybe close this one as there is no response in 10 days?

closed 2016-08-24 by thinkyhead

thinkyhead, 2016-08-30:
@guerradr I've been looking into another issue where there's a freeze in the middle of a long Z move at a very high rate of steps (37,333 steps per second). I wonder if the Rumba board's CPU just can't keep up with 32x microstepping once there comes a point where several operations just happen to concur with each other.

Of course it also comes down to feedrates too. Your settings are much more modest than the other freeze issue I'm tracking. But maybe you are printing with smaller layer heights and with faster rates…?

Try doing some prints with the microsteps reduced to 16 and see if it stops freezes. If it does, then you may be able to get away with simply reducing microstepping on the Z axis alone, but leave 32x on the other axes. You could also try to _cause a freeze_ by increasing the feedrate during a (small) print session. If you find that increasing the feedrate leads to a freeze, then we might be onto something.

pmbroth, 2016-12-18:
I and several; others have had this happen several times in long prints.  We use the TREX (+) printer from formbot, and it will pause, and you cannot continue.  If you use a pc or octoprint, no issues

thinkyhead, 2016-12-19:
@pmbroth Are you using the current `RCBigFix` branch? Are you able to force this failure to happen (in a shorter print) by increasing speed or by other tweaks? It would help to pinpoint the problem if it can be shown to be related to high speed or some other definitive cause.

adammhaile, 2017-11-15:
Just ran into this on a FormBot T-Rex 2+, running from SD. Seems no one had this issue when tethered? Guess it's time to setup Octoprint...

thinkyhead, 2017-11-15:
@adammhaile You experienced this while using the latest `bugfix-1.1.x` code?

adammhaile, 2017-11-15:
Can't say for sure. FormBot won't release the firmware source for the T-Rex
2+ :( they say they will once more printers have IDEX
And yes, I have already told them multiple times that is not how the GPL
works.

adammhaile, 2017-11-15:
I asked and they did at least say it was based on bugfix-1.1.x
Beginning to wonder if it was a corrupted SD card.

pmbroth, 2017-11-15:
What date did the sayvthe big fix 1.1x did they use. It changes all the
time.

adammhaile, 2017-11-15:
They didn't say. Will ask.

pmbroth, 2017-11-15:
I upgrade my trex plus almost every time a new build is released.

BR

adammhaile, 2017-11-15:
> 4521#issuecomment-344631793
> AA6a6hJBcXjLmiovN844eCJlbjYt0S3Kks5s2wYFgaJpZM4Jbh3M

Which version of the T-Rex? (Have the 2+) and do you mean firmware hex
files from Formbot or direct from Marlin main source?
Formbot refuses to release the source for the 2+

pmbroth, 2017-11-15:
I have the Trex plus.   The one before Trex 2.

adammhaile, 2017-11-16:
> 4521#issuecomment-344685160
> CFP54PZQtoDks5s2y4jgaJpZM4Jbh3M

Formbot says May, 2017

pmbroth, 2017-11-16:
Wow that is old.   Many revisions and enhancements since then

adammhaile, 2017-11-16:
> 4521#issuecomment-344708154
> yuXUupD4k_dnkGYks5s20C1gaJpZM4Jbh3M

Sigh :( guess I'll keep hounding them to actually release the source... I
yell at them daily now.

mjc619, 2017-11-24:
I just got this error (or a very similar one) from octoprint, lcd was still working and have the log, overnight 2 hours in on a 4 hour print it hanged and on pausing and resuming had started moving the extruder again but not xy and lcd gave a "homexy" notification before I stopped it due to the filament ruining the print, my pause command needs to be fixed too as I haven't used octopritnt's pause for weeks and could be a M91 bug, put pause and resume below the log.

I did not set the fade hight according to the M420 either

[octoprint.log](https://github.com/MarlinFirmware/Marlin/files/1502179/octoprint.log)

Pause: 
    ; relative XYZE
    G91
    M83

    ; retract filament, move Z slightly upwards
    G1 Z+5 E-5 F4500 

    ; absolute XYZE
    M82
    G90

    ; move to a safe rest position, adjust as necessary
    G1 X0 Y180 

Resume:

; relative extruder
M83

; prime nozzle
G1 E-5 F4500
G1 E5 F4500
G1 E5 F4500

; absolute E
M82

; absolute XYZ
G90

; reset E
G92 E{{ pause_position.e }}

; move back to pause position XYZ
G1 X{{ pause_position.x }} Y{{ pause_position.y }} Z{{ pause_position.z }} F4500

; reset to feed rate before pause
G1 {{ pause_position.f }}

thinkyhead, 2017-11-24:
Marlin can't parse that "Resume" g-code, though possibly a host could fill in those values.

For pause/resume, Marlin has the `PARK_HEAD_ON_PAUSE` feature, which adds the `M125` G-code to pause and move to a park position. It waits for an LCD click before resuming. (It also responds to `M108` if you have `EMERGENCY_PARSER` enabled). On resume, the nozzle moves back to its previous position before it was paused.