MarlinFirmware/Marlin #4524: Want option to easily disable LCD beeper/buzzer/speaker in configuration.h

funbobby2001, 2016-08-03:
It drives me batty.

Blue-Marlin, 2016-08-03:
```// SPEAKER/BUZZER
//
// If you have a speaker that can produce tones, enable it here.
// By default Marlin assumes you have a buzzer with a fixed frequency.
//
//#define SPEAKER

//
// The duration and frequency for the UI feedback sound.
// Set these to 0 to disable audio feedback in the LCD menus.
//
// Note: Test audio output with the G-Code:
//  M300 S<frequency Hz> P<duration ms>
//
//#define LCD_FEEDBACK_FREQUENCY_DURATION_MS 100`//!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
//#define LCD_FEEDBACK_FREQUENCY_HZ 1000                  //!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

```

funbobby2001, 2016-08-03:
Sorry, I thought I just had a "buzzer" and so those "speaker" settings didn't apply, hence I didn't read much past //#define SPEAKER in that section. Probably because I watched Tom Sanladerer digging through the other code modules to turn it off, not realizing that video was old. Not trying to be a nuisance.

closed 2016-08-03 by funbobby2001