MarlinFirmware/Marlin #4526: RC7 Experimental filament change display error

WZ9V, 2016-08-04:
On RC7 after doing a filament change using a XXL LCD the screen is not completely cleared prior to displaying the info screen.  On the third line of the display, the letters "to" remain on the screen between % and SD and the letter "e" remains between % and the timer.  There should be white space in those three places.

thinkyhead, 2016-08-04:
Interesting. The screen should always be cleared in-between displays, because the `lcd_goto_screen` function always sets `lcdDrawUpdate = LCDVIEW_CLEAR_CALL_REDRAW`. But maybe the `lcd.clear()` function isn't actually doing anything…?

To help debug this, please modify the `lcd_implementation_clear` function in `ultralcd_impl_HD44780.h` and check the output in your host console:
```cpp
static void lcd_implementation_clear() {
  SERIAL_ECHOLNPGM("Called lcd.clear()");
  lcd.clear();
}
```

WZ9V, 2016-08-04:
I'll try that tonight.  It is just my best guess that it is the clear, it could be elsewhere.  You agree with what my instincts say is the best place to look first.

WZ9V, 2016-08-05:
lcd.clear is not called after the screen saying

CHANGE FILAMENT
Wait for print 
to resume

so some of the test from the third line shows through on the info screen.

WZ9V, 2016-08-05:
I don't know how to create a pull request but I did this in ultralcd.cpp and it's working now.

  static void lcd_return_to_status() { 
    lcd_implementation_clear();  // add this line
    lcd_goto_screen(lcd_status_screen); 
  }

thinkyhead, 2016-08-06:
Very strange. Your solution is fine, but I would rather find the root cause as to why the screen is not being cleared based on the clear flag. Were you able to determine any pattern as to which screens are clearing properly versus which are not? Or maybe there is a bug where `lcdDrawUpdate` is being set to something else after it is set to `LCDVIEW_CLEAR_CALL_REDRAW`. I will have to keep looking.

WZ9V, 2016-08-06:
No idea, but it does seem contained to the last screen before switching back to the I for screen.

thinkyhead, 2016-08-06:
Well, I decided to go for the simplest solution and clear the screen explicitly in `lcd_goto_screen` instead of setting the flag to do it on the next loop. I'll merge that pretty soon.

closed 2016-08-07 by thinkyhead