MarlinFirmware/Marlin #4529: "LCD >Level Bed" after print

sarf2k4, 2016-08-04:
This happened twice on my printer and wondering if anyone else got this issues. From what I understand by using the the level bed from the lcd menu g28 will be automatically executed before the g29 (correct me on this one).

After print I tried to use the "lcd level bed" option and I noticed that marlin doesn't do g28 first but just straight go to g29 which will crash the axis and probes at the wrong area. This only happened after printing from sdcard (please confirm this one for me as well).

It seems that a new flag must be added 

```cpp
if (stepper_disabled)
   g28 then g29
else
   g29
```

I also included my config file here
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/401102/Configuration.txt)

thinkyhead, 2016-08-04:
If the machine knows where it is then it allows G29. Does your machine lose track of its position in the process of printing, or after completing the print?

sarf2k4, 2016-08-05:
It is after, because i disabled everything after print, stepper and heater.
The axis are also blinking

thinkyhead, 2016-08-06:
Probably I will submit this change, requiring `G28` if steppers are disabled:
```diff
   static bool axis_unhomed_error(const bool x, const bool y, const bool z) {
-    const bool xx = x && !axis_homed[X_AXIS],
-               yy = y && !axis_homed[Y_AXIS],
-               zz = z && !axis_homed[Z_AXIS];
+    const bool xx = x && (!axis_homed[X_AXIS] || !axis_known_position[X_AXIS]),
+               yy = y && (!axis_homed[Y_AXIS] || !axis_known_position[Y_AXIS]),
+               zz = z && (!axis_homed[Z_AXIS] || !axis_known_position[Z_AXIS]);
```

sarf2k4, 2016-08-06:
I am sorry i didnt report anything about this, the level bed were good to
begin with, it was just mispositioned head and the level bed were okay. But
i kinda wonder is it normal to have those axis still blinking indicating it
were never homed in the lcd panel?
Considering that the last known position of marlin after stepper disabled,
the axis might accidentally moved by user and straight go for level bed
which might gonna crash the axis to either sides without proper g28 =)

thinkyhead, 2016-08-06:
The blinking was added as a clear indication that the position is untrusted. Given the potential for user confusion, we might as well require homing any time a stepper has been disabled.

closed 2017-10-31 by thinkyhead