MarlinFirmware/Marlin #4531: Z Axis leveling

mperdue, 2016-08-04:
This is more of an inquiry than an issue. I use two stepper motors each connected to its own driver. I also have a BLTouch probe mounted to the extruder for bed leveling. Occasionally the two motor get a bit out of sync and the z-azis ends up not parallel to the print bed. This is usually caused by the extruder crashing into the bed or some other nastiness. What I'd like to be able to do is add two limit switches to the z-axis so it can be auto leveled by running each motor until the limit is reached. Alternatively, it could be done with the existing probe at one side of the bed and a switch on the opposite side.

Does this capability exist in Marlin? If yes, how is it set up? If no, then I me need to figure out how to add it.

Blue-Marlin, 2016-08-04:
Look for `Z_DUAL_STEPPER_DRIVERS` and `Z_DUAL_ENDSTOPS` in `Configuration_adv.h`

mperdue, 2016-08-05:
I have enabled Z_DUAL_STEPPER_DRIVERS to get the two motors working so that has been OK all along.

I defined Z_DUAL_ENDSTOPS and set Z2_USE_ENDSTOP to _ZMAX_ as that is where my second z limit switch is connected. the first limit switch is built the probe and it's connected to ZMIN

I also enabled USE_ZMAX_PLUG in configuration.h

With these setting the printer homes the X and Y but not the Z

I'm probably missing something here so I'm open to suggestions.

brainscan, 2016-08-05:
the first limit switch is built the probe and it's connected to ZMIN

Im not sure but I don't think you can use one of the dual end stops as a probe too. You might need two endstops for z and another one for the probe.

mperdue, 2016-08-05:
I suppose that makes a certain amount of sense. I can add another switch and look at the configuration files to see what needs to be changed. This would necessitate having the Z leveling done near the upper limit of movement as opposed to the lower limit so it won't interfere with the Z Probe. Is this leveling initiated by a separate G code? If it's kicked off as part of G28 then it will really slow down starting up a print.

brainscan, 2016-08-05:
Depends on probe design and type of switch, could be done at z min without problems if you have a retractable probe or capacitance probe. Sorry haven't got this set up yet so don't know that much about its workings.

-

mperdue, 2016-08-05:
Looking at the code, it may be overkill for what I'm attempting to do. It looks to have an additional offset for the second motor. It seems to me if you have a probe with it's Z offset and you have a a second, adjustable switch set to trigger at the same point as the probe you can move the Z axis closer to the bed stepping both motors until one encounters a limit. Stop that motor and continue stepping the other until it trips the limit. Then back them both up and do a slow verify on each. At that point the Z of both stepper motors should be equal to the probe Z offset and off you go.

I guess I need to explore the code more. It's been a few years since I've done any programming so I'm a bit rusty. Retirement will do that to you.

brainscan, 2016-08-05:
I think it's better to have the axis level with the frame rather than the bed which can warp or move, and then use the probe and bed levelling to compensate for the bed.

mperdue, 2016-08-05:
I guess I have always tried to build my devices so the print bed is level to the frame so I tend to think relative to the bed. You have a point though and it makes a lot of sense. Currently I just manually verify it from time to time so maybe I'm the one over thinking things. :-)

mperdue, 2016-08-07:
I'm not having any luck getting this going. If anyone has done this I'd appreciate any feedback you might have on the proper configuration and connections.

boelle, 2016-08-07:
I have been wanted this too

and we should be able to use the touch probe for this

but only if the bed and whole Y carriage is up for it. as @mperdue my bed is level to the frame and i have verified that with a dial gauge.

i also use the same dial gauge to level the Z axis... which is a time consuming job.

btw... calling it Z axis leveling is kind of wrong as it's really the X axis you level relative to the bed/frame... Z axis is the one that goes up/down

boelle, 2016-08-07:
how i do this on my mendel90:

the 2 corner posts at the back of the bed are staic and not adjusted. i mave X carriage with dial cauge monted left and right at the back to get it level with the bed (or the 2 screw heads at the corners)

once i have left right 100% level i move the bed so i can do the same at the 2 corner posts at the front. 

these 2 are adjustable and i make sure that the bed is now level in the front/back direction by adjusting these 2 posts up and down.

i then crost check that all 4 posts are at the same level. they allways are but i check anyway.

BEFORE you do any comments please spend the time to see how a mendel90 bed looks like

and YES the bed can warp... but at least the 4 corners are now the same height. to overcome the warp issue i use a ikea mirror and i now adjust my Z height with the bed at 110 degrees and nozzle at 220

i always get a level 1st layer...

but it would be damm nice to have all this work automated...

mperdue, 2016-08-07:
I agree that it should be doable with the z probe as one of the sensors. In reality it could probably be done by positioning the probe at opposite ends of the x axis. It would be tricky because the pivot points of the axis would cause a bit of parallax to deal with.  

So far, I've been unable to get it to do anything that resembles checking the level.  There has been no movement that suggests that a leveling routine is being run.

thinkyhead, 2016-08-08:
@mperdue I've submitted a PR to fix `Z_DUAL_ENDSTOPS`, which appears to have been broken if homing to to the min end. The code for the feature in `Conditionals.h` was incomplete and flawed as well. #4562 should make `Z_DUAL_ENDSTOPS` behave correctly for you.

strips, 2016-08-08:
I need some advice as I want to do what's discussed here. I have a Bigbox and want to enable dual Z stepper drivers. I also have a couple of [optical switches of this kind](http://www.pibot.com/pibot-optical-endstop-rev16.html). They allow the bed to continue to move past without hitting anything physical. 

My problem are kids turning the Z couplings when I'm not watching :( So I often need to manually re level the bed.

I have not mounted the new optical Z endstops yet. Where is the best position for these Z endstops, max or min?. I do also have a IR Z probe that today only functions as Z endstop and hope it could be used for automatic MBL one day. Could it be used with the endstops to set the Z offset more correctly?

Blue-Marlin, 2016-08-08:
The algorithm for levelling x with the probe alone could be:

1. Measure the height of two points on the bed, as far apart as possible.
2. If the difference is below a threshold stop. (go 5)
3. Lift the lower side by the difference of the measured height.  
Because the spindles are more far apart then the measured points the raise will not be high enough, but the right direction. No overshot is possible.
4. go 1
5. End

In case the threshold is smaller then the probes error this will run endless. So limit the number of iterations.

strips, 2016-08-08:
In my case I will have stepper Z and stepper Z2 raising/lowering the bed. In this case I would like to have an automated Z and Z1 leveling to keep the bed platform as level to the frame as possible.

An added bonus of having the Z endstops at max would be a extra protection against the bed hitting the frame if it is already at max and have automatic raise when homing or I have misconfigured something. This has happened a few times when I was messing about.

mperdue, 2016-08-08:
Blue-Marlin - your algorithm for leveling with the probe only is pretty much what I had settled on as a way to do it. I need to take the time to learn how this software is put together so I can start making some of these changes myself.

mperdue, 2016-08-08:
I'll have a look at the latest code as soon as I can. Hopefully later today.

thinkyhead, 2016-08-09:
`Z_DUAL_ENDSTOPS` requires 2 endstops. It doesn't support using a Z probe. If you use a probe to home Z then you should just disable `Z_DUAL_ENDSTOPS`.

mperdue, 2016-08-09:
OK, fine. I'll see if I can code my own solution to this.

strips, 2016-08-09:
@thinkyhead: maybe I should make a separate feature request for this?

A nice feature would be to use both Z_DUAL_ENDSTOPS (2 Z endstops) for initial leveling and the probe for ABL / MBL. 

Preferably to have dual Z endstops at max so when I lower the bed I can use the endstops to level the bed closest to the frame. Then raise the bed up and do a ABL or MBL in the future. An added bonus for this is the limit on Z max.

mperdue, 2016-08-09:
That would certainly work for me. I plan to still look at things in the code but I'm retired and haven't done any coding for awhile - I'm pretty rusty. Still, it won't hurt for me to figure out how the code is setup and what it is doing.

mperdue, 2016-08-10:
I'm pretty new to github. How does one know when an item has been addressed?

thinkyhead, 2016-08-11:
> maybe I should make a separate feature request for this?

Yes. It would have to be a new feature.

> How does one know when an item has been addressed?

The original question was "Does this capability exist in Marlin?" The answer is clearly "no." If you need guidance in how to implement the feature, we can continue to discuss it.

> I'll see if I can code my own solution to this.

We love and appreciate it when people are into coding features themselves and contributing them. We're all very busy!

mperdue, 2016-08-11:
@thinkyhead, I understand that my original post was a question and that was addressed. However, @strips suggested adding the feature and refered to it in another item. My most recent question was just trying to get information on how this stuff is followed on github. Should the feature be added, how does one know when it has actually been implemented?

In the meantime, I'm digging through the code as my schedule permits to see how things are set up. Is there a programmers guide that helps or is it all in the comments?

thinkyhead, 2016-08-11:
> how this stuff is followed on github

Check regularly. I don't personally keep a list of everyone who wants to be contacted when something is done. Of course, if you're the implementer then you'll post a PR when your code is ready.

> how does one know when it has actually been implemented

I wouldn't worry about that for a long, long, _long_ while. I certainly don't have the time or inclination to add this, and I don't have dual-z motors, so it would require assistance from someone who does.

> Is there a programmers guide…?

No. You'll have to glean from the existing code how to implement such things. That's how we all worked it out. You might read the contribution and style guidelines [in the wiki](https://github.com/MarlinFirmware/Marlin/wiki).

thinkyhead, 2016-08-11:
Josef Prusa recommends moving Z all the way to the top until the motors grind, and then the two Z carriages will be level with each other, whether or not one uses one or two drivers.

mperdue, 2016-08-11:
@thinkyhead I'm not worried about or watching for the release of anything at this point, I was just asking a general question. I don't expect you or anyone else to drop what you're doing to jump on this. I'll take a look at the guidelines. Thanks.

cfelicio, 2016-11-29:
So, my Z gets out of sync once in a while, I'm thinking of using 2 endstops (one on each side), and also dual Z drivers (on ramps 1.4). I don't have a bltouch probe or anything. Would enabling these options get my Z to stay in sync when homing?

Thanks!

thinkyhead, 2016-11-30:
@cfelicio Josef Prusa recommends for his machines that user move all the way to the top until the steppers grrrrinnnddd, at which point the Z axes are level.

mperdue, 2016-11-30:
> Josef Prusa recommends for his machines that users move Z all the way to the top until the steppers grrrrinnnddd, at which point the Z axes are level.

Just guessing here but I think a better solution can be devised... :-)

Blue-Marlin, 2016-11-30:
What's better? In the categories cost effective, simple and reliable it's hard to beat the Prusa way.
If you want it the complicated way look for Z_DUAL_STEPPER_DRIVERS + Z_DUAL_ENDSTOPS.

mperdue, 2016-11-30:
I think the complicated way will lead to an increased lifespan of the drive components. One can ship a manual transmission in a car without using the clutch but it typically doesn't last as long. The same is true of the printer, I suspect.

cfelicio, 2016-11-30:
I use prusa method, but what I'm saying is that over time, Z gets out of sync (for me, after a few prints). Then I have to do the "grind" again. Since I have an unused driver on the ramps board, I was thinking of using  Z_DUAL_STEPPER_DRIVERS + Z_DUAL_ENDSTOPS, so when I home the printer, the left stepper stops when the left endstop trigger, and the right stepper stops when the right endstop trigger. Does that make sense? Would it work with the current Marlin implementation of  Z_DUAL_STEPPER_DRIVERS + Z_DUAL_ENDSTOPS? Thanks! :)

thinkyhead, 2016-12-02:
> left stepper stops when the left endstop trigger, and the right stepper stops when the right endstop trigger

@cfelicio Yes, that is how it's designed to work. The `M666` command is used to add an offset to correct for any inaccuracy.

cfelicio, 2016-12-02:
Awesome thanks! :)

closed 2016-12-05 by thinkyhead