MarlinFirmware/Marlin #4532: Kossel mini z-probe retraction issue

unnamedini, 2016-08-04:
I'm having some problems with the z-probe auto level when using the latest Marlin Firmware 1.1.0-RC7 

My Kossel Mini is using the Allen key retractable z-probe and the configuration file can be found here: [configuration.txt](https://github.com/MarlinFirmware/Marlin/files/402367/configuration.txt)

First see this gif of my kossel when trying to retract the probe: http://imgur.com/NhRPGJS
I'm going to include some commands from the configuration file to better explain this problem.

As you can see in the gif, when I execute the g30 command the z-probe hits the bed and then goes to the retract position:
```
  #define Z_PROBE_ALLEN_KEY_STOW_1_X -65.0
  #define Z_PROBE_ALLEN_KEY_STOW_1_Y 78.0
  #define Z_PROBE_ALLEN_KEY_STOW_1_Z 31.0
```

It continues to move the nozzle down further to push the probe into retracted position:
 ```
#define Z_PROBE_ALLEN_KEY_STOW_DEPTH 16
#define Z_PROBE_ALLEN_KEY_STOW_2_X  Z_PROBE_ALLEN_KEY_STOW_1_X
#define Z_PROBE_ALLEN_KEY_STOW_2_Y  Z_PROBE_ALLEN_KEY_STOW_1_Y
#define Z_PROBE_ALLEN_KEY_STOW_2_Z  (Z_PROBE_ALLEN_KEY_STOW_1_Z-Z_PROBE_ALLEN_KEY_STOW_DEPTH)
```

But when the allen key touches the screw, the z-min endstop is activated sending the TRIGGERED signal making the Marlin to move the nozzle up as if it's executing the next lines of command and leaving the probe in the deployed position.

```
// Raise things back up slightly so we don't bump into anything
#define Z_PROBE_ALLEN_KEY_STOW_3_X  Z_PROBE_ALLEN_KEY_STOW_2_X
#define Z_PROBE_ALLEN_KEY_STOW_3_Y  Z_PROBE_ALLEN_KEY_STOW_2_Y
#define Z_PROBE_ALLEN_KEY_STOW_3_Z  (Z_PROBE_ALLEN_KEY_STOW_1_Z+Z_PROBE_ALLEN_KEY_STOW_DEPTH)
```

After the commands above, the nozzle is moved to the center of the bed:
```
#define Z_PROBE_ALLEN_KEY_STOW_4_X 0.0
#define Z_PROBE_ALLEN_KEY_STOW_4_Y 0.0
```

And I get this on the console:
```
>>> g30
SENDING:G30
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Error:Z-Probe failed
[ERROR] Error:Z-Probe failed

Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
[ERROR] Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)

Bed X: 0.00 Y: 18.00 Z: 0.00
echo:endstops hit:  Z:230.69
```

My printer is working perfectly when I use the Marlin V1 firmware, I just thought in testing the new firmware and see if it is working with my printer. Apparently it is not and I don't know where to fix this issue in the code.

Blue-Marlin, 2016-08-04:
Please try
In MarlinMain.cpp
```
@@ -2069,10 +2066,11 @@ static void clean_up_after_endstop_or_probe_move() {
       // If endstop is already false, the Z probe is deployed
       if (_TRIGGERED_WHEN_STOWED_TEST == deploy) { // closed after the probe specific actions.
                                                    // Would a goto be less ugly?
       //while (!_TRIGGERED_WHEN_STOWED_TEST) { idle(); // would offer the opportunity
       // for a triggered when stowed manual probe.
+      if(!deploy) endstops.enable_z_probe( deploy ); //Switch off triggered when stowed probes early
     #endif
 
     #if ENABLED(Z_PROBE_SLED)
       dock_sled(!deploy);
     #elif HAS_Z_SERVO_ENDSTOP
```

unnamedini, 2016-08-06:
Thanks, that worked!

closed 2016-08-06 by unnamedini