MarlinFirmware/Marlin #4533: Fail to compile when using max6675 on RAMP 1.4 FSB

erikkallen, 2016-08-04:
Hi,

When I try to compile the latest commit 2c83e68 with E0 temp sensor -2 (MAX6675) on RAMPS1.4 FSB I get the error **'DIODIO52_DDR' was not declared in this scope**

Steps to reproduce:

* clone marlin form git
* change  TEMP_SENSOR_0 to -2 in Configuration.h
* Click compile

I am using Arduino version 1.6.9

The entire error log can be found [here](https://gist.github.com/erikkallen/ba0b84a30e51bba66e9946a604f3f2f8#file-error_log_marlin-txt)

Kind regards,

Erik

esenapaj, 2016-08-06:
PR #4419 (Switch Sd2CardLib to fastio) is causing the issue.
I tried to fix, but failed.
Because SCK, MISO, MOSI, SS in the [fastio.h](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/fastio.h#L960-L963) are macro,
```cpp
  #define SCK         DIO52
  #define MISO        DIO50
  #define MOSI        DIO51
  #define SS          DIO53
```

but those in the [pins_arduino.h](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/variants/mega/pins_arduino.h#L38-L41) are variables...
```cpp
static const uint8_t SS   = PIN_SPI_SS;
static const uint8_t MOSI = PIN_SPI_MOSI;
static const uint8_t MISO = PIN_SPI_MISO;
static const uint8_t SCK  = PIN_SPI_SCK;
```

thinkyhead, 2016-08-07:
#4554 is merged, and should address this.

erikkallen, 2016-08-07:
That seems to fix the compilation issue, however now it continuously reboots the devices upon connect. Should I create a new issue for that?

erikkallen, 2016-08-08:
SDSUPPORT was disabled, enabling it fixed the problem! Thanks! It looks like the SPI bus is not initialized without SDSUPPORT, should this be fixed?

closed 2016-08-08 by erikkallen

esenapaj, 2016-08-08:
I also confirmed the continuously reboots (freeze) when using MAX31855 without `SDSUPPORT`.
When freezing, red LED on RAMPS is flashing.
It can be reproduced without real thing of MAX31855. 

But when I enable `SDSUPPORT`, MINTENP Err is occurs on every startup...
But I can't judge that this is problem of Marlin or not.
Because this is my first using of MAX31855, thus maybe I'm mistaking something.

erikkallen, 2016-08-08:
I had the same MINTEMP issue, and it turned out I was using the wrong slave select pin marlin now uses pin D66 which can be locaed using this pinout http://www.geeetech.com/wiki/index.php/File:Ramps_interfaces.jpg

Blue-Marlin, 2016-08-08:
Please try:
```diff
@@ -1019,11 +1019,12 @@ void Temperature::init() {
   #if ENABLED(HEATER_0_USES_MAX6675)
 
     #if DISABLED(SDSUPPORT)
       OUT_WRITE(SCK_PIN, LOW);
       OUT_WRITE(MOSI_PIN, HIGH);
-      OUT_WRITE(MISO_PIN, HIGH);
+      SET_INPUT(MISO_PIN);
+      WRITE(MISO_PIN,1);
     #else
       OUT_WRITE(SS_PIN, HIGH);
     #endif
 
     OUT_WRITE(MAX6675_SS, HIGH);
```
MISO should be an input.

erikkallen, 2016-08-08:
Jep, that did the trick, it now works without SDSUPPORT awesome!

esenapaj, 2016-08-08:
I've tested the solution, but freezing wasn't solved for some reason.
Continued tomorrow...

・A branch that it's used for test: https://github.com/esenapaj/Marlin/tree/testes

esenapaj, 2016-08-13:
I'm still stucking, but now I'm doubting that [my hardware is broken](https://github.com/MarlinFirmware/Marlin/issues/4509#issuecomment-239603069).

ckcampbell248, 2016-09-02:
I'm still getting the error from the original post (DIODIO52_DDR was not declared in this scope) when trying to compile the latest build using TEMP_SENSOR_0 -2. #4554

thinkyhead, 2016-09-04:
@ckcampbell248 Please post your configuration so I can find out the cause.

ckcampbell248, 2016-09-04:
Thanks for the reply. Configuration.h attached (rename from .txt). 
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/454074/Configuration.h.txt)

Let me know if you need me to try anything. Thanks!

Grogyan, 2016-09-04:
+thinkyhead I reckon it may be a good idea to have in sanity_check for if
temp sensor is -2 or -3 make sure SDSUPPORT is enabled.

Seems to be a common overlooked item

Blue-Marlin, 2016-09-04:
I can compile his config without changes without error with the newest RCBugFix.
Have not tested on actual hardware but should also work without SDSUPPORT now.

ckcampbell248, 2016-09-04:
@Blue-Marlin where did you download the code? I got RC7 from the main site and it still fails after uncommenting SDSUPPORT only now I get the error on DIODIO53_DDR. What'd I miss?

Thanks.

Blue-Marlin, 2016-09-04:
Today. RC7 != RCBugFix!

ckcampbell248, 2016-09-04:
Found it. The code is compiling now and deployed to my board. Thanks!

alphapilot380, 2016-09-05:
I have this reboot issue also on the latest RCbugfix, I'm also using TEMP_SENSOR_0 -2
The reboot issue dissappears when I change 

#if DISABLED(SDSUPPORT)
  #define MAX6675_SS       66
#else
  #define MAX6675_SS       66
#endif

to

#if DISABLED(SDSUPPORT)
  #define MAX6675_SS       53
#else
  #define MAX6675_SS       66
#endif

in "pins_RAMPS.h"
I had the MAX6675 also like this in RC2 and it worked perfectly.

Blue-Marlin, 2016-09-05:
@alphapilot380 
What pin number to use totally depends on where __you__ connected the SS-line. Don't expect Marlin to know what you have done until you told Marlin what you have done. For every unusual setup, you have to edit pins_*.h .

thinkyhead, 2016-09-05:
@alphapilot380 Do you think that should be applicable to all setups? I notice that the `RIGIDBOARD` aims to use pin 53 under the same circumstances (pin 49 otherwise). But there are so many RAMPS variants that use `pins_RAMPS.h` I'd be hesitant to make this a universal change.

Grogyan, 2016-09-06:
@alphapilot380 do not use D53 as that pin is used for SD Card SS.
Do not use D49 as that is used for the SD Card Card Detect

I think that there is still comments in the code which express this, if there is a compile issue, then it is somewhere else.

I haven't tested the latest RCBugFix.  I'm using an version a few months old

thinkyhead, 2016-09-07:
Do you think it's worth adding a sanity check (with certain boards) to ensure that SDSS and SD_DETECT don't overlap with the 6675 pin?

Grogyan, 2016-09-07:
Possibly, as more and more people use thermocouples over thermistors, it is an easy thing to over look.
Especially when they use something like Instructibles.com as the main reference, which is laughably out of date.
I look forward to RAMS V2, or RAMBo, and others have another SPI breakout, or a SPI Breakout and have on board SD slot

thinkyhead, 2016-09-08:
So…
```cpp
#if ENABLED(SDSUPPORT) && HEATER_0_USES_MAX6675
  #if SDSS == MAX6675_SS
    #error "MAX6675_SS thermocouple pin conflicts with SDSS pin."
  #elif SD_DETECT_PIN == MAX6675_SS
    #error "MAX6675_SS thermocouple pin conflicts with SD_DETECT_PIN."
  #endif
#endif
```

korpx, 2016-09-19:
Why is this closed? I am still getting the error with RC7 and RCBugFix

thinkyhead, 2016-09-20:
@korpx It was closed by the original author because he solved it. That's how it works. Please post a new issue.