MarlinFirmware/Marlin #4536: Why Homing speeds  at mm/m

fabtopia, 2016-08-05:
Why must the homing speed be defined in mm/m instead if of mm/s like the rest

thinkyhead, 2016-08-07:
It's historical. And it's not just `HOMING_FEEDRATE_XY` and `HOMING_FEEDRATE_Z` are mm/m. Also…

- `X_PROBE_SPEED`
- `MANUAL_FEEDRATE`
- `M207 F`, `M208 F`
- `G0 F`, `G1 F`, `G2 F`, `G3 F`, etc.

Perhaps some others I'm forgetting…

Roxy-3D, 2016-08-07:
I'ld love to change it.   I would like everything consistent and in the same units.  But remember, we have a very large user base.  Changing the Feed Rates to different units will need to be done very carefully.   We need to provide a migration path.    One idea might be to have a //#define ALL_UNITS_IN_MM_PER_SEC that is turned off.   If the user turns it on, they do so knowing they need to convert different tables.

It is kind of like a doctor with the Hippocratic oath.   "Do no harm!"   If we could go back in time it would be a lot easier to make the shift.  But right now, we should be very careful and not just rush to 'Fix' something when we know it will cause great pain.   (That also doesn't mean we shouldn't fix it.  We should fix it.  But we need to do it in a very careful and considerate manner.)

@ThinkyHead   Do you have an opinion on the idea to allow the user to control the migration to consistent units?
~~~~
//#define ALL_UNITS_IN_MM_PER_SEC 
~~~~

thinkyhead, 2016-08-07:
@Roxy-3D A configuration option won't cut it. Either we change the configuration options to use mm/s, or we leave them as they are. CNC has historically use mm/m as the standard unit when specifying a feedrate through the `F` parameter in GCode. I believe we should preserve that. It seem to me that, if anything, we should convert every other user-facing setting to mm/m to be consistent with the CNC world that we are flirting with.

Roxy-3D, 2016-08-07:
Oh?   That was the exact opposite of what I was thinking.  I was thinking we should convert everything to mm/sec.  But if the standard is mm/min,  maybe that is the right direction to go.   Mostly, I would prefer all of the units are consistent and not mixed.    If that means moving to mm/min is more standard, I can live with that.

The whole idea for a configuration option is we we could use that to insulate the format of the user's configuration.h settings from how we do things in the code.  We could start moving towards consistent units but for some period of time we could respect the old format of units.    

I'm really not pushing hard for anything.  I'm mostly just suggesting we think about how to start getting things consistent.

thinkyhead, 2016-08-07:
Storage for mm/m units is 2 byte short, while mm/s units require 4 byte floats.

Internally, some of the code that uses mm/s as the stored unit still uses mm/m for both input and output. The disadvantage is that you need to add 0.001 to round up for display. Forward-facing units may generally remain mm/m, but internally all the storage could be done in mm/s. (Acceleration is always specified in mm/s/s because there it makes more sense. And mm/m/s just seems weird.)

thinkyhead, 2016-08-07:
#4556 makes things more internally consistent. Note the usage of `MMM_TO_MMS` (etc.) macros in various places, but not as many as before.

AnHardt, 2016-12-10:
How about closing this?

closed 2016-12-11 by Roxy-3D