MarlinFirmware/Marlin #4538: Mesh Bed Leveling bug

Roxy-3D, 2016-08-05:
The Mesh Bed Leveling routines have a bug...   With the numbers I have cell_index_x() should be returning 4 but it returns 3.
~~~~
    int8_t cell_index_x(float x) {
      int8_t cx = int(x - (MESH_MIN_X)) / (MESH_X_DIST);
      return constrain(cx, 0, (MESH_NUM_X_POINTS) - 2);
    }

    int8_t cell_index_y(float y) {
      int8_t cy = int(y - (MESH_MIN_Y)) / (MESH_Y_DIST);
      return constrain(cy, 0, (MESH_NUM_Y_POINTS) - 2);
    }
~~~~
This code (implemented as M45) will demonstrate the rounding problem.   The good news is...  It wasn't a compiler bug!
~~~~
inline void gcode_M45() {
float f1, f2, f3, f4, f5, f6, f7;
float start_x = test_get_cell_index_x( 90.997 );
float start_x2= bed_leveling_mesh.get_cell_index_x( 90.997 ); 

//	f1 = 91.08;		// The destination X coordinate of where the problem happens
	f1 = 90.997;		// The start X coordinate of where the problem happens
	f2 = 22.555555;		// MESH_X_DIST	
	f3 = MESH_X_DIST;

	f4 = f1 / f2;
	f5 = f1 / f3;

	dump_float("f1: ", f1);
	SERIAL_ECHO("\n");
	dump_float("f2: ", f2);
	SERIAL_ECHO("\n");
	dump_float("f3: ", f3);
	SERIAL_ECHO("\n");
	dump_float("f4: ", f4);
	SERIAL_ECHO("\n");
	dump_float("f5: ", f5);
	SERIAL_ECHO("\n");
	dump_float("start_x : ", start_x);
	SERIAL_ECHO("\n");
	dump_float("start_x2: ", start_x2);
	SERIAL_ECHO("\n");
	SERIAL_ECHOPAIR("     f1/f3  ", f1/f3 );
	SERIAL_ECHO("\n");
	SERIAL_ECHOPAIR("int( f1/f3) ", int(f1/f3) );
	SERIAL_ECHO("\n");
	dump_float("MESH_MIN_X: ", MESH_MIN_X );
	SERIAL_ECHO("\n");
	dump_float("MESH_X_DIST: ", MESH_X_DIST );
	SERIAL_ECHO("\n");
	dump_float("constrain(f4, 0,(MESH_NUM_X_POINTS)-2) : ", 
			constrain(f4, 0, (MESH_NUM_X_POINTS) - 2) );
	SERIAL_ECHO("\n");

int8_t cx = int(f1 - (MESH_MIN_X)) / (MESH_X_DIST);
int8_t cx1= int(f1) / (MESH_X_DIST);

	SERIAL_ECHOPAIR("cx = int(f1 - (MESH_MIN_X)) / (MESH_X_DIST) = ", cx );
	SERIAL_ECHO("\n");
	SERIAL_ECHOPAIR("cx1= int(f1)) / (MESH_X_DIST) = ", cx1);
	SERIAL_ECHO("\n");

	SERIAL_ECHO("\n");
} 
~~~~

epatel, 2016-08-05:
I wonder how the different versions from here stack up in your test https://github.com/MarlinFirmware/Marlin/issues/3968#issuecomment-224661850

Roxy-3D, 2016-08-06:
I had to run out the door so I didn't really spend much time once I found the bug.    But I think putting the parenthesis in the right place fixes things.   I'll definitely check it with all of the corner cases to make sure!    :
~~~~
    int8_t cell_index_x(float x) {
      int8_t cx = int( (x - MESH_MIN_X) / MESH_X_DIST );
      return constrain(cx, 0, (MESH_NUM_X_POINTS) - 2);
    }
~~~~
I also think I need to check all the boundary conditions for:
~~~~
    int8_t find_closest_x_index(float x) {
      int8_t px = int(x - (MESH_MIN_X) + (MESH_X_DIST) / 2) / (MESH_X_DIST);
      return (px >= 0 && px < (MESH_NUM_X_POINTS)) ? px : -1;
    }
~~~~
Taking an integer function on some floating points and then dividing that result by a floating point numbers makes me think we are going to have some numbers that cause trouble there too.    

@epatel   Have you been seeing artifacts on the outside surface of your prints at the mesh boundaries?    I think that bug was causing that.  But it was also causing the nozzle to make random movements off path.   They hardly ever happen.  But they do happen.   I only saw 1.5" stray movements but I was using a 10x10 grid.  One of my Beta testers saw much bigger unexpected deviations.

epatel, 2016-08-06:
@Roxy-3D Artifacts on boundaries? No, have not seen that at all. But, I am not running Scotts changes to MBL, or much from latest Marlin changes. This is my "Production" version I use, https://github.com/epatel/MarlinDev/tree/epatel/RigidBot_Big+RAMPS_1_4+BulldogXL
I have printed large objects over 20 hours coming out as good as they can, not seeing any MBL artifacts. Stepper? Planner? I see issues like this  https://github.com/MarlinFirmware/Marlin/issues/4521 that scares me a lot. I think I have explained enough about those methods. I have tried to point to obvious mistakes when I see them, but I can imagine there are a lot of small calculation mistakes that do not crash the printers but lowers the quality of prints.

Roxy-3D, 2016-08-06:
@epatel  Artifacts on boundaries? No, have not seen that at all. 

Yes, I understand why.  You are immune to the rounding error I found.   And there is a complication in mesh_buffer_line().   Because of the way it calculates and uses the n_distribution number, if you are close to a cell boundary and get the wrong index, the error for the other part of the coordinate gets extreme.   You have this code.   
~~~~
  int select_x_index(float x) {
    int i = 1;
    while (x > get_x(i) && i < MESH_NUM_X_POINTS - 1) i++;
    return i - 1;
  }
~~~~
I was looking at using your binary search code just because when we get the UBL code running on Delta's we are going to need High Resolution Mesh's.   If we are at 10x10 or 15x15 we aren't going to want to do an linear search.

I'm also thinking seriously about doing what you already investigated.  You turned get_probe_xy() into indexing a floating point array.   I'll have to re-do your timing of that because it sure seems like that should be faster than doing a floating point multiply. 
~~~~
    float probe_x[MESH_NUM_X_POINTS];
    float probe_y[MESH_NUM_Y_POINTS];
...
    float get_probe_x(int8_t i) { return MESH_MIN_X + (MESH_X_DIST) * i; }
    float get_probe_y(int8_t i) { return MESH_MIN_Y + (MESH_Y_DIST) * i; }

    // Use the array instead
    float f_get_probe_x(int8_t i) { return probe_x[i]; }
    float f_get_probe_y(int8_t i) { return probe_y[i]; }
~~~~
But with all that said, I think the biggest speed up is going to come from re-writing  the mesh_buffer_line() routine.  Right now it is recursive and it takes a bunch of calls to these support routines to get set up each time so the routine can recursively execute.   And then there is all the overhead to keep track of what has been split and not split.    I have an iterative version almost done that drastically cuts the calculations.   And there is another optimization it will do.  Since it is splitting on Mesh Lines, it doesn't have to do bi-linear interpolation for Z.  It can simply interpolate between the two Z values for that Mesh line where the print line intersects.   That will simplify the work and setup that calc_z0() does and what is need to call the non-bilinear calc_z() routine.   
~~~~
    float calc_z0(float a0, float a1, float z1, float a2, float z2) {
      float delta_z = (z2 - z1) / (a2 - a1);
      float delta_a = a0 - a1;
      return z1 + delta_a * delta_z;
    }
~~~~

thinkyhead, 2016-08-07:
> `cell_index_x()` should be returning 4 but it returns 3

Sounds like a float rounding error. You're most likely correct about the rounding failing here:
```cpp
int8_t cx = int(x - (MESH_MIN_X)) / (MESH_X_DIST);
```
…since `int(3.999999)` will give 3, when what you really want is 4.

Moving the parentheses is fine. This probably also fixes the issue just as effectively:
```cpp
int8_t cx = int((x + 0.0001) - (MESH_MIN_X)) / (MESH_X_DIST);
```
The result will be an `int` divided by an `int`, rounding down.

Roxy-3D, 2016-08-07:
> int8_t cx = int(x + 0.5 - (MESH_MIN_X)) / (MESH_X_DIST);

I don't think that fixes the problem.  We need the cell_index_x/y() function to return the cell it is in, not the cell line that is closest.      In mesh_buffer_line() there are 4 cases to decide how to handle things based on the direction of travel.   If we are close to a mesh line but return the wrong cell, really bad things happen because of how normalized_dist is leveraged to extrapolate the other 1/2 of the (x,y) coordinate.
~~~~
 if (dest_x > start_x && TEST(x_splits, dest_x)) {
    nx = bed_leveling_mesh.map_x_index_to_bed_location(dest_x) + home_offset[X_AXIS];
    normalized_dist = (nx - current_position[X_AXIS]) / (x - current_position[X_AXIS]);
    ny = current_position[Y_AXIS] + (y - current_position[Y_AXIS]) * normalized_dist;
    nz = current_position[Z_AXIS] + (z - current_position[Z_AXIS]) * normalized_dist;
    ne = current_position[E_AXIS] + (e - current_position[E_AXIS]) * normalized_dist;
    CBI(x_splits, dest_x);
  }
~~~~
In a few days, I'll have a faster version that is not recursive.  (I really like using recursion, but Delta's are out of CPU resources.  We have to minimize the overhead.)     Right now, I have the cell_index_x/y() routines working reliably and I've tested it against the problem cases I've found.    Give me a few more days to finish cleaning things up.

thinkyhead, 2016-08-07:
> We need the `cell_index_x/y()` function to return the cell it is in, not the cell line that is closest.

That is what the function does!

thinkyhead, 2016-08-07:
> I don't think that fixes the problem.

You're right! Actually, the cast to int is premature. The whole thing can be done in float, and cast to int at the end.

boelle, 2016-08-11:
Will this one be fixed in RC8 ?

thinkyhead, 2016-08-11:
@boelle It's already patched, see the merged PR linked above. #4555

Roxy-3D, 2016-08-11:
@thinkyhead   Is floating point multiplication faster than division?

thinkyhead, 2016-08-12:
> Floating point division is typically at least twice as expensive

Indeed! This is why I recently added the `planner.steps_to_mm[]` array. It allows axis steps to be converted to mm by multiplication rather than division.

boelle, 2016-08-12:
if its allready patched/fixed why is the issue still open?

Roxy-3D, 2016-08-12:
Probably the answer to the question is the people involved in the discussion still see value in continuing to understand the topic at a deeper level.   I'm not sure what ThinkyHead's exact motivation is, but I did a lot of work to figure out and design the UBL System.  And I did it with the primary intent of having it 'fix' the calibration problems that Delta printers seem to always run into.

The big problem is Delta's are almost out of processing power.  The UBL System adds another layer of kinematics under the hood of what Delta's have to do to move around on the bed.   If I can optimize some of the floating point calculations away to free up CPU cycles, I view that as more important than reducing the Issue Count by one.

@thinkyhead I have mesh_buffer_line() almost rewritten in an iterative fashion instead of doing its work recursively.  That saves a bunch of floating point calculations for two reasons.  First, the routine does not need to do all of the setup work to figure out which Mesh Cell it is in for each smaller section of the move that it gets called to do.  But the other opportunity for reducing floating point operations comes because as we cross the grid lines of the Mesh, we know we don't have to do a bi-linear approximation (which is done using 3 calls to calc_z0() )  We can accomplish it with a single call to calc_z0() in that special case.

I have code written in Visual Studio to verify the 'correctness' of what it is doing for a whole set of different lines.  And it does the parsing (correctly!!!) regardless of the direction it traverses the line.   I'll get more confidence in the re-write as soon as I get it integrated into the UBL code base.   I think once it is fully shaken out, it is just a matter of extracting that function and replacing the existing mesh_buffer_line() if you want to give Mesh Bed Leveling a boost in the RC family.

thinkyhead, 2016-08-16:
> I have mesh_buffer_line() almost rewritten in an iterative fashion

And its output is identical to the recursive one? That's good. I was trying to accomplish that, or at least save on some recursion, but didn't get far.

closed 2016-08-16 by thinkyhead

Roxy-3D, 2016-08-16:
Actually...  I should be further along than I am.  But I spent the 2 hours I had free to code on adding robust debugging (or really...  Verification code) to prove exactly that.    The curly braces are where the line started.   But I'm reversing the order of the print for easy verification that both directions produce the same intersections with the mesh lines.

This has some particularly trouble some lines in it.   I put the Mesh lines on nice multiples of 10 just to simplify the analysis:
~~~~
Starting...
========================================================
 [96.00,62.00] >>>--->  [35.00,59.00] {35.00, 59.00} [40.00, 59.25] [50.00, 59.74] [55.33, 60.00] [60.00, 60.23] [70.00, 60.72] [80.00, 61.21] [90.00, 61.70] [96.00, 62.00]
========================================================
 [96.00,62.00] >>>--->  [35.00,59.00] [35.00, 59.00] [40.00, 59.25] [50.00, 59.74] [55.33, 60.00] [60.00, 60.23] [70.00, 60.72] [80.00, 61.21] [90.00, 61.70] {96.00, 62.00}
========================================================
 [85.00,56.00] >>>--->  [34.00, 6.00] {34.00,  6.00} [38.08, 10.00] [40.00, 11.88] [48.28, 20.00] [50.00, 21.69] [58.48, 30.00] [60.00, 31.49] [68.68, 40.00] [70.00, 41.29] [78.88, 50.00] [80.00, 51.10] [85.00, 56.00]
========================================================
 [85.00,56.00] >>>--->  [34.00, 6.00] [34.00,  6.00] [38.08, 10.00] [40.00, 11.88] [48.28, 20.00] [50.00, 21.69] [58.48, 30.00] [60.00, 31.49] [68.68, 40.00] [70.00, 41.29] [78.88, 50.00] [80.00, 51.10] {85.00, 56.00}
========================================================
 [51.00,23.00] >>>--->  [34.00, 6.00] {34.00,  6.00} [38.00, 10.00] [40.00, 12.00] [48.00, 20.00] [50.00, 22.00] [51.00, 23.00]
========================================================
 [51.00,23.00] >>>--->  [34.00, 6.00] [34.00,  6.00] [38.00, 10.00] [40.00, 12.00] [48.00, 20.00] [50.00, 22.00] {51.00, 23.00}
========================================================
 [97.00,16.00] >>>--->  [14.00,56.00] {14.00, 56.00} [20.00, 53.11] [26.45, 50.00] [30.00, 48.29] [40.00, 43.47] [47.20, 40.00] [50.00, 38.65] [60.00, 33.83] [67.95, 30.00] [70.00, 29.01] [80.00, 24.19] [88.70, 20.00] [90.00,
 19.37] [97.00, 16.00]
========================================================
 [97.00,16.00] >>>--->  [14.00,56.00] [14.00, 56.00] [20.00, 53.11] [26.45, 50.00] [30.00, 48.29] [40.00, 43.47] [47.20, 40.00] [50.00, 38.65] [60.00, 33.83] [67.95, 30.00] [70.00, 29.01] [80.00, 24.19] [88.70, 20.00] [90.00,
 19.37] {97.00, 16.00}
========================================================
 [93.00,27.00] >>>--->  [25.00,19.00] {25.00, 19.00} [30.00, 19.59] [33.50, 20.00] [40.00, 20.76] [50.00, 21.94] [60.00, 23.12] [70.00, 24.29] [80.00, 25.47] [90.00, 26.65] [93.00, 27.00]
========================================================
 [93.00,27.00] >>>--->  [25.00,19.00] [25.00, 19.00] [30.00, 19.59] [33.50, 20.00] [40.00, 20.76] [50.00, 21.94] [60.00, 23.12] [70.00, 24.29] [80.00, 25.47] [90.00, 26.65] {93.00, 27.00}
========================================================
 [87.00,78.00] >>>--->  [33.00,73.00]   line type == Horizontal {33.00, 73.00} [40.00, 73.65] [50.00, 74.57] [60.00, 75.50] [70.00, 76.43] [80.00, 77.35] [87.00, 78.00]
========================================================
 [87.00,78.00] >>>--->  [33.00,73.00]   line type == Horizontal [33.00, 73.00] [40.00, 73.65] [50.00, 74.57] [60.00, 75.50] [70.00, 76.43] [80.00, 77.35] {87.00, 78.00}
========================================================
 [13.00,68.00] >>>--->  [11.00, 3.00]   line type == Vertical {11.00,  3.00} [11.22, 10.00] [11.52, 20.00] [11.83, 30.00] [12.14, 40.00] [12.45, 50.00] [12.75, 60.00] [13.00, 68.00]
========================================================
 [13.00,68.00] >>>--->  [11.00, 3.00]   line type == Vertical [11.00,  3.00] [11.22, 10.00] [11.52, 20.00] [11.83, 30.00] [12.14, 40.00] [12.45, 50.00] [12.75, 60.00] {13.00, 68.00}
========================================================
 [ 6.00,78.00] >>>--->  [57.00,76.00]   line type == Horizontal {57.00, 76.00} [50.00, 76.27] [40.00, 76.67] [30.00, 77.06] [20.00, 77.45] [10.00, 77.84] [ 6.00, 78.00]
========================================================
 [ 6.00,78.00] >>>--->  [57.00,76.00]   line type == Horizontal [57.00, 76.00] [50.00, 76.27] [40.00, 76.67] [30.00, 77.06] [20.00, 77.45] [10.00, 77.84] { 6.00, 78.00}
~~~~

thinkyhead, 2016-08-16:
I look forward to seeing the final results.

> I'm going to collapse the 4 different directions down to one next.

Good!

I see you like 4-space tabs. Code bound for Marlin should use only 2 spaces.

Better than the 4-tilde method, three back-ticks plus the letters "cpp" will make a nice C++ syntax-colored code block.
~~~~
```cpp
~~~~

Roxy-3D, 2016-08-16:
I'll convert to two spaces per indent before I fold it in.  It is easier for me to see things with it indented deeper.   Part of the problem is we have spaces and tabs intermixed everywhere.   I wonder if we should turn everything into spaces with uniform indentation prior to promoting RC-8 to 'Stable' ???   Right now, it is so random.

I see what you mean by the 3-tilde-cpp method.   That does look nicer.

thinkyhead, 2016-08-19:
> Part of the problem is we have spaces and tabs intermixed everywhere

Nope! The only file that still has tabs is `Makefile`.

Roxy-3D, 2016-08-19:
Oh?   Well..   the Tabs have certainly snuck back into the .h and .cpp file in the devel_ubl branch!
I'll start searching for them and get them converted to spaces.

thinkyhead, 2016-08-19:
> Tabs have certainly snuck back

You'll have to watch your text editor settings. What is your text editor of choice?

Roxy-3D, 2016-08-19:
Pretty much, I use Vim.