MarlinFirmware/Marlin #4539: The Language Translations

yhfudev, 2016-08-05:
Is it possible to update the language engine?

The code use two libs to handle the low level display driver, the u8g and others (such as Hitachi HD44780). The u8g uses a strange way to handle various languages by re-code the chars. There exists a wrapper that uses a straight way to use the strings by a pre-processed manner (https://github.com/yhfudev/u8glib-fontutf8.git). For example, in file language_cn.h, we can define the value in readable manner:
#define WELCOME_MSG MACHINE_NAME  _U8GT("已备"),
instead of
#define WELCOME_MSG "\xa4\xa5\xa6\xa7"

So the developer can easily to update the translations without worry about both the 'code' and the fonts.

thinkyhead, 2016-08-07:
This is a good idea. You will implement this for us?

yhfudev, 2016-08-07:
I would like to implement it. But before I start work, we need to discuss the details of the current implementation with the developers who finished it. We have to change a lot of files in the source code folder.

thinkyhead, 2016-08-08:
@yhfudev The great thing about Github is that you can do this work entirely in your own fork, get the implementation ready, work out the bugs. And we only have to start thinking about integration later, after the code has been fully written. In this way you are not restricted to following our current methodologies, we don't have to divert the main project at all while you do your thing, and you will have, at least, a reference implementation in hand. It helps in the process if you make the effort to keep up-to-date in your fork with the latest changes (using `git rebase`). In the end you'll need to do a `git rebase` and patch it up for submission anyway.

boelle, 2016-08-14:
@yhfudev are you doing this one?

thinkyhead, 2016-08-19:
@yhfudev We're very much looking forward to seeing how your work progresses. Keep us posted!

closed 2016-08-19 by thinkyhead

yhfudev, 2016-08-19:
Yes, I'm working on this. Unfortunately, I can't promise to have enough time because of other personal issues(job and visa). But I would like to discuss with whom can involved in this module.

yhfudev, 2016-08-21:
After looking through the source code, I found we don't need to change much to support both U8G and HD44780 at the same time. We can replace most of the language code with one or two files. The only extra step is to use a script to parse the UTF8 strings from the new language engine and generate the font and codes, which needs be done only when the language file changed.

yhfudev, 2016-08-23:
Finished HD44780 part, Here's the language definition source:

#ifndef LANGUAGE_JP_H
#define LANGUAGE_JP_H 1

#define MACHINE_NAME _UxGT("3Dプリンター") //_UxGT("3D Printer")

#define WELCOME_MSG                         MACHINE_NAME _UxGT(" レディー.") //MACHINE_NAME _UxGT(" ready.")
#define MSG_SD_INSERTED                     _UxGT("カードガソウニュウサレマシタ") //_UxGT("Card inserted")
#define MSG_SD_REMOVED                      _UxGT("カードガアリマセン") //_UxGT("Card removed")
#define MSG_LCD_ENDSTOPS                    _UxGT("エンドストップ") //_UxGT("Endstops") // Max length 8 characters
#define MSG_MAIN                            _UxGT("メイン") //_UxGT("Main")
#define MSG_AUTOSTART                       _UxGT("ジドウカイシ") //_UxGT("Autostart")

#endif // LANGUAGE_JP_H

And the the attached pic is the results.

[image]

TODO:
* support U8GLIB
* integrate both the HD44780 and U8GLIB module to Marlin

yhfudev, 2016-08-24:
Finished:
* UTF-8 converstion for HD44780
* UTF-8 converstion for U8G, the conversion tools and source code

TODO:
* integrate both module to Marlin

I also ordered a Russian HD44780, and I'll test it with Russian language once I received it.

Here's the result for the Russian language mixed with English showing in a I2C graph display:
[image]

And Chinese version:
[image]

Finally, the Japanese version:
[image]

thinkyhead, 2016-08-24:
It looks very good. Now we just have to figure out how to integrate the necessary conversion into the workflow. I suppose we can just ask our contributors to run the conversions.

yhfudev, 2016-08-26:
I get it worked.

It's now supports UTF-8 language translation header files, and generate the font data automatically.
So the language maintainer can use unicode to translate their languages and don't need care about
the font if the font contains all of the glyph of their languages.
 
The HD44780 is also works.

The document(howto) for the new language engine is in the file buildroot/share/fonts/uxggenpages.md.

There's some issues I can't handle:
1. some glyph are missing from the default 9pt font (WQY), this cause missing char of words in Greek(pl).
2. It would be better to change the suffix of the language file to reflect the language/country.
For example: jp_kana(Japaness with kana), zh_cn(Simplified Chinese), zh_tw(Traditional Chinese) etc.
Since all of the language files are UTF-8 encoded, so we don't need the utf8 suffixes now.

Here's the results:
[image]
[image]
[image]

thinkyhead, 2016-08-26:
Quite an incredible effort. Thank you for putting in the time. I see you did in fact write up some documentation. Great! With the additional code cleanups, this will be a most welcome addition to the project.